patman15/BMS_BLE-HA #2: some minor things missing on type A

gkathan, 2024-04-01:
hi sir,

I gave it a final look and detected following issues here in my setup (type-A battery)

1) two values are always "0" (Current and Power) - it looks like "Current" matches what I see in the offgridtec app (see screenshot), "Power" is not shown in the app ... but should also not be zero

2) very important (at least for me) is to see (as displayed in the app) that the battery is either "charging", "discharging" or "idling" => but I suspect that this is just a calculated value derived from simple rule to check whether "Current" is positive, negative or zero value => but would be nice to gave that as an entity too in the integration

3) the switching between using your integration and the offgridtec is currently also a bit quirky => tested to disable the integration => this worked after some time (30 min) that the offgridtec app again could connect BUT from then on it was no longer possible to re-enable succesfully the integration => it just showed "unavailable error" => only removing the device and the integration and re-adding the integration cured the situation 

OK thats it from my side - i hope we can fix these last glitches => then it will be perfect and maybe even point 3) is irrelevant as all relevant data is provided by HA

cheerz 
greetings from vienna
[image]
[image]

patman15, 2024-04-01:
Hi!

> I gave it a final look and detected following issues here in my setup (type-A battery)
> 
Thanks a lot!

>     1. two values are always "0" (Current and Power)
> 
This should be fixed in [version 1.0.1](https://github.com/patman15/BLE_BMS-HA/releases/tag/v1.0.1). Need to dig deeper into the topic, but it should work now. Thanks for spotting this!

> 2. very important (at least for me) is to see (as displayed in the app) that the battery is either "charging", "discharging" or "idling" => but I suspect that this is just a calculated value derived [...]
Yes, the app just goes with the current. I could add a binary sensor `battery_charging` which is foreseen by HA. Is that what you would be looking for? (This would not cover the state `idle` though.)

>     3. the switching between using your integration and the offgridtec is currently also a bit quirky
>
True, did not think of this as a real use-case ... Anyway, to switch from the app to the integration, you need to click `Scanning` on the right top of the app and confirm to close the bluetooth connection. If you just close the app it still blocks the connection to the BMS and the integration cannot read values. The other way round I need to dig much deeper, as I have not seen how HA cleans up connections on deactivation actively.

gkathan, 2024-04-01:
hej,
just updated to 1.01 - to be 100% sure the updated version is actually used I even removed the device and re-added the integration which found the battery immediatly ...

ad 1) this still shows zero at "Current" and "Power"  but there is a router + raspi4 running in my van eating around 0,6A constantly - which i can see that the SOC percentage slooowly is going down (so in general it seems to work) - again - do you have a type-A battery to test your code ? or shall I jump into this role ;-) what I anyhow already do :-) 

ad 2) yez - should not be a big deal to add this binary sensor ... would appreciate it 

ad 3) agree this sounds a bit more complicated especially if connection handling and termination seems to be quite vague :-) 

thanx & cheerz
[image]

patman15, 2024-04-01:
ad 1) weired, I tested it with my battery and also with my emulated A type battery. Could you please enable debug mode, ensure there is some current (in or out) and send attach the log? (If you leave debug on for some minutes and then disable it, HA will automatically offer you the download.)
Log would be great as next step for analysis. Thanks.

gkathan, 2024-04-02:
good morning,

* ok did also just fully reboot HA to be sure 
* still zeroes in Current and Power 
* and yez - there is definitely current in the battery (as you eg see in the SOC history graph - this is going down accordingly, and as I said it's a POE switch, peplink router and raspi4 HA running ... otherwise I would not be able to access the van here from home 💚✨)
* ok turned on debug now for 5 minutes - log attached
see ya later[image]
[image]

gkathan, 2024-04-02:
just one more information:
* it's a rather low current ... I guess around 0,6Ah ... as these 3 devices consume around 20W ... 
* ... in 2 weeks two fixed solar panels (250W) will be mounted on my van roof and will hopefully keep my lipo4 battery up with that constant load 
cheerz

patman15, 2024-04-02:
Morning! Please provide me a debug log (see my previous message on how to do it), so I can check what's happening! Thanks!

gkathan, 2024-04-02:
should also be attached in my previous GitHub comment ... but here again :-)

gerold kathan :: +43 699 1973536

2024-04-02 06:22:29.888 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration bms_ble which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-04-02 06:22:29.891 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration victron_ble which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-04-02 06:22:29.894 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-04-02 06:23:00.454 ERROR (MainThread) [metno] Access to https://aa015h6buqvih86i1.api.met.no/weatherapi/locationforecast/2.0/complete returned error 'ClientConnectorError'
2024-04-02 06:23:01.247 WARNING (MainThread) [motioneye_client.client] Connection failed to motionEye: Cannot connect to host 127.0.0.1:28765 ssl:default [Connect call failed ('127.0.0.1', 28765)]
2024-04-02 06:23:02.706 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-04-02 06:23:35.331 WARNING (MainThread) [motioneye_client.client] Connection failed to motionEye: Cannot connect to host 127.0.0.1:28765 ssl:default [Connect call failed ('127.0.0.1', 28765)]
2024-04-02 06:23:45.591 WARNING (MainThread) [motioneye_client.client] Connection failed to motionEye: Cannot connect to host 127.0.0.1:28765 ssl:default [Connect call failed ('127.0.0.1', 28765)]
2024-04-02 06:24:05.046 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for hamburgesp @ 192.168.1.199: Timeout while connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='192.168.1.199', port=6053))] (TimeoutAPIError)
2024-04-02 06:25:11.860 DEBUG (MainThread) [custom_components.bms_ble] BMS SmartBat-A42491 data update
2024-04-02 06:25:11.860 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] BMS SmartBat-A42491 already connected
2024-04-02 06:25:11.860 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] command: +RAA0201
2024-04-02 06:25:11.861 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] ble cmd frame bytearray(b">GTT%\'%$")
2024-04-02 06:25:12.053 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] Received BLE data: bytearray(b">GQ9%\' ,\x18\x1f")
2024-04-02 06:25:12.054 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] response: +RD,0259
2024-04-02 06:25:12.054 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] Decoded data: reg: battery_level (#2), raw: 89, value: 89
2024-04-02 06:25:12.055 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] command: +RAA0403
2024-04-02 06:25:12.055 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] ble cmd frame bytearray(b'>GTT%!%&')
2024-04-02 06:25:12.253 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] Received BLE data: bytearray(b'>GQ9%!"W!!%T\x18\x1f')
2024-04-02 06:25:12.254 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] response: +RD,047B440A
2024-04-02 06:25:12.254 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] Decoded data: reg: cycle_capacity (#4), raw: 175310, value: 175.31
2024-04-02 06:25:12.255 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] command: +RAA0802
2024-04-02 06:25:12.256 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] ble cmd frame bytearray(b">GTT%-%\'")
2024-04-02 06:25:12.553 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] Received BLE data: bytearray(b'>GQ9%-PV&&\x18\x1f')
2024-04-02 06:25:12.554 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] response: +RD,08EC33
2024-04-02 06:25:12.554 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] Decoded data: reg: voltage (#8), raw: 13292, value: 13.292
2024-04-02 06:25:12.555 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] command: +RAA0C02
2024-04-02 06:25:12.555 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] ble cmd frame bytearray(b">GTT%V%\'")
2024-04-02 06:25:12.753 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] Received BLE data: bytearray(b'>GQ9%V!,%W\x18\x1f')
2024-04-02 06:25:12.754 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] response: +RD,0C490B
2024-04-02 06:25:12.754 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] Decoded data: reg: temperature (#12), raw: 2889, value: 15.8
2024-04-02 06:25:12.755 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] command: +RAA1003
2024-04-02 06:25:12.755 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] ble cmd frame bytearray(b'>GTT$%%&')
[... cut ...]
2024-04-02 06:30:54.260 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] Decoded data: reg: cycles (#44), raw: 8, value: 8
2024-04-02 06:30:54.260 DEBUG (MainThread) [custom_components.bms_ble.plugins.ogt_bms] Data collected: {'battery_level': 89, 'cycle_capacity': 2328.7584, 'voltage': 13.292, 'temperature': 15.8, 'current': 0.0, 'runtime': 530040, 'cycles': 8, 'power': 0.0}
2024-04-02 06:30:54.261 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching SmartBat-A42491 data in 1.399 seconds (success: True)

patman15, 2024-04-02:
Sorry, overlooked the file below the huge picture. 😎 I'll have a look at it soon, first glance indicates that A type batteries use a different data format for the current. Strange implementation.

patman15, 2024-04-02:
Hey @gkathan, I just drafted a fix for the current issue. Can you check the v1.0.2-beta version? You can download it via HACS by enabling beta versions after clicking "download again". Let me know if it helps. Thanks a lot!

gkathan, 2024-04-02:
ok
will try right now ...

gerold kathan :: +43 699 1973536

gkathan, 2024-04-02:
FANTASTIC 🤘💚

you nailed it ! now both values are accurate 

awesome

so to keep you busy ;-) one more (very minor) feedback for the "Runtime" Sensor .. would be cool to have it also split in days (d) when Runtime is larger than 24h as current format is not really human readable ( how many days are 123h again 🤭) and if you still would consider to add the boolean value (charging/discharging) we have a perfect version 1.2 

curious as I am ... what was the issue withe two values ?[image]

cheeerz from vienna

patman15, 2024-04-02:
>     * the Current and Power values seem a bit too low ... will check tomorrow by comparing what the offgridtec app shows with the integration
> 
I think I know. Factor of 10 missing, give me a minute ...

patman15, 2024-04-02:
Fixed. I did not do a new version, just select `main` under beta-versions.

gkathan, 2024-04-02:
...trying to select "main" ... but always jumps back to 1.2-beta ... seems to be sticky ... cannot see r anything else than 1.2-beta 

any ideas ? what does the comment window tell me ? "..use service call for update entities .." 

🤔[image]

[image]

gkathan, 2024-04-02:
ok ... got it ... stupid me was on the wrong HA instance 🤭😂

values look very reasonable now 

good night - let's follow up tomorrow
c

patman15, 2024-04-03:
@gkathan so, good news, just published [v1.1.0-beta](https://github.com/patman15/BLE_BMS-HA/tree/v1.1.0-beta) which has:
- the strongly missed charging indicator
- feature to disconnect the battery on disabling it in HA (to allow access by the app again, remember to click `scanning` in the app to disconnect the battery if switching back to the integration)
- fixed exception handling

Hope that solves your request stuff in this issue. :smile:

> "Runtime" Sensor .. would be cool to have it also split in days

I haven't found a way to influence the presentation by HA. Also I did not find an example where this is handled, e.g. my UPS integrations acts the same. If you have hints, let me know.

Enjoy and let me know if you see issues so I can make it a final version or fix it.

patman15, 2024-04-04:
> just installed beta 1.1.0 .. but on first try I had just "unknown" on all entities ...

I removed the initial update, so time to first values was longer. This was to speed up HA startup time. Nevertheless, all integrations block for the initial update, so I will reintroduce it.
Should not affect functionality in any way, just takes 30s longer to show first values.

gkathan, 2024-04-04:
hi,

so just drove home and tested also the switch to the app by disabling the integration => works like a breeze - great (forth and back)

regarding the initial startup => fully agree with what you said 

one last thisng => could you replace the text in "Charging" sensor / entity to "discharging" instead of "not charging" as this makes clear that something is pulling from the battery => "not charging" could also mean the battery is idling - just some minor semantic comment ;-)

thanx

patman15, 2024-04-04:
Hi @gkathan,

some further  solutions for you:
> "Runtime" Sensor .. would be cool to have it also split in days (d)

You can use a template sensor or a template card, e.g. [Mushroom template card](https://github.com/piitaya/lovelace-mushroom) with the following template:
`{{ timedelta(seconds=int(states("sensor.smartbat_aXXXXX_runtime"), 0)) }}` results in `4 days, 4:20:00`

> could you replace the text in "Charging" sensor / entity to "discharging"

Sorry, the [HA semantic](https://github.com/home-assistant/core/blob/cceea6dac2bbff2958996f77be4352180b6deb6a/homeassistant/components/binary_sensor/__init__.py#L46) says that the sensor needs to be called `BATTERY_CHARGING`. Nevertheless, you can create your own [threshold sensor](https://www.home-assistant.io/integrations/threshold/) based on the current that does exactly what you want. You can even define something other than 0 or hysteresis in addition. :smile:

Are we good then with version 1.1.0-beta? I did some further fixes on the startup issue that I would put into the final version.

gkathan, 2024-04-05:
did not mean the name of the sensor ;-) but the value of the sensor when it is not charging should be "discharging"

gkathan, 2024-04-05:
ok got it 🤭 ... yes the threshold thing makes sense ... thänx 🤘 too early in the morning

yez - ready to go for 1.1

patman15, 2024-04-05:
issue(s) fixed by [release v1.1.0](https://github.com/patman15/BLE_BMS-HA/releases/tag/v1.1.0) thanks @gkathan

closed 2024-04-05 by patman15

gkathan, 2024-08-12:
ahoi 
the integration is still working like a breeze 💚🤘 ... short question ... are you aware that offgridtec released a new variant of it's LiPo4 battery ?
 https://www.offgridtec.com/offgridtec-12-200-lifepo4-pro-200ah-2560wh-lithiumbatterie-12-8v.html

do you have a clue whether this new variant is working with current integration ? 
cheeerz from vienna

patman15, 2024-08-12:
Hoi!

> are you aware that offgridtec released a new variant of it's LiPo4 battery ? https://www.offgridtec.com/offgridtec-12-200-lifepo4-pro-200ah-2560wh-lithiumbatterie-12-8v.html
> 

I did some fixes on the weekend (not yet in a release) and noticed that Offgridtec has deleted all apps from the Play Store. :angry: So I cannot check against the app anymore because I temporarily deleted it from my phone.

> do you have a clue whether this new variant is working with current integration ?
>
I can't see any hint that this battery supports Bluetooth, so I'd say no. And, as said, there are no apps in the store anymore which is also a clue towards that.

Grüße aus dem heißen Wien,
Patrick

gkathan, 2024-08-12:
> Offgridtec has deleted all apps from the Play Store. 😠 So I cannot check

that's more than weird ... how about all customers with legacy batteries ?
lucky that your Intengration works rock solid 🤘
gerold kathan | +43 699 19743536 | www.kathan.at

patman15, 2024-08-13:
Hi!
I have no idea how existing customers can use their batteries now. In any case they shouldn't delete their app. B-)
Since I lost it l, I need to be especially careful with any future changes. Luckily, I still have my emulator and 100% test coverage. :-D
Enjoy the hot weather!

gkathan, 2024-08-13:
I still have it on my phone ... shall I extract an APK for you ?
gerold kathan | +43 699 19743536 | www.kathan.at