patman15/BMS_BLE-HA #19: Support for JK BMS

downset, 2024-05-30:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I would love to see support for the new JK Inverter BMS.

### Describe the solution you'd like

Found your project through the Offgrid Garage video and have the same problem as Andy with the Batmon addon, because i would like to use the ESPHome BLE proxy. 
So your project would be a perfekt solution.
Are you planing to add the JK Inverter BMS to your solution?
I opened this Feature Request to track the progress (if integration is planed) and maybe can help Beta testing
Best Regards

patman15, 2024-05-30:
Hi @downset, as I do not own such a BMS, I would need more information in order to be able to add it to the supported BMS as I need to be able to test the implementation to assure some quality. Can you specify in more detail, which BMS you are looking at? JK seems to have a number of different ones ...
Do you have any documentation regarding the protocol? Is there other software supporting this BMS? Last resort would be that (with your extensive support) we could try to reverse engineer the most important communication parts.

downset, 2024-05-30:
Thanks for the reply.
For me I'm speaking about the JK-PB2A16S-20P but i think all JK-PB'X'A16S-'XX'P have same protocol
offgrid garage is using this BMS with the Batmon HA Addon in his Video und seams as it work with the from him called "JK" protocol, but not with Bluetooth proxy caused this is an Addon and no Integation in HA as yours, thought maybe it would be easy to Integrate this code as well... but maybe its not so simple...
Please tell me if i can help in any way
The Data to the BMS:
https://www.jkbms.com/product/jk-pb2a16s20p-jk-pb2a16s-20p/

patman15, 2024-05-31:
Thanks a lot, that is already good information for me. If it works with the Batmon HA and the protocol "JK" it should be doable. I tried to build a fake battery to be able to test the setup but failed so far. The BMS seems to have some very [strange/specific properties](https://github.com/fl4p/batmon-ha/issues/83#issuecomment-1514602250) regarding BT. Can you confirm that your BMS works with this app https://play.google.com/store/apps/details?id=com.jktech.bms ? I might need some more info to properly detect the BMS later. :thinking:

downset, 2024-05-31:
Sorry I can not confirm this app cause I don't own an Android Phone - but it is the app JK QR code in the Manual links for this BMS
I am using this app:
https://apps.apple.com/de/app/%E6%9E%81%E7%A9%BAbms/id1425725691
But I think it is the Same just for Apple

patman15, 2024-06-01:
Hi! I spent yesterday quite some time to try to get a fake battery that I can test with the app but it didn't work out for unknown reason. So testing against a known good seems to be out of reach. :disappointed: Anyhow, if you like, I can port the [implementation from batmon](https://github.com/fl4p/batmon-ha/blob/master/bmslib/models/jikong.py), do some dry runs with samples I found and you test it real life?
(There is a list of unknowns as it seems that all the other implementations have been reverse engineered, thus are inconsistent or weakly documented.)

downset, 2024-06-01:
Thank you very much for your efforts.
Yes i have the newest firmware V15.17 where V15 is the HW revision.
Of cause i would like to test if you can port the batmon implementation untested :)

Greets downset

patman15, 2024-06-03:
Ok, great thanks. I already started working on it, device detection works, shouldn't take too long till I have an alpha version for testing. Allow me some time, I can only work some evenings or weekend. I'll give you a ping when I have something ready to be checked.
Greetings!

patman15, 2024-06-06:
Hi @downset, I have a test implementation you could try on the [jikong_bms branch](https://github.com/patman15/BMS_BLE-HA/tree/jikong_bms). You can install it [manually](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual). If you do not succeed, I could also generate a draft release. Let me know the outcome, a debug log would be great, independent of the actual result. That would help me to verify some assumptions.
Thanks!

downset, 2024-06-07:
The Integration now finds both BMS but no data in the Entitäten. Hope This is the logfile you need
[home-assistant_bms_ble_2024-06-07T07-18-29.574Z.log](https://github.com/user-attachments/files/15721672/home-assistant_bms_ble_2024-06-07T07-18-29.574Z.log)
Tell me if i can do anything to address the problem
Greets
Downset

patman15, 2024-06-07:
Hi! 
> For me I'm speaking about the JK-PB2A16S-20P but i think all JK-PB'X'A16S-'XX'P have same protocol
>
I actually doubt that. Had a look at the log and it looks like you have a totally different BMS (protocol). Did you ever try it with BatMON? Basically, BatMON supports a subset of the BMS listed here https://github.com/syssi/esphome-jk-bms and yours is not listed at all.

downset, 2024-06-07:
mhh thats strange,
 I haven`t tried Batmon by my self cause my Homeassistant VM has no Bluetooth i have to use bluetooth Proxy. 
 I just watched the Videos from Andy on Youtube the "off-grid-garage" and he uses Batmon with many variations of this BMS HW version 14 and 15 and also different max current models and a github post on Batmon Repository in conversation wit Andy pointed me to your implementation
 
 https://youtu.be/Iq0vvR-mQUs?si=i9ikFvMm_VU6aYL4

patman15, 2024-06-08:
Hi! I had a look at the video and I think we are talking two different BMS types here. Andy is always talking about is "battery#2" and "battery#3" is testing with BatMON. In the video, timecode 11:38 I can see his Bluetooth scan where those batteries are mentioned and there is another JK_PB2A16S20P-06 he is never talking about. Unfortunately, he doesn't mention the type of BMS is actually using/testing and I'm quite sure he did not try that as his configuration does not contain the MAC address of the JK_PB2A16S20P-06.

Your BMS seems to have a separate UART BT module that needs to be configured. A hint is here https://github.com/syssi/esphome-jk-bms/discussions/412#discussioncomment-8131531. I assume that this is the issue as I can see the BMS sending "AT" messages.
I totally lack some documentation or more information to do put this BMS into the integration. We could start reverse engineering the system, but this will take a while and several (failing) iterations. Easier would be to have someone who can give information about the protocol required. Andy?

patman15, 2024-06-09:
@downset I added some more debugging to the implementation, could you please update your installation from the [branch](https://github.com/patman15/BMS_BLE-HA/tree/jikong_bms) and try again so I can get more info?
Also could you please add to your configuration.yaml the logging configuration:
```
logger:
  default: info
  logs:
    custom_components.bms_ble: debug
```

downset, 2024-06-09:
I`ll try tomorrow, I am not at home at the moment.
Thanks again for your effort 
I`ll attach the lock afterwords

patman15, 2024-06-10:
@downset thanks again for your patience. I think I found a (stupid) mistake. Can you please try again with the recent update on the [branch](https://github.com/patman15/BMS_BLE-HA/tree/jikong_bms) and send me the logs? If I'm really lucky then you should see at least some values. Thanks!

downset, 2024-06-10:
Here it is :)
but no data visible until now
[home-assistant_bms_ble_2024-06-10T19-30-40.230Z.log](https://github.com/user-attachments/files/15778238/home-assistant_bms_ble_2024-06-10T19-30-40.230Z.log)

patman15, 2024-06-11:
> but no data visible until now
>
I can see some, so the fix did bring us significantly forward. Do the values look plausible to you?
```
2024-06-10 21:30:35.915 DEBUG (MainThread) [custom_components.bms_ble] BMS data sample {'temperature': 18.4, 'voltage': 52.235, 'current': 0.0, 'battery_level': 42, 'cycle_charge': 117.212, 'cycles': 2, 'cycle_capacity': 6122.56882, 'power': 0.0, 'battery_charging': False, 'rssi': -83}
2024-06-10 21:30:35.915 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching BatteryOben-00 data in 0.917 seconds (success: True)
```
The protocol is still messed up so readout is not stable. I'll have detailed look soon, but at least values are arriving now. :smiley:

downset, 2024-06-11:
That sounds good.
Values are looking plausible out of current and power. Can`t beleve that at this moment i created the Log file the current was realy zero.

Greets
downset

patman15, 2024-06-12:
Hey @downset, I again did an update to fix the current and the general protocol handling. Could you give it another test, same procedure as before and provide me with logs? Thanks a lot!

downset, 2024-06-12:
and here it is:
[home-assistant_bms_ble_2024-06-12T10-01-36.116Z.log](https://github.com/user-attachments/files/15802932/home-assistant_bms_ble_2024-06-12T10-01-36.116Z.log)
Charging state looks correct
SOC is correct
Cycles is a bit low but the same as the BMS iphone app tells me :)
Stored Energie is correct
Load is correct
Voltage is correct
Current is correct
Temp is correct,
Runtime is completly jumping and I do not have any such Value also in the IPhone BMS APP

everything looks very nice... looking forwoard what you say about the Log
Would love to see also Cell Volt Diff and all Cell voltages  but this is an other request i Think :D

Many Thanks 
Downset

downset, 2024-06-12:
ok this was to early :)
while I am discharging the battery i see strange high numbers and all Positiv. No negativ values an it shows charging.
At the Time of the Screenshot i was discharging with around 5-10A for more than 5 Minutes - very cloudy here at the moment
[image]

downset, 2024-06-12:
and here a log in this situation at 13:09 MEZ i startet 3kw car charger and so i discharged both batterys with around 30a
[home-assistant_bms_ble_2024-06-12T11-10-00.822Z.log](https://github.com/user-attachments/files/15803721/home-assistant_bms_ble_2024-06-12T11-10-00.822Z.log)

patman15, 2024-06-12:
Hey! Sorry, had to fix the current again, now it should work. The runtime is the expected remaining time the battery has till 0% SoC. It's calculated so cannot be found in the app. If you can confirm the values now to be correct, I would be mainly interested in the stability, i.e. are the values always available. Would be cool if you can again send me a log, I'd like to do some more optimizations to reduce the BT load so likelihood of interference is lower. Thanks so much for taking the effort.

downset, 2024-06-12:
Found one thing,

at negativ values the remaining Time is unknown
[image]

patman15, 2024-06-13:
> at negative values the remaining Time is unknown

Signed values do not seem to be my strength. :laughing: Fixed.

I now cleaned the code and the tests and significantly reduced the BT traffic (and logging). Let me know if it still works as expected. If so, I would do further cleaning over the next days and then merge it to master for a release. Also if you could have a look at graphs to see if the values are continuously reported that would be great.

patman15, 2024-06-13:
Hey @downset! Ok, the behavior is really strange, there is no way I can replicate this on my side. I tested the old and modified version and all my tests (artificial, emulated battery) passed. Consequently, I decided to revert the BT optimization and just kept the fixes for the remaining time. Hope it works this time again and we can conclude here and move over to the cell voltage values as a new feature. :smile:

downset, 2024-06-13:
This is working again, but still no negativ Runtime value :)
[home-assistant_bms_ble_2024-06-13T20-21-27.668Z.log](https://github.com/user-attachments/files/15828331/home-assistant_bms_ble_2024-06-13T20-21-27.668Z.log)

It is up to you how to move foreword if you are willing to make more optimation tries of cause i will test it :) 
Thanks again for your Work
greets
downset

patman15, 2024-06-14:
> This is working again, but still no negativ Runtime value :)

Are you sure, you replaced all files, in this case especially `plugins/basebms.py`? My tests say it's correct, but who knows :dark_sunglasses: there is some evidence in the past I still might be wrong. :smile: 

> It is up to you how to move foreword if you are willing to make more optimization tries of cause i will test it :)

I'll go with first completing, because then there is always a stable version to go back to. Makes also the deltas (in code) smaller.

Greetings, patman15

downset, 2024-06-14:
Oh you are right I total missed the base bms... I have always only updated the new files cause it is easier with the homesisstant file editor

can this also play together with the not working optimations version?

patman15, 2024-06-15:
> can this also play together with the not working optimations version?
> should I try this again?

I don't think that it was related. For the optimizations I only changed the `jikong_bms.py` which I suppose you updated. I'll give it a try after a release is out to have a stable base. Thanks for your offer, I'll come back to it. :+1: 

> runtime is working while discharge now it is unknown with positiv numbers is this intended? or should it show the remaining charge time?

Runtime is intended to show the time till the battery is empty, thus it is unknown during charging. I don't think I should mess with charge time estimation, as this is a science of it's own due to changing parameters and it is heavily non-linear ...

patman15, 2024-06-15:
@downset thanks for your help in getting the BMS included. I just release [v1.3.0](https://github.com/patman15/BMS_BLE-HA/releases/tag/v1.3.0) which you can simply install via HACS. I'll come back on the BT optimization and the cell voltages after upgrading to the newer HA version. I'll keep the issue open till then. 

Let me know if you spot some issue in the meanwhile and don't forget to remove the `custom_components.bms_ble: debug` line from your configuration to avoid your log to get spammed with messages. In case something happens you can still enable it via the GUI.

downset, 2024-06-15:
I have to thank yo for your efforts :)
just updated to 1.3.0 and everything seams fine
Thanks for the hint to disable debug log again

if you have anything to test also unreleated to the jk just in general for the integration, feel free to contact me

patman15, 2024-06-23:
Hi @downset, I'm thinking again about a small BT optimization for your BMS. I have seen in the offgrid garage video that the BMS beeps on the first connection (when the integration starts). Can you confirm? I'd like to avoid that happening every 30 seconds ...

downset, 2024-06-23:
Yes this is correct it beeps every 30 sec :)
And after longer testing i can see from time to time that the Values are unknowen, but only a short time... not long enough to see it in the graf

patman15, 2024-06-23:
> Yes this is correct it beeps every 30 sec :)
> And after longer testing i can see from time to time that the Values are unknown
>
Do you prefer to 
- get rid of the beeps and fix the unknown values, or 
- keep the beeps and reduce the traffic via Bluetooth (saves a bit of energy and reduces disturbances with other BT devices).

Any preferences? Then I would give it a try again.

downset, 2024-06-23:
for me the beep is in the Basement so no problem
I absolutely would prefer less traffic on the Bluetooth proxy :)

patman15, 2024-06-23:
Hey! Can you give this [branch](https://github.com/patman15/BMS_BLE-HA/tree/feature/reduced-traffic-jk) a try? Note, it's based on 1.4.1, so either upgrade and replace one file, or replace all ...
It's sufficient if you enable logging via the GUI and leave it for a few minutes, send me the log and let me know how it behaves. Thanks!

downset, 2024-06-23:
just updated to 1.4.1 everything working than updated just the jkbms file correct?
seems only one of the tow batteries are there the one with address 00 
here is the log:
[home-assistant_bms_ble_2024-06-23T17-33-40.708Z.log](https://github.com/user-attachments/files/15945479/home-assistant_bms_ble_2024-06-23T17-33-40.708Z.log)
switching back to the 1.4.1 jkbms file also second batt there

downset, 2024-06-23:
did an other try to dubble check...
now both are not connecting
[home-assistant_bms_ble_2024-06-23T17-53-50.694Z.log](https://github.com/user-attachments/files/15945519/home-assistant_bms_ble_2024-06-23T17-53-50.694Z.log)
on 1.4.1 again both working

patman15, 2024-06-24:
> did an other try to dubble check... now both are not connecting

After updating the files, did you restart HA? Otherwise, reloading of the new functionality is not guaranteed.

downset, 2024-06-24:
yes i did both times

patman15, 2024-06-24:
This BMS behaves so strange. Since I cannot find a clear documentation, I will need to rely on the BT traces that are available, meaning that I cannot stop the BMS from permanently sending data (although not necessary). But the good news is, I can fix the missing values (I think). :smile:

patman15, 2024-06-24:
Could you try this [branch](https://github.com/patman15/BMS_BLE-HA/tree/feature/cnt-tx-jk)? It should now basically never show you "unknown" data and probably it removes the regular beeps (just a wild guess).

downset, 2024-06-24:
It is working and no more Beeping :)

[home-assistant_bms_ble_2024-06-24T17-07-46.873Z.log](https://github.com/user-attachments/files/15959115/home-assistant_bms_ble_2024-06-24T17-07-46.873Z.log)

I will monitor and update this post if I get "unknown" values again 

Thank you

downset, 2024-06-25:
it happens that it don`t work anymore :)
[image]
do you need a log?

patman15, 2024-06-25:
> do you need a log?

No thanks. I think we have reached quite the optimum of what is doable with the BMS. I'm currently working on cell voltage readout to give you something useful for all your efforts.

downset, 2024-06-25:
if its easy to do in this step all Temp sensors and the direct from the bms calculated cell differenze would also be nice :)

downset, 2024-06-25:
Looks Perfect
Not to much info for all time but if I like to inspect I can use the array
Better than I as a front end noob could thought of :)

patman15, 2024-06-25:
Not fully happy with my code yet, but I think it should be working for you and show you cell voltages using this [branch](https://github.com/patman15/BMS_BLE-HA/compare/main...feature/cell-voltages). I want to add the feature for all BMS types before releasing, so it might still take a while till it comes to a release.

downset, 2024-06-26:
It`s working fine

Thanky you very much

patman15, 2024-06-26:
Hi! Ok, looks like it works. 👍 The user presentation is a bit strange but defined by HA. If you take a look at the [developer tools](https://www.home-assistant.io/docs/tools/dev-tools/) the presentation is with 3 digits as the readout from the BMS is.
What I noticed is that the `delta voltage` was reported in one of your logs as 0.002 by the BMS while all voltages were actually the same. (By that fact you can see that it is a readout and not calculated from the cell voltages. 😄)

downset, 2024-06-26:
Again i would like to thank you for all your efforts to make the JK work.
I like it very much how it turned out
how can I give you a donation
Do you have an idea how you can give me a paypal link in Private? :)
not so firm with github can you send me a private message?
greets Downset

patman15, 2024-06-27:
@downset thanks a million for that nice motivating feedback. Highly appreciated! If you want to contact me personally, I have a [discord link ](https://discordapp.com/users/693860058479722517) in my profile and I now added a [buymeacoffee](https://buymeacoffee.com/patman15) button to the project. Actually, I was thinking that there are people who are in much more need of money than me so I would appreciate if you could donate the amount you were thinking of to an organization of your choice helping people in need. :blush:

patman15, 2024-06-30:
I now released v1.5.0 which should have all requested features. Thanks for your support, if there is anything more, let me know by contacting me or opening a new issue!

closed 2024-06-30 by patman15

downset, 2024-07-01:
Just updated to this Version, All working fine

Thank you