patman15/BMS_BLE-HA #23: JBD BMS

stobbie0, 2024-06-16:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

no

### Describe the solution you'd like

Support for JBD BMS

### Additional context

I think JBD are among the most used BMS, are there plans to integrate JBD?

stobbie0, 2024-06-16:
I'm using Batmon for my JBD BMS, but it would be great to be able to use a BT proxy

patman15, 2024-06-17:
Hi @stobbie0 shouldn't be a big issue as the JBD is rather simple to implement as I have seen. I can take it onto my agenda but need finish some other stuff first. Are you volunteering as beta tester? If you want to try it yourself, see the contribution guidelines and the other implementations for reference.
Best regards, Patrick

stobbie0, 2024-06-17:
Hi, yes I would volunteer as a beta tester. I'm not much of a programmer myself..

patman15, 2024-06-18:
Ok great, I'll have look and let you know.

patman15, 2024-06-22:
@stobbie0 I have added support for JBD BMS with the draft release [v1.5.0rc1](https://github.com/patman15/BMS_BLE-HA/releases/tag/v1.5.0rc1). Could you please test it and send me a log? Thanks!

stobbie0, 2024-06-24:
Thanks, that's quicker than I expected! Just went on holiday, but will test
after that.

Thanks!

Met vriendelijke groet,

Paul Stroink

One-Ten Land Rover Specialist
www.One-Ten.nl <http://www.one-ten.nl/>
www.Facebook.com/onetenlandrovers

LaroRent Land Rover verhuur
www.Larorent.nl

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
24-06-24,
07:28:34

patman15, 2024-06-30:
Please try v1.5.0 which I just released. Another user tried it with Supervolt batteries, which seam to use JBD BMS so I would assume the version should work for you as well. I'll keep the issue open till you can confirm. Enjoy your holidays!

patman15, 2024-07-16:
> Thanks for this excellent project! 👍

Thanks, you are welcome! :blush: 

> > individual cell voltage are available as attribute to this sensor
> 
> Would you mind sharing some details how to do this?
>
Sure, no problem, I added a new [FAQ entry](https://github.com/patman15/BMS_BLE-HA#how-do-i-get-the-cell-voltages-as-individual-sensor-for-tracking). Let me know if that helps or which information you are missing.

> Or might this only be possible in a future version? As you wrote under "Outlook":

Sorry, my mistake, forgot to remove this after implementing. Thanks for the hint, fixed.

> Why is signal strength deactivated by default? 

I saw that this is the case for most BLE devices in HA and in the end, a common user is not really interested in as long as the main functions work. So since it's diagnostic, I disabled it to save database storage, but *advanced* users can easily enable it. It's only once after installation.

> I assume it is the BLE signal strength which is helpful in case of connectivity issues. Is it?

For a rough guess I would say yes, also if you have something that disturbs the connection at certain times only. If it is severe, you will see outages in the sensor data.

Goaheadz, 2024-07-19:
I see that you already solved the template cell voltage extraction.
If you just want the list, without index, you can use "for" to get all cells voltages:

{% set VOLTAGES = state_attr('sensor.xxxxxxxxxxx_delta_voltage', 'cell_voltages') %}
{% for voltage in VOLTAGES %}
voltage {{ loop.index }}: {{ voltage }}
{% endfor %}

[image]

closed 2024-09-20 by github-actions[bot]