patman15/BMS_BLE-HA #34: Max number of devices

davebamford, 2024-07-13:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate feature request of [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I only seem to be able to access 3 devices with ESP Bluetooth Proxy. I have 8 BMS boards. Is there any way to access more than 3. If I use more than 1 ESP bluetooth proxy esp32 can I specify which devices they connect to?

### Describe the solution you'd like

Some way to specify which BLE mac ids you access.

davebamford, 2024-07-13:
I have managed to connect to more than 3 devices by using 2 ESP32 devices :-)

patman15, 2024-07-14:
Hi! Please have a look at the big warning on the [Bluetooth Proxy](https://esphome.io/components/bluetooth_proxy#bluetooth-proxy) page. A maximum of *3* active connections is supported, so this is a limit of the proxy, not the application. 
I have prepared code that automatically opens and closes connections when querying, I can enable that if you want, but I expect a significant lower reliability on readout then.
Is your solution as easy as adding another ESP32 device, or is there additional configuration necessary? If you could share, I could put it in the description to also help others. Thanks!

davebamford, 2024-07-14:
Adding another ESP32 device is all that is needed to connect to another 3 BMS BT devices. Because the first 3 are locked by the first device it just connects to the next 3, so no extra configuration is needed. I did see the big warning about max of 3 devices so I should have expected this behaviour. 
The nice thing is that once a connection is open it is locked. Thanks

patman15, 2024-07-14:
Thanks for the feedback, I updated the documentation!

closed 2024-07-14 by patman15