patman15/BMS_BLE-HA #38: Hi this looks perfect having BLE proxy support

timmills, 2024-07-23:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

Hi, I have auto installed using HACS (Do have other HACS add on running) I'v restarted HA. When I go to add the BLE Battery integration. It just pops up instantly with "No devices found on the network" There doesnt appear to be any way to configure a MAC address or let it know to look via BT Proxy. Thanks

timmills, 2024-07-23:
I did install and then uninstall the the original battery management add in if that makes any difference? Have approx 6 BT proxy in ESP being online

patman15, 2024-07-24:
Hi! There is nothing you need to configure if you have
- the device in range of any of the proxies
- at least one of the three possible connections per proxy available
- and a compatible BMS.
Yes, you most likely need to deactivate (or uninstall) the original add-in and be sure you are not connected via a phone app.
Which BMS are you trying to use?

timmills, 2024-07-24:
Setup a new proxy closer and bam - worked first time.Sorry my bad!Sent from my iPhone.

timmills, 2024-07-24:
> So everything now works as expected?
> Touched, not typed.

Absolutely - thanks! Adding some data now. Just needed to have a BT proxy closer to the battery Sent from my iPhone.

patman15, 2024-07-29:
Closed as done.

closed 2024-07-29 by patman15