patman15/BMS_BLE-HA #40: [runtime] in offgridtec battery no longer working

gkathan, 2024-08-22:
### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Bug%22+).

### Describe the issue

hi,
i just upgraded via HACS from v1.1.0 to 1.6.2 and now the entity "runtime" has no value anymore 
[image]

anything I can do ? or is there some issue on integration side ...
cheerz from vienna
 

### Reproduction steps

just updated  and before it worked 

### Debug logs

```text
13:17:01 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9$%"%%&%%\x18\x1f')>}, []]
13:17:02 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9$-SSSS\x18\x1f')>}, []]
13:17:02 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9'V'#%%\x18\x1f")>}, []]
13:17:32 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9%'#!\x18\x1f")>}, []]
13:17:32 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9%!! 'W%T\x18\x1f")>}, []]
13:17:32 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9%-S!&&\x18\x1f')>}, []]
13:17:32 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9%VTW%W\x18\x1f')>}, []]
13:17:32 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9$% ,%&%%\x18\x1f')>}, []]
13:17:32 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9$-SSSS\x18\x1f')>}, []]
13:17:33 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9'V'#%%\x18\x1f")>}, []]
13:18:03 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9%'#!\x18\x1f")>}, []]
13:18:03 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9%!! 'W%T\x18\x1f")>}, []]
13:18:03 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9%-S!&&\x18\x1f')>}, []]
13:18:03 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9%VTW%W\x18\x1f')>}, []]
13:18:03 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9$%'W%&%%\x18\x1f")>}, []]
13:18:04 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9$-SSSS\x18\x1f')>}, []]
13:18:04 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9'V'#%%\x18\x1f")>}, []]
13:18:34 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9%'#!\x18\x1f")>}, []]
13:18:34 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9%!! 'W%T\x18\x1f")>}, []]
13:18:34 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9%-S!&&\x18\x1f')>}, []]
13:18:34 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9%VTW%W\x18\x1f')>}, []]
13:18:34 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9$%!S%&%%\x18\x1f')>}, []]
13:18:34 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9$-SSSS\x18\x1f')>}, []]
13:18:35 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9'V'#%%\x18\x1f")>}, []]
13:18:52 DEBUG [client] Sending CONNECT (u0, p0, wr0, wq0, wf0, c1, k60) client_id=b''
13:18:52 DEBUG [client] Received CONNACK (0, 5)
13:19:05 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9%'#!\x18\x1f")>}, []]
13:19:05 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9%!! 'W%T\x18\x1f")>}, []]
13:19:05 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9%-S!&&\x18\x1f')>}, []]
13:19:05 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9%VTW%W\x18\x1f')>}, []]
[... cut ...]
13:23:45 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b'>GQ9$-SSSS\x18\x1f')>}, []]
13:23:45 DEBUG [client] received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0/dev_48_70_1E_77_3E_68/service0023/char002d): ['org.bluez.GattCharacteristic1', {'Value': <dbus_next.signature.Variant ('ay', b">GQ9'V'#%%\x18\x1f")>}, []]
```

patman15, 2024-08-22:
Hey!
Strange all tests still pass. Do you have a current that is negative?
Can you provide me please logs (no idea where you got the above one from) from here:
[image]
and can you check the status from the device itself?
[image]

gkathan, 2024-08-22:
> hi, ... ok that could answer the issue ... currently my solar panels on
> roof are producing ... no negative current ... that might explain it ...
> previous 1.1 version showed some value also with positive current ... will
> check when the sun goes down
> cheerz
> gerold kathan | +43 699 19743536 | www.kathan.at

OK ... this is new behaviour ... if there is negative current all is good
again 🤘 ... just when there is positive current no longer an estimate of
runtime is calculated ... is this intended ?
and other short question ... is it safe to upgrade to HACS version 2.0 ?
cheeerz from vienna
gerold kathan | +43 699 19743536 | www.kathan.at

patman15, 2024-08-22:
> OK ... this is new behaviour ... if there is negative current all is good again 🤘
>
That's the new behaviour, because the number during charging was useless, i.e. 0xFFFF minutes ~ 45 days. It just resulted in spikes that ruined the scaling for looking at the remaining discharging time.

> when there is positive current no longer an estimate of runtime is calculated ... is this intended ?

Yes, see https://github.com/patman15/BMS_BLE-HA/releases/tag/v1.6.2 or more detailed https://github.com/patman15/BMS_BLE-HA/pull/39 :-)

> and other short question ... is it safe to upgrade to HACS version 2.0 ?
>
Sorry, no idea yet, haven't upgraded so far. I hope yes ...

gkathan, 2024-08-26:
just to let you know - upgraded now to HACS 2.0 in my van setup => ALL GOOD, everything is WORKING

patman15, 2024-08-26:
Thanks for letting me know, I assumed that the HACS version is not relevant for the integration, but you never know ...

closed 2024-10-18 by patman15