patman15/BMS_BLE-HA #41: BT Pin code supported?

HansWegman, 2024-09-01:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

I just tried to install the integration to connect by XiaoXiangElectric Battery (JBD) which went like a charm, unfortunately although my battery was detected nicely through the ESPHome proxy and 11 entities were added, none have any data. I suspect this is because I have added a bluetooth pin to the battery which unfortunately cannot be removed. I have set it to "000000" now but still no data and the integration reports: "Failed to initialise BMS Hans Battery, continuing". Any idea if it is possible to enter the pin somewhere?

patman15, 2024-09-01:
Looks like the device does not respond, which is highly likely due to the set pin. Unfortunately, there is currently no function to set a pin for a battery in the integration. Would need to take a look on how to implement that. Would you volunteer as beta-tester (since I have no battery with pin available)?
Thanks for the log, simplified the response a lot!

patman15, 2024-09-01:
You could also try to pair the battery from the terminal using bluetoothctl see https://commandmasters.com/commands/bluetoothctl-linux/ chapter 4.

HansWegman, 2024-09-01:
Thanks for your quick reply! I will be very happy to do the beta test for sure! And I will try to pair from the terminal right away, will let you know asap.

HansWegman, 2024-09-01:
Hmm, tried to pair but maybe from the wrong terminal? I have a ssh and terminal add-on in HA which reports: 
unable to open mgmt_socket, device A4:C1:37:04:2D:BE not available.

HansWegman, 2024-09-01:
And maybe a silly suggestion since my programming skills are very low but maybe this could be solved (by someone with te right skills) with a small setting added to the ESPHome proxy based upon this: 

on_passkey_request
This automation is triggered when the BLE device requests a passkey for authentication.

ble_client:
  - mac_address: XX:XX:XX:XX:XX:XX
    id: ble_itag
    on_passkey_request:
      then:
        - ble_client.passkey_reply:
            id: ble_itag
            passkey: 123456

patman15, 2024-09-01:
> unable to open mgmt_socket, device A4:C1:37:04:2D:BE not available.
>
You see that as reply to which command?

When you enter the terminal you should see:
```
       ▄██▄           _   _                                    
     ▄██████▄        | | | | ___  _ __ ___   ___               
   ▄████▀▀████▄      | |_| |/ _ \| '_ ` _ \ / _ \              
 ▄█████    █████▄    |  _  | (_) | | | | | |  __/              
▄██████▄  ▄██████▄   |_| |_|\___/|_| |_| |_|\___|          _   
████████  ██▀  ▀██      / \   ___ ___(_)___| |_ __ _ _ __ | |_ 
███▀▀███  ██   ▄██     / _ \ / __/ __| / __| __/ _` | '_ \| __|
██    ██  ▀ ▄█████    / ___ \\__ \__ \ \__ \ || (_| | | | | |_ 
███▄▄ ▀█  ▄███████   /_/   \_\___/___/_|___/\__\__,_|_| |_|\__|
▀█████▄   ███████▀

Welcome to the Home Assistant command line.

System information
  IPv4 addresses for enp0s18: 192.---------
  IPv6 addresses for enp0s18: ---------------------------

  OS Version:               Home Assistant OS 13.1
  Home Assistant Core:      2024.8.3

  Home Assistant URL:       http://home.local:8123
  Observer URL:             http://home.local:4357
➜  ~ 
```
- Next, run `bluetoothctl devices`, you should get a list of available (and unavailable) devices and their MAC addresses.
- Next, try `bluetoothctl pair <MAC_of_your_BMS>`

HansWegman, 2024-09-01:
Yes, I tried it there indeed but my BMS device is not listed there... It is in the log of the proxy though, see the attached logfile.
[logs_m5stack-atom-lite-167c28_run.txt](https://github.com/user-attachments/files/16830241/logs_m5stack-atom-lite-167c28_run.txt)

HansWegman, 2024-09-02:
I changed the log settings in the proxy to VERY_VERBOSE and copied the parts with the Battery which seem to indicate it gets connected if I understand it well. Still do not understand why it does not appear in HA though. I attached a screenshot of the bluetoothctl devices output as well. Also strange, I reloaded your integration without the proxy and it failed on the entities. When I connected the proxy again it reloaded well. 
[image]
[Proxy_very_verbose Log battery part.txt](https://github.com/user-attachments/files/16837005/Proxy_very_verbose.Log.battery.part.txt)

patman15, 2024-09-02:
Which install type did you choose for Homeassistant? Are you using HASS or other?
--
Touched, not typed.

patman15, 2024-09-02:
Can you please check the output of `bluetoothctl list` to show the adapters?
I think it is an issue with the proxy. Pairing needs to be between the adapter and the device, IMHO. That means that since you are using a proxy the BMS needs to be paired with the proxy (which you can't do unless it shows up in `bluetoothctl`). Probably you can use the automation you have used so far?
Sorry, just guessing a bit, but maybe it helps to move forward. What is a bit irritating is the message `unable to open mgmt_socket` that should not happen. Can you bring the battery closer to the actual host so that we can check the pairing works there?

HansWegman, 2024-09-02:
Attached is the result of the list command. I guess however looking at the log that the MBS is paired with the proxy. I also sniffed the communication between the original app and the device and it looks like there is some special communications. I will include the log as well. 

What I did, I removed and reinstalled the app so it would ask for the password again.
I started the log
I opened the app and entered the password 123123
Than the readings came in, battery 99% charged
I switched to some other screens in the app
I closed the app
I reopened the app and this time I did not need to enter the password, I guess it saves it and uses the same or another way to connect
The data reappeared

Im am looking at it and trying to make some sense out of it: 
[XiaoXiang log 2.log](https://github.com/user-attachments/files/16841104/XiaoXiang.log.2.log)

[image]

patman15, 2024-09-03:
Which app did you reinstall that works and asks for the pin again? Just for reference so I can have a look.
--
Touched, not typed.

HansWegman, 2024-09-03:
The XiaoXiangElectric app on my iPhone, there is also an Android version, see https://xiaoxiangbms.com/download/.

patman15, 2024-09-04:
Hi Hans,

I have not too much hope, but you could try this code: https://github.com/patman15/BMS_BLE-HA/tree/feature/pairing
See https://github.com/patman15/BMS_BLE-HA/tree/feature/pairing?tab=readme-ov-file#manual on how to get the code manually to your system.
Please send me the debug log to see what happens.

krasnoukhov, 2024-11-23:
@HansWegman Did you have any luck with on_passkey_request? I'm trying to set up this integration with ESPHome BT proxy and JK BMS that apparently does require PIN, but no luck so far

HansWegman, 2024-11-28:
Hello Dmitri,

unfortunately not. I have found a possible solution though with https://github.com/syssi/esphome-jbd-bms. The author has figured out the encryption and is working on implementing that into the software so hopefully there is a solution soon.