patman15/BMS_BLE-HA #43: QUESTION: JK BMS - is the remaining capacity supported by this BMS-BLE HA integration ?

typxxi, 2024-09-06:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

I have checked the docs and I am still using BAT-MON, but wanna move ahead cause I have a JK V15 inverter BMS. 
The remaining capacity is shown in the JK BMS windows app but was not reported by BAT Mon therefore I am asking if that item  is part of your integration or not.

Due to the missing discussion page I had to put it here under "issues" even though it might not being one or maybe only one of the documentation or that I have overlooked it.

[image]

patman15, 2024-09-06:
You can get exactly this value by using a template sensor with a value template
`{{ float(states("sensor.smartbat_..._stored_energy"))/float(states("sensor.smartbat_..._voltage")) }}`

> Due to the missing discussion page I had to put it here under "issues" even though it might not being one or maybe only one of the documentation or that I have overlooked it.

No worries, available information is listed in the [README](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#provided-information).

typxxi, 2024-09-13:
thanks a lot.

Of cause kWh / voltage gives me Ah but I had assumed that it would be "available" as it is in the JK BMS out of the box.
But your solution is good and will do what I want / need.

I have opened a new question about a migration path from old bat mon to your bat mon - maybe you have a suggestion that can make it easier ? 

FINALLY: I have started reading the README part where i had missed the cellvoltages and what not till I found the solution here in an example which has a typo you can easily fix.

`"{{ iif(has_value("sensor.smartbat_..._delta_voltage"), state_attr("sensor.smartbat_..._delta_voltage", "cell_voltages")[0], None) }}"`

You can remove the second i in the beginning from **iif** cause I bet you meant **if**

closed 2024-09-13 by typxxi

patman15, 2024-09-13:
Home Assistant does not provide a native sensor with Ah as unit. Thus, I expose only the Wh one.

The double I in the iif is actually correct for templating.

Will have a look at your other question later.

typxxi, 2024-09-15:
``

> The double I in the iif is actually correct for templating.

Great, but I have never seen that before and therefore assumed a mistake. Lesson learned.