patman15/BMS_BLE-HA #44: Only 1 temperature sensor BMS Daly

VitalyIak, 2024-09-12:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Bms Daly type K has many temperature sensors, but I can see only one. Most importan itemperature sensors are  internal one and first external, but better to see all of them. 

### Describe the solution you'd like

Add sensors at integration for Daly

patman15, 2024-09-12:
Hi! I'll take a look if I can implement it easily as an attribute to the main sensor. Will let you know.

downset, 2024-09-14:
Would like to see this also for the JK BMS :)

patman15, 2024-09-20:
@VitalyIak, @downset can you test the [branch](https://github.com/patman15/BMS_BLE-HA/tree/44-only-1-temperature-sensor)?

VitalyIak, 2024-09-21:
> @VitalyIak, @downset can you test the [branch](https://github.com/patman15/BMS_BLE-HA/tree/44-only-1-temperature-sensor)?

Update on 1.6.3 - no changes concern BMS temperature... 
Or may be I have to test manually change files from this branch?

downset, 2024-09-21:
Looks fine for the JK Thank you @patman15 
do you need any logs to verify thinks?

downset, 2024-09-21:
@VitalyIak yes you have to manual exchange the files from the Branch it is not in the final HACS Release until now

downset, 2024-09-21:
@patman15 ok this was a little bit to early :) 
I get often just zero values for all parameters, not only the new Temp value, so it is switching back and forth from zero to real values. Temp on pack "oben" sometimes shows 7,1°C wile all other Parameters are zero
[home-assistant_bms_ble_2024-09-21T07-20-16.557Z.log](https://github.com/user-attachments/files/17083264/home-assistant_bms_ble_2024-09-21T07-20-16.557Z.log)

VitalyIak, 2024-09-21:
Oh, I can see 4 sensors of  temperature at attributes.  But only external sensors, but there is no internal bms temperature  sensor, unfortunately... is it possible to add? Thank you!

VitalyIak, 2024-09-21:
Anyway, now it's much better! Adding of external bms will be excellent!

patman15, 2024-09-21:
@VitalyIak 
> Oh, I can see 4 sensors of temperature at attributes. But only external sensors, but there is no internal bms temperature sensor, unfortunately... is it possible to add? Thank you!

I'm not aware that the daly BMS provides such information. If you have documentation about it, please let me know ...

> Anyway, now it's much better! Adding of external bms will be excellent!

What do you mean by _external bms_ ?

patman15, 2024-09-21:
@downset 
> I get often just zero values for all parameters, not only the new Temp value, 

Will check, probably I already messed up with v1.6.3, despite all tests passed. Thanks for the logs.

patman15, 2024-09-21:
@downset had a look at the logs, but they look very weird. I can see all messages from the BMS 6 times at the exact same time, e.g.
```
2024-09-21 09:18:30.638 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (BatteryOben-00) Rx BLE data (cnt.): bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd4\x00\x00\x00\x00\x00')
2024-09-21 09:18:30.638 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (BatteryOben-00) Rx BLE data (cnt.): bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd4\x00\x00\x00\x00\x00')
2024-09-21 09:18:30.638 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (BatteryOben-00) Rx BLE data (cnt.): bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd4\x00\x00\x00\x00\x00')
2024-09-21 09:18:30.638 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (BatteryOben-00) Rx BLE data (cnt.): bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd4\x00\x00\x00\x00\x00')
2024-09-21 09:18:30.638 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (BatteryOben-00) Rx BLE data (cnt.): bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd4\x00\x00\x00\x00\x00')
2024-09-21 09:18:30.638 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (BatteryOben-00) Rx BLE data (cnt.): bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xd4\x00\x00\x00\x00\x00')
```
Also I miss _connect_ messages of the integration after startup. It just looks like the system works without connecting. Did you restart HA? Or, do you probably have multiple ESPproxies that all catch the same, maybe some issue with them? It does not really look like an issue with the integration at first hand. But I'm totally confused at the moment, because the log looks like something that cannot happen.

downset, 2024-09-21:
of cause i restarted after updating the files, but the activation of the debug log was after the restart so maybe this is the reason there is no connecting. i have just one bluetooth proxy until the last version 1.6.2 everything was fine.
Sometimes there were some connection problems with missing values but never zero values

did am other fresh restart, also rebooted the bluetooth proxy:
Still zeros and one pack isn`t connecting at all
[home-assistant_bms_ble_2024-09-21T15-07-01.568Z.log](https://github.com/user-attachments/files/17084504/home-assistant_bms_ble_2024-09-21T15-07-01.568Z.log)

downgrade to 1.6.2. its working again will now test 1.6.3 main branche and will edit hier with feedback

patman15, 2024-09-21:
@downset you are a pretty good tester! I think I found the issue (#51). I updated the branch, hope it works now!

downset, 2024-09-21:
Perfekt seams to work again, no more zero values. Thank you!
I have now the main Temp Value and 3 additional Attributes.
in the App i have 4 Bat sensors and one MOS Sensor. 
I think thease 4 are the Battery sensors right?
Can i also get the MOS Sensor? :)
[image]

Greets
Downset

patman15, 2024-09-22:
@downset 

> Perfekt seams to work again, no more zero values. Thank you!
>
Thank you for the testing!

> in the App i have 4 Bat sensors and one MOS Sensor.
> I think thease 4 are the Battery sensors right?
>
Unfortunately, I don't know which sensor is which I just found 3 temperature values. Probably you can find out by comparing? Documentation and references are pretty weak on that topic.

> Can i also get the MOS Sensor? :)
>
I did a further update. I found 3 further temperature values, maybe you can tell what is what and check if it works?
Will need some cleanup for the release, but I think it works.

downset, 2024-09-22:
Thanks again!
It is working

Now i have the main Value and 5 extra Attributes, two of them seams to be the same, but maybe It is the real data at the moment or is it a dublication?

[image]

VitalyIak, 2024-09-23:
> I'm not aware that the daly BMS provides such information. If you have documentation about it, please let me know ...

it's seems that at UART/RS-485 BMS send this info at message_id=92, where status of BMS:
[Part 4_ Daly RS485+UART Protocol.pdf](https://github.com/user-attachments/files/17095489/Part.4_.Daly.RS485%2BUART.Protocol.pdf)

I'm ready to test...

VitalyIak, 2024-09-23:
@patman15 Also please check https://github.com/dreadnought/python-daly-bms I suppose there is also about Daly BMS, may be there will be usefull smth...

VitalyIak, 2024-09-23:
from test_daly_bms.py:
```
    def _response(
        self, char_specifier: BleakGATTCharacteristic | int | str | UUID, data: Buffer
    ) -> bytearray:
        if char_specifier == normalize_uuid_str("fff2") and data == (
            self.HEAD_READ + self.CMD_INFO
        ):
            return bytearray(
                b"\xd2\x03|\x10\x1f\x10)\x103\x10=\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00<\x00=\x00>\x00?\x00\x00\x00\x00\x00\x00\x00\x00\x00\x8cuN\x03\x84\x10=\x10\x1f\x00\x00\x00\x00\x00\x00\r\x80\x00\x04\x00\x04\x009\x00\x01\x00\x00\x00\x01\x10.\x01\x41\x00*\x00\x00\x00\x00\x00\x00\x00\x00\xa0\xdf"
            ) # {'voltage': 14.0, 'current': 3.0, 'battery_level': 90.0, 'cycles': 57, 'cycle_charge': 345.6, 'numTemp': 4, 'temperature': 21.5, 'cycle_capacity': 4838.400000000001, 'power': 42.0, 'battery_charging': True, 'runtime': none!, 'delta_voltage': 0.321}
```
there is a temperature, i suppose this is temp of BMS, because external NTC are at different place. Please check...

VitalyIak, 2024-09-23:
> What do you mean by _external bms_ ?

@patman15  Forget please about it,  ))

patman15, 2024-09-23:
> from test_daly_bms.py:
> ...
> there is a temperature, i suppose this is temp of BMS, because external NTC are at different place. Please check...
>
No that is the average temperature calculated from the 4 that are now visible as attributes.

> Also please check https://github.com/dreadnought/python-daly-bms I suppose there is also about Daly BMS, may be there will be usefull smth...
>
That could be interesting but is totally different to the current approach, so might take a while.

@VitalyIak can you please send me a debug log of the integration and a screenshot from your app? Does the app show the MOS temperature? If yes, I can probably infer from the debug log. Thanks!

VitalyIak, 2024-09-23:
@patman15 
[image]
[home-assistant_bms_ble_2024-09-23T14-31-07.391Z.log](https://github.com/user-attachments/files/17099174/home-assistant_bms_ble_2024-09-23T14-31-07.391Z.log)

patman15, 2024-09-23:
@VitalyIak I had a look at the resources you provided, but they are all for the UART protocol not Bluetooth, so they unfortunately do not help. The temperature you are looking for is not present in the record that I can query and also the app does not show it from the record. I know the app requests more info from the BMS, but I can't find documentation on how to do so.
The only way would be to record the data transfer between the app and the BMS, e.g. on Android using the Bluetooth debug setting. If you know how to do that, please provide me such a file.
I will mark this issue as solved for the moment for a new release, but feel free to open a new feature request with the specific value for Daly BMS.
Sorry that I have no better news.

closed 2024-09-23 by patman15