patman15/BMS_BLE-HA #45: any suggestions for a migration from the previous fl4p batmon to patmans ? Or can they run in parallel ?

typxxi, 2024-09-13:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

I found this Batmon here as recommendation by someone else who had asked if fl4p's [batmon-ha](https://github.com/fl4p/batmon-ha) is still maintained or not.

1.
My biggest question is if I can try this one in parallel with fl4ps (which I doubt cause I can not even access the BMS from my phone anymore due to a setting).

2.
And I want to know if there is a smooth way for migration from my current 
fl4p's [batmon-ha](https://github.com/fl4p/batmon-ha) to this one.

I am using JK inverter BMS V15 with firmware V15.30

3.
My biggest complain about fl4p's solution is the missing "remaining capacity" sensor - one of the most important features in my opinion. 

I want to avoid to make mistakes others had already made therefore I am asking.

Thanks a lot

patman15, 2024-09-14:
ad 1) you can _install_ it in parallel, but I'm quite sure (cannot test) you cannot _run_ them in parallel. In short, you can install both and deactivate always one of the two.

ad 2) depends on what you understand by _smooth migration_. Hard to answer, as I don't know your setup.

ad 3) ok, great that is solved!

Let me know if you need more information!

typxxi, 2024-09-15:
thanks for replying soon. 

1 - good to know to have both installed and being able to switch between them in a kind of transition phase
2 - hard to say but at the end everbody has already his sensors in batmon and then it would be nice to change the source from fl4ps batmon to you patman15 batmon . Setup is JK inverter BMS V15 running firmware 15.30 under current HA OS

the key question is a new one I had not considered. It is caused by someone who had linked your BMS BLE Solution on the batmon side as a kind of "the successor solution" and simply fooled me cause I had thought it would be a 1:1 solution but never thought about missing sensors of a JK BMS like the active balancer current. 

I have read the feature request which made me wonder how much more could be missing and how someone could be so stupid to promote your batmon without telling that things are not there and might even never come which I  might be used to from the current solution.

Of cause not your fault , but that fault of the guy who had given the recommandation without having any clue or a limitation like "just for whatever BMS owner only". He played a blank card and has lost. 

I will try your solution sooner or later but the story has changed a lot by that guy who had fooled me and gave an expression like this bat mon would be a successor of the previous one.

patman15, 2024-09-16:
> 2 - hard to say but at the end everbody has already his sensors in batmon and then it would be nice to change the source from fl4ps batmon to you patman15 batmon .

That was never the intention of this integration and a drop in replacement would require a migration code to adopt the sensors created by batmon via MQTT. This is for sure quite some work and not sure it would work flawless.

> but never thought about missing sensors of a JK BMS like the active balancer current. I have read the feature request which made me wonder how much more could be missing ...

The sensors/information supported is listed in the [README](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#provided-information).
Nevertheless, I did not intend to build a replacement but a native integration. There are good reasons to stick with batmon, e.g. MQTT, stand-alone mode, or the active control. Also I [tried to align](https://github.com/fl4p/batmon-ha/discussions/213) with Fabian on the protocol implementation to join forces, but sadly did not receive a response since quite some time.

Long story short, if it's useful to you, feel free to try it out, if you have questions let me know!

typxxi, 2024-09-16:
yes, it is usefull and what you said is right and new to me cause one guy over there had recommanded your solution in a way that lead me into the wrong direction cause  I had assumed that this is "successor" like . 

And hence my questions cause someone without much clue had invited me that this would be the solution for JK BMS owners, but not telling the missing details. 

I know that you are working on things like showing the active balancer sensor datas and I have seen other feature requests that are pointing in the same direction therefore I am quite happy. 

And I guess that you now can understand better why I had come with these questions . 

Fabian sounds like a german guy, so I might try to write him in german and get behind what is going on cause many things are going on while his bat mon is slowly falling behind - or what he thinks about a new appoach as you said with joined forces.

Thansk a lot !

closed 2024-11-17 by github-actions[bot]