patman15/BMS_BLE-HA #46: Seplos v3 not recognize

Alexandros89, 2024-09-13:
Hi
I have 2 batteries with the seplos v3 BMS. From my phone I can connect through Bluetooth without any problem. Using the esphome proxy, ha doesn't recognize anything. The esphome proxy works because it see my beacon's phone

patman15, 2024-09-14:
Your description does not really help to exclude potential issues. Can you please provide a debug log of the esphome proxy as well as one of the integration in order to be able to narrow it down?

- Did you ensure that you have disconnected the phone app before trying?
- Did you make sure not more than 2 devices are connected to the esphome proxy?
- Does it work without the proxy via a direct interface?
- Are you sure the esphome proxy is in range of the battery?

patman15, 2024-09-19:
Ok, now I think I know the issue.
Can you please change the following line
https://github.com/patman15/BMS_BLE-HA/blob/8e9c1b7c934d6837fb3f93f2032b290c7677ed2d/custom_components/bms_ble/plugins/seplos_bms.py#L108 
and this one:
https://github.com/patman15/BMS_BLE-HA/blob/8e9c1b7c934d6837fb3f93f2032b290c7677ed2d/custom_components/bms_ble/manifest.json#L25
both to read `SP1*` and see if that helps?
I need to look into how to autodetect Seplos batteries better as the name seems not to follow a pattern.

Alexandros89, 2024-09-19:
Before you answer me I change only the manifest and it doesn't recognize it. After that I change the script now it appears.

patman15, 2024-09-19:
Does it fully work as expected?

Alexandros89, 2024-09-20:
now it works, i see the basic information that is better than nothing

patman15, 2024-09-20:
Ok, thanks for the feedback, I'll patch the detection for the next release.
Compared to the screenshots I see the same information, so not sure about your expectations.
Thanks again for the confirmation that it works!

patman15, 2024-09-20:
@Alexandros89 sorry, my commit comment automatically closed the issue. If you still need something, feel free to reopen.

closed 2024-09-20 by patman15

Alexandros89, 2024-09-20:
These are the other information that I can see from the app

[image]

[image]

commit 75faccfb (2024-09-20): fixes #46 Seplos BMS detection