patman15/BMS_BLE-HA #47: Pace BMS support?

JiMcNulty, 2024-09-14:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

No support for Pace BMS server rack batteries 

### Describe the solution you'd like

Pace BMS is one of the most used bms's on server rack batteries, it support BLE (and WIFI)
It will be great to have support for this BMS

patman15, 2024-09-14:
I can have a look, but first I would need some documentation about the protocol that is used. Can you provide links or documents that describe the BT communication?

JiMcNulty, 2024-09-14:
Sorry, didn't find docs like that
But there is an [android app](https://play.google.com/store/apps/details?id=com.paicheng.bms&hl=en)
So I guess maybe decompiling it would help?

patman15, 2024-09-16:
Ok, an Android app helps to reverse engineer the protocol. Can you provide information about the characteristics using e.g. [nRF connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&pcampaignid=web_share) and a recording of the communication between the app and the BMS during the first ~30s?
I don't want to interfere with legal issues, so no decompiling.

Zodiac-69, 2025-08-16:
@patman15 
Here are the PACE BMS RS 232 documentation
I have tested this on my Sunsynk 10.56kW (BMS Version: P16S200A-A21084-1.05)  pack with RS232 connection.

[PACE-BMS-RS232-communication-protocol-20180705.pdf](https://github.com/user-attachments/files/21816419/PACE-BMS-RS232-communication-protocol-20180705.pdf)

Zodiac-69, 2025-08-19:
@patman15 

My current setup does not have Bluetooth connectivity.
My plan was to use an ESP32 (ESPHome) and create a serial to BT bridge.

What i can supply you is:
- Raw Data
- Constants
- Decoded data

patman15, 2025-08-19:
Can't you just use the [ESPHome Bluetooth proxy](https://esphome.io/components/bluetooth_proxy) to add BT functionality or a [good and cheap USB BT dongle](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters)?