patman15/BMS_BLE-HA #48: Balance current for JK

downset, 2024-09-14:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Would like to see the Balance current of the BMS

### Describe the solution you'd like

Would like to see the Balance current of the BMS to verify that the Balancer is working.
And all temp sensors but therefore I have seen is also a request for the Daly so... duplicate :)

typxxi, 2024-09-15:
I can only agree cause I have assumed that this one would incl. all the sensors we already had in fl4ps BatMon.
Someone has recommanded this solution as the proper successor for a JK BMS but I was not aware of such missing sensors cause I am still on the verge to migrate now, but still have not done it yet.

patman15, 2024-09-25:
Will have a look in a few weeks on how to integrate the balance current in a smooth way.

> And all temp sensors but therefore I have seen is also a request for the Daly so... duplicate :)

That is solved by release [v1.6.4](https://github.com/patman15/BMS_BLE-HA/releases/tag/v1.6.4).

patman15, 2024-11-20:
@downset, @typxxi  I thought about the balance current. I do not want to do another sensor for it, as this is a specific item for JK. Would it be sufficient for you, if I add the value as attribute to the current sensor?

downset, 2024-11-20:
Absolutely fine for me.
I understand the idea to keep the sensors as generic as possible.
With a helper Template I can make a Sensor for my needs.
good solution in my opinion.

Greets downset

typxxi, 2024-11-20:
> @downset, @typxxi  I thought about the balance current. I do not want to do another sensor for it, as this is a specific item for JK. Would it be sufficient for you, if I add the value as attribute to the current sensor?
> 

sure,
i'm not the expert , but i guess i know what attribute means.
You are using an existing sensor and add that  attribute. 

BUT: im quite sure that my topic was a different one.

sorry, I'm on my phone and wanted to give you a quick reply.

What I was missing was the Ah meter cause that is important for us .

Balancing was not a major topic, but it is quite a while.
I can look it up later, cause I instantly started typing and have only a phone with me.

But I know that you are the bt bms master for  home assistant. 

But i can't remember that balance current was a topic of mine cause I rarely check that. 

hope to check it later when i am on the laptop again

thanks

typxxi, 2024-11-20:
> @downset, @typxxi  I thought about the balance current. I do not want to do another sensor for it, as this is a specific item for JK. Would it be sufficient for you, if I add the value as attribute to the current sensor?
> 
ok,

my core topic is Ah missing.

Back then I wrote that out of astonishing cause someone had told us on the bat mon and give the impressikn that your solution would be a kind of successor or replacement of thst.

And therefore I had written.

I can live with a sensor and an  attribute of the balancing current.

Priorities have shifted cause the balancing is now working flawlessly after a jk firmware upgrade.

sorry for the confusion, cause when I pressed save I could see the topic + my comment and could remember what had happened back then which was not your fault.

thanks a lot

patman15, 2024-11-21:
@typxxi 

> my core topic is Ah missing.

This can be solved! Just go to [Settings > Devices & Services > Helper](https://my.home-assistant.io/redirect/helpers) and [add a template sensor](https://my.home-assistant.io/redirect/config_flow_start?domain=template) with the following state template and there your are.

`{{ iif(has_value("sensor.smartbat_..._voltage"), float(states("sensor.smartbat_..._cycle_capacity"))/float(states("sensor.smartbat_..._voltage")), None) }}`

> I can live with a sensor and an attribute of the balancing current.

Ok, then I will go forward with this approach soon and hope that it solves your issues!

> But I know that you are the bt bms master for home assistant.

I'm not, I'm just a random guy who made this integration for his own purpose and offers it to others so that the amount of time I invest is arguable for me. For me alone it consumes far too much of my spare time. So I really hope the community benefits from it.

patman15, 2024-11-22:
@downset, @typxxi just pushed a commit to the branch, can you verify the balance current works for you?

downset, 2024-11-23:
for me there is the problem, that the weather is so bad that there is actual no chance to get anything in the Battery, and so no chance to get to an SOC where balancing kicks in.

I installed the new Branch and I see the Attribute, but it will take some time until I will see a value different from 0 

Later, i will try to lower the balance start voltage in the Settings of the BMS for a moment to check if I get the balancing values correct

downset, 2024-11-23:
Not so easy to test at the moment, also with lowering the balance start Voltage there is no balance current with a pack with only 0,002V deviation at the actual SOC

But i can partly confirm that it is working, for a short moment the balancer kicked in an I have seen a positiv balance value, but it was so short that I can not confirm if also the negativ currents show up correct cause at the next 30 sec update the balancing was already done down to 1mV :)

Have you seen my edited Post above?

I it easy to also get the attribut witch cell is balancing?

Thanks a lot for your Work again

patman15, 2024-11-23:
> Is it easy to also get the attribute witch cell is balancing?

Sorry, missed that. Had a look, but couldn't figure out, where to get that information, so it's not easy, sorry.

downset, 2024-12-26:
Still no day with full batteries in Winter to fully test.
would suggest you merge it as it is so i can test with actual version and come back if there are problems later.
Maybe the next days i would get sun to charge an see it working for longer than a few seconds last time i forced Balance in the bms
Regarding witch is the active cell witch is balancing, i think it is not a separate Value, in the original app i also can't see it als a number, it is the color of the Voltage sting of the highest and lowest cell.
Maybe there is an attribut for color or it is done by the App itselfe as it colores the max/min cell

closed 2024-12-26 by patman15