patman15/BMS_BLE-HA #50: Energy Dashboard

ThomasCr, 2024-09-19:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

Hi,
can anybody hive a hind, how to integrate the battery into the Energy Dashboard?

https://www.home-assistant.io/docs/energy/battery/

I mean, do I really need a extra enegery sensor to meshure In and Out values?
Or can anybody give a hint how to create a template?

Regards,
Thomas

patman15, 2024-09-20:
You can use for power out a template sensor:
`{{ [states("sensor.smartbat_..._power") | float(0) *-1, 0] | max}}`
and for power in:
`{{ [states("sensor.smartbat_..._power") | float(0), 0] | max}}`
both with
`Unit of measurement 	W`
`Device class 			power`
`State class 			measurement`
Then create via the `helpers` an `integral sensor` for each of the above `template sensors`.
You can then add the `template sensors` as in/out to the energy dashboard in the category `battery`.

ThomasCr, 2024-09-21:
ok, Step 1 looks like this:
```
  - sensor:

    - name: "DWCE00531J-029 Leistung Rein"
      unique_id: "dwce00531j_029_leistung_in"
      state: '{{ [states("sensor.dwce00531j_029_leistung") | float(0) *-1, 0] | max}}'
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"

    - name: "DWCE00531J-029 Leistung Raus"
      unique_id: "dwce00531j_029_leistung_out"
      state: ' {{ [states("sensor.dwce00531j_029_leistung") | float(0), 0] | max}}'
      unit_of_measurement: "W"
      device_class: "power"
      state_class: "measurement"
```

But how I should do the `helper` with the `integral sensor`?

patman15, 2024-09-21:
Ok, schaut gut aus. :+1: 

You could have done it [using the GUI](https://my.home-assistant.io/redirect/config_flow_start?domain=template), but YAML works as well.
Next is to generate one [integration sensor](https://my.home-assistant.io/redirect/config_flow_start?domain=integration) for each, or via [YAML](https://www.home-assistant.io/integrations/integration). This will then result in (k)Wh so you can add to the energy dashboard.

closed 2024-11-21 by github-actions[bot]