patman15/BMS_BLE-HA #52: Adding Fliteboard Efoil battery

alros100, 2024-09-22:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Hello
I tried the integration with my flight board efoil battery bms. But the integration reports no compatible device found.

would you please consider adding this battery ? I realize you have left instructions to add different bms. But I am not up to the task.

this is what I know about this battery from a Reddit post

thanks so much for considering
Al

The Flitecells are designed by Battery Pack Brand Official Store | Em3ev. The BMS is a DXB46A that is custom-made for these packs by Tianjin D-powercore Co., Ltd their software is kinda junk and they do not publically release firmware. It has two temperature probes that are glued to cells 25% and 75% up one side of the pack.

### Describe the solution you'd like

Adding support for the bms DXB46A please

patman15, 2024-09-23:
Hi! In order to help I need more technical information about Bluetooth and the used protocol. Please have a look at this [comment](https://github.com/patman15/BMS_BLE-HA/issues/47#issuecomment-2353274285) about what I would need.
The reason is, I do not own any (except one) of the supported batteries so I cannot figure it out on my own.

alros100, 2024-09-23:
Hi

I was chatting with an iOS app developer who added the batter for me. I will reach out to him and see if he can help with the protocol and get back to you

Thanks so much for your help
Al

alros100, 2024-09-23:
I do have a log of the connection failed and then connecting to the iOS app Strom. Is there a way I can send you the log, rather than post a long log here ?

Al

alros100, 2024-09-23:
Here is the log that connects. I am waiting to here back from the IOS developer as well
[log.zip](https://github.com/user-attachments/files/17100188/log.zip)

patman15, 2024-09-23:
Hmm, I have an idea how to do this, need to check, cannot promise much, especially not too soon. Will keep you posted.

alros100, 2024-09-23:
The developer of IOS app is willing to help. I gave him link to this issue. Thanks so much !

patman15, 2024-09-23:
Ok, I tried to emulate the BMS, but I got stuck. The app from Tianjin D-powercore does use different commands than your Fliteboard (I guess its a custom made version) and the app from Fliteboard does not work without account. I think I know the BT characteristics that are relevant, but ideally I'm now looking for an emulation or documentation of the commands and responses.

patman15, 2024-09-24:
> Is there a way I could send that to you privately in a private message?

You could DM [me](https://community.home-assistant.io/u/patman/) in the HA community or on [Discord](https://discordapp.com/users/693860058479722517), would that work?

alros100, 2024-09-24:
Thanks, I PM'd you on home assistant committee with the contact 

Thanks again
Al

patman15, 2024-09-25:
@alros100 can you try the code on the [branch](https://github.com/patman15/BMS_BLE-HA/tree/52-fliteboard-efoil-battery)? Note it is very alpha, only roughly tested and might contain some major flaws. I need to take a break from coding the next weeks, so not sure how far I can get still.

alros100, 2024-09-25:
Thank you so much. I will try it and post back. Please take all the time off you need - no pressure at all . I am very grateful that you are assisting!!

Al

alros100, 2024-09-26:
Hi

Thanks so much for your assistance on this.

I uninstalled the original integration from hacs and I installed the branch manually as per the instruction and I confirmed that the updated files are present.

I have both the raspberry pi Bluetooth enabled as well as running the Bluetooth proxy on esp32. And I also tried with each separately 

I can find and add the integration, but it instantly indicated that no compatible batteries are found. I tried searching for logs but I could not find anything. I also ran the integration with the Bluetooth proxy log running and did not see any attempts at a search or any changes. 

Does the integration log to a specific area? I just want to confirm that it's trying to connect as the message of no compatible batteries comes up so instantly that I wonder if there is something wrong with with the installation rather than the new code ?

Al

alros100, 2024-09-26:
Will Do ! 
Thank you
Al

alros100, 2024-09-28:
Hi
 Please see attached zip of what you requested. I am really confused. in terminal HA built in bluetooth seems to see the battery only randomly and bluetooth proxy ESP32 doesn't seem to see it
[HA Log.zip](https://github.com/user-attachments/files/17174959/HA.Log.zip)

not sure if its my home assistant Bluetooth that is the issue ?

Al

patman15, 2024-09-29:
The log shows that the device characteristics are not resolved via Bluetooth, so the integration does not detect it as a valid device.
My guess would be that you run into the 3 device limit of the BT Proxy, as I can see 2 Hue Lamps and a Bose speaker being resolved.
It's definitely a BT issue, either add more BT proxies or due a static assignment of the BMS to one proxy so it is guaranteed to be connected. That would be by hint, hoping it is correct.

alros100, 2024-09-29:
Hi

Thanks for you assistance and patience. Although I have bluetooth proxy running, I am only using the raspberry pi 4 built in bluetooth module. Also yes in the proxy I did have active enabled.

so I installed a new clean homeassistant on a spare raspberry pi 3 I had have. The only thing I put on it is  bluetooth and the custom_component / bms_ble.  The battery is sitting right next to the raspberry pi 3.

Now the bluetooth integration debug log and the  terminal command bluetoothctl info B0:D2:78:19:59:36 consistently shows the battery. Not randomly like on my main homeassisstant.

the first time I tried to add the BMS ble integration there was a quick error message that I wish I had my phone to photo. it was something like a component caused an error (I looked in your strings file and it was not one of those messages). The second time I tried the integration there was no error message but it once again reported "no supported devices found via bluetooth"

logs attached, 
[Logs.zip](https://github.com/user-attachments/files/17181074/Logs.zip)

thanks for your patience - no rush on this
Al

patman15, 2024-09-30:
The log does not mention the battery. There is still an issue with the BT setup.
- did you try to pair the device? Use `bluetoothctl pair <mac_address>` from the terminal.
- are you sure nothing else is connected to the BMS?

I'm still wondering via the device does not advertise any services, which is required for detection and operation.

alros100, 2024-10-01:
Hi
 yes the main log doesn't show the battery, but the bluetooth integration download diagnostics - shows it . Yes I have the fliteboard app and the ios app from adreas that are able to connect to the battery and I am absolutely sure that both are not running on my phone and nothing is connected to the battery during these tests.

so I rebuild a brand new Home assistant on the pi 3. once again the info command shows the device, but the pair command attempts but doesnt appear to work see picture.

WI keep seeing that message n terminal "unable to open mgmt_socket" both on the old and new home assistant. Is this what is telling you that my bluetooth is not configured properly ? I wouold think the a brand new fresh install of HA should solve the issue unless the problem is the bluetooth chip on the Pi 3 and 4 . 

since I have no other compatible connectable bluetooth devices compatible with home assistant. I ordered a new Pi 5 and a bluetooth thermometer on the HA compatible list to see if I can show that this connects and works.  not sure how else to trouble shoot  the bluetooth integration.

thanks so much for your patience - I'll report back in the next few days when I receive the new stuff.
 
[image]

Al

alros100, 2024-10-02:
Hi Patman
so I recieved the pi5 and set up a clean install of just bluetooth alone then added bluetooth proxy. in both cases bluetooth picks up and adds a known HA supported Govee bluetooth thermometer. also this thermometer is picked up by my full home assistant on a Pi4 as well as the Pi 3 clean install. So I am assuming bluetooth is working on these three devices. but I have had no luck with these batteries.  bluetoothctl info with mac works as above but pair does not.

I continue to be able to connect and retrieve all the cell data using Djorn-Andreas' strom app and the fliteboard app . so is it that home assistant bluetooth does something upfront before the Patmon integration has a chance to find/connect to the battery ?

thanks for your patience
Al

patman15, 2024-10-02:
The message with the management socket doesn't tell me anything. I tried already previously to figure out under which conditions it occurs but did not succeed.
Honestly, I'm running out of ideas why your device does not show up. Maybe you need to set the flite board into some pairing mode? Can you unpair it from your phone and try again? Just a random idea ...
Did app just work out of the box? (No pairing needed) If yes to the latter my best guess is still an issue with the Bluetooth setup.
The device must have services advertised in order to be able to detect and control it. For some strange reason it does not do so. 😞

alros100, 2024-10-02:
I understand - this is super frustrating. No there is no pairing with either of the phone apps. But both apps immediately report the TBA battery. Before Andreas updated his app, the app immediately reported the battery but would not connect. After his update it connected and reported all the parameters.

I ran a verbose log of the ESPHome Bluetooth proxy and searched the log for the battery MAC address, but could not find it. The built in Pi Bluetooth at least seems to detect the battery name. 

By the way, I had originally tried the Batmon bms integration. That one "sees" the battery but fails to connect using the included protocols as they don't have a em3ev BMS protocol. I reached out to you as that project seems to be inactive.

Hey I appreciate all the time you have spent and I understand if you can't devote anymore time to this.

My motivation for this was to simply create a home assistant charge to 60% automation for battery storage. The flite app and their incredibly overpriced charger has a fixed nearly full charge algorithm only. It was getting to be a pain timing the charge or watching it reach 60% !

Greatly appreciate your efforts 

Al

patman15, 2024-10-02:
I won't give up until I totally run out of ideas, or have figured out the problem. So no worries, the initial implementation took most of the time, so I want to make use of it. 😄 
Batmon uses the same Bluetooth methods as my integration, but it assumes after the configuration that a compatible device is reachable via the MAC address you specify. I assume it fails connecting because the characteristics are not resolved. That would be the same reason why my integration doesn't even show it in the first place ...
Do you have logs from Batmon you could share to validate that hypothesis?
Do you have an Android phone where you could run `nRF connect` to check the Bluetooth properties and send a log /screenshot?
I would need to play around a bit with the `bluetoothctl` and similar commands to get more information about the device, but currently I have no infrastructure at hand. Will take a few weeks till I can go deeper.
Best regards, Patrick

alros100, 2024-10-02:
Thanks so much Patrick ! i am responding now but please take the time you need !

nrf connect has ios version. It connects to the battery and I have the logs, I also included the logs before and after Bjorn-Andreas fix the strom app. He also seemed very open to reconnecting if needed.

I will re-install Batmon and get some logs.

Also if do you suggest any relatively cheap supported BMS that is known to work with the integration I could purchase to test ? or is the raspberry pi platform a potential problem for bluetooth ? should i try another home assistant platform

thanks !
Al

[Logs.zip](https://github.com/user-attachments/files/17237100/Logs.zip)

alros100, 2024-10-02:
here are the Batmon Logs when trying battery as JK or JBD
[JK Batmon Log.txt](https://github.com/user-attachments/files/17237442/JK.Batmon.Log.txt)
[JBD Batmon log.txt](https://github.com/user-attachments/files/17237443/JBD.Batmon.log.txt)

alros100, 2024-10-02:
OMG ! so after uninstalling batmon, I just wanted to make sure everything was back to normal so I tested the terminal bluetooth commands and it all of sudden showed the TBA battery and attributes. So manually re-copied patman  to the custom component and restarted  it found the battery and connected and let me add the details into an entity card.

I have no idea way it worked now but you must be right it is some bluetooth issue that seems to have self corrected ??

amazing!
Al

alros100, 2024-10-03:
update. The connection to battery 1 is robust. I can reboot home assistant and it reconnects. I updated the core and it reconnected. The battery bms is activated with a magnet. I deactivated and activated the battery multiple times and it immediately connects.
now the weird part. Battery two is behaving like before. I can't add it and it won't connect. in terminal it shows as before without showing the TBA name or the extra attributes as from the screen grab above. so strange that one battery has a completely robust connection while the other one is behaving like the currently connected battery before it connected.

maybe this behaviour can point to the challenges with initial connection. Is it something with my bluetooth ?
Al

patman15, 2024-10-03:
> I have no idea way it worked now but you must be right it is some bluetooth issue that seems to have self corrected ??

- did you still have Batmon running? If yes -> that was the issue
- Which [method](https://www.home-assistant.io/installation/#advanced-installation-methods) did you use to install Home Assistant? If you used core and copied Batmon in parallel then there is definitely a conflict. Especially if you chose to update the Bleak library (Batmon does use sometimes a special version) which is the core Bluetooth library. That could cause incompatibilities to Home Assistant.
- For the 2nd battery not working I would guess you run into the 3 device limit of the ESPproxy, if you are using it. Check die logs of the proxy!

alros100, 2024-10-03:
Hi Patrick

Success!

ok I figured out something. and I have been able to connect both batteries. the Solution is to first connect it with Batmon. It connects but can't get any data as it is the wrong protocol. then I shut down batmon and add the device with Patman and it immediately connects.

so Batmon must be doing something different to connect.

the weird thing is before connecting with batmon the "bluetoothctl info B0:D2:78:19:59:36" either doesnt find the battery or displays it not connected. after connecting with batmon, the same command shows that it is now connected. shows the TBA name and characteristics see screen grab,

when connected with patman, I have all the characteristics , including the individual cell voltages in the logs. in home assistant, I think you did not assign the individual cell voltages.

Al

[image]

patman15, 2024-10-03:
Hmm that sounds like the BMS only sends correct announcements when you connect. Very weird! (And against the BT spec I would say.)
Anyhow can you try the following:
1. Check with `bluetoothctl info <MAC address> that name, characteristics **are not** there
2. Use `bluetoothctl connect <MAC> to connect
3. Check with `bluetoothctl info <MAC address> that name, characteristics **are** there.

I'm wondering how the App finds the BMS, if it does not identify via name or characteristics. Does the app use the MAC address?

alros100, 2024-10-04:
Hi Patrick 

Now that the two batteries have been "found". The terminal command seems to always show the battery connected and displaying the characteristics as in the screen capture from My post above.

Before I did the Batmon trick. The terminal command never showed the battery name or characteristics, just its MAC address and the Bluetoothctl pair command always failed. One of my posts above had a screen capture of this

I will do a fresh install of home assistant on the pi 3 and retest your questions and then see if batmon allows the batteries to connect like it did on the Pi 5.

I can email Bjorn-Andreas to ask about how the app finds the battery.

Thanks again !
Al

patman15, 2024-10-04:
Hmm, makes sense that it works now, because the information is cached. Still can't figure out a reason why using Batmon helped to achieve that.
Did you install Home Assistant according to https://www.home-assistant.io/installation/raspberrypi ?
Just to verify...

Also,.if using [RPi internal Bluetooth there is this warning](https://www.home-assistant.io/integrations/bluetooth/):
[image]

alros100, 2024-10-04:
HI Patrick

Ok did a fresh install of on the Pi 3 - using the correct method. I did not apply those patches but the raspberry pi 5 was still connected to batteries 2 days going before I turned it off to test the pi3 

initially, the bluetooth info command shows not available, then I enter the pair command. in the screen grap you can see I did this several times with it returning device not available, then it returns the message advertisment monitor path registered. redoing the info command shows the full battery info now. The second battery took more attemps

so I guess this means you have to attempt a pair. and  secondly the randomness with the bluetooth command is strage when Batmon seems to work. I have to rush to work but I will install patman after work and see if in now finds the batteries. 

[image]

alros100, 2024-10-05:
Hi Patrick

so to follow up. on the clean pi 3 install - if I repeatedly try the pair command, eventually, I get  the path registered message. sometimes the full battery info shows up but most of the time not.  when I get the message, I immediately try to run patman but it doesn't find the battery. However, as before, if I run Batmon with the battery mac address , it attempts to connect, but seems to through exceptions getting data obviously due to the wrong protocol. Once again, running patmon finds the battery and once added to home assistant, the connection is not lost. I can then repeat for battery 2 and then I can shutdown and uninstall Batmon and both batteries are connected and return data almost continuously . the connections are stable after reboot, core updates, etc etc.  

I don't know enough about ble to offer any ideas but I will email Bjorn andreas and see if he has any ideas to offer why his app always sees and connects to the battery
Al

patman15, 2024-10-05:
Hey! Probably try `bluetoothctl connect <MAC>` instead of `pair`. I would guess this is also what Batmon does.

alros100, 2024-10-05:
Hi Patrick

ok that may be the solution. I did another clean install on raspberry pi 3. only Patmon installed. for some reason, the info commanded reported the battery name this time. I tried patmon but didn't find the battery. then I tried the connect command followed immediately by patmon and it found the battery and connected .  

So i guess that is the solution. you  must try to connect to the battery first ! 

now i'll see if bluetooth proxy can find it but so far no

thanks so much Patrick
Al

patman15, 2024-10-05:
I think you don't need to do a clean install all the time, HA can be reset by stopping it and deleting the `/config` directory. It will reinitialize next time it starts (of course all configuration and data is lost!). While HA being stopped you can also delete BT devices by using `bluetoothctl remove <mac_address>` That should give you the same starting point as with a clean install, but much faster.

> thanks so much Patrick

You're welcome! Sorry that it takes so long, but remote debugging with not having a similar device at hand is quite difficult. 😎

alros100, 2024-10-27:
Hi Payton

Everything is working great. Thank you again

Just a quick question. How frequently does your integration poll the battery state ? 

My automation currently stops at battery % reaches 60% or timer greater than 25 min. Just in case the connection is lost or the % level is missed between update ?

Thank

patman15, 2024-10-28:
Hi! Polling interval is default of HA, i.e. every 30 seconds. I will keep the issue open till I have completed my cleanup (need to write tests to keep it working from release to release) and merged it into the main line so it gets into the next release. **Till a new release having your device in the release notes, avoid updating!** :wink:
What might drive you to update is the [new sensor in v1.7.0 `link quality`](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#provided-information) that tells you whether the queries were successful. **But** be patient, see above.

patman15, 2024-11-05:
Hi @alros100, I just release [version 1.8.1](https://github.com/patman15/BMS_BLE-HA/releases/tag/v1.8.0) which contains the official support for your battery. Could you please check if there are no issues? I did some code changes and fixes. Installation should work normally through HACS. Since I renamed the BMS type, please make sure to first delete your device, then update, then you can re-add it. History should be maintained. I'll close this issue finally, when you can confirm the release is fine. Thanks!

alros100, 2024-11-05:
Awesome ! Thanks Patrick. I will try it out and post back here 

all the best
Al

alros100, 2024-11-05:
Hi Patrick

I can confirm the new update works as expected 

I used the Bluetoothctl connect - then integration finds battery and all is good

Thanks so much !

Al

[image]

patman15, 2024-11-06:
> I can confirm the new update works as expected
> 
Thanks a lot for confirming! Case closed, please open a  new issue in case you encounter some troubles, thanks!

> I used the Bluetoothctl connect - then integration finds battery and all is good
>
Strange, I changed detection to be purely on the name and thought that would fix this issue, but it seems it is not enough. :thinking:

closed 2024-11-06 by patman15