patman15/BMS_BLE-HA #55: Daly BMS MOS Temperature

VitalyIak, 2024-09-24:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

There is no MOS (BMS) temperature of Daly

### Describe the solution you'd like

Add MOS temperature for Daly

### Additional context

I made 3 debug tests from Android with original APP, where was different MOS temperature
**1-st test (29-28C deg)**
```
d2031200000000753000000045ffffffffffffffff787ed203183232314a4131353233303030333300000000000000000000ffebd2030200003d96
d2031200000000753000000044ffffffffffffffff75eed203183232314a4131353233303030333300000000000000000000ffebd2030200003d96
```
Values 44 and 45 means 29 and 28C deg...

**2-nd test - 25C deg**
```
d2031200000000753000000041ffffffffffffffff4abed203183232314a4131353233303030333300000000000000000000ffebd2030200003d96
```
Values 41 means 25C deg...

**3-rd test - 39-40-39-38C deg**
```
d203120000000075300000004fffffffffffffffff06de
d2031200000000753000000050ffffffffffffffff8aee
d203120000000075300000004fffffffffffffffff06ded203183232314a4131353233303030333300000000000000000000ffebd2030200003d96
d203120000000075300000004effffffffffffffff0b4e
```
Values 4f means 39C deg...
Values 50 means 40C deg...
Values 4e means 38C deg...

At all tests we can see changes of the 13-th byte which provide MOS temperate +40 in HEX
I don't know which requests are used for this response, I'll add all 3 debugs logs for Wireshark
[debug_hci_snoop.zip](https://github.com/user-attachments/files/17113130/debug_hci_snoop.zip)
in text format:
[test1.txt](https://github.com/user-attachments/files/17113145/test1.txt)
[test2.txt](https://github.com/user-attachments/files/17113146/test2.txt)
[test3.txt](https://github.com/user-attachments/files/17113147/test3.txt)

kirillch2000, 2024-09-24:
Yes, very useful improvement - I also really need to monitor MOS overheating

VitalyIak, 2024-09-25:
Also analyzed requests to BMS, 
at first we can see requests:
    Handle: 0x0028 (Unknown)
    Value: 48694c696e6b

    Handle: 0x0014 (Unknown)
    Value: d21000d400031809180d05351c43

    Handle: 0x0025 (Unknown)
    Value: 41542b5645523d3f0d0a

    Handle: 0x0018 (Unknown)
    Value: 41542b42414e443d3f

    Handle: 0x0018 (Unknown)
    Value: 41542b50524f44554354494e464f3d3f

    Handle: 0x0014 (Unknown)
    Value: d20300800029965f

    Handle: 0x0014 (Unknown)
    Value: a54002080000000000000000ef

    Handle: 0x0014 (Unknown)
    Value: 81100f05000200000000876a

Then, requests are repeated cyclically

    Handle: 0x0014 (Unknown) 
    Value: d2030000003ed7b9

    Handle: 0x0014 (Unknown)
    Value: d20300a900208791  

    Handle: 0x0014 (Unknown)
    Value: d20300c90003c656  

    Handle: 0x0014 (Unknown)
    Value: d20300cc00021797  

    Handle: 0x0014 (Unknown)
    Value: d203003e0009f7a3  

    Handle: 0x0014 (Unknown) 
    Value: d2030057000ce7bc

    Handle: 0x0014 (Unknown)
    Value: d20300d700012791  

    Handle: 0x0014 (Unknown)
    Value: d20300800029965f

VitalyIak, 2024-09-25:
@patman15 Please check all info there, I suppose it will be not really difficuult to add one request and parse it for MOS temperature? Thank you in advance!

patman15, 2024-09-25:
@VitalyIak will definitely do so, thanks a lot for all the information, it really looks like the things I need. Overall it looks like in addition to the query I currently run (which contains all info so far), there are additional commands that need to be integrated. I can add those for sure. Unfortunately, I run out of time, I need to take a pause the next weeks, so not sure I can quickly hack them in, but I will definitely come back to it and fix it when I can resume.

patman15, 2024-09-25:
Found some time, gave it a quick short, can you try the [branch](https://github.com/patman15/BMS_BLE-HA/tree/55-daly-bms-mos-temperature)?

patman15, 2024-09-26:
> But I think it'll be better to split one object Temperature to MOS temperature of BMS and Battery's temperature with average value.

Yeah, I know, but I'm hesitating to change it, because the different BMSs have very different logic on temperature sensors. There are e.g. some who have sensors also for power or per battery pack, so it would be very specific to the actual BMS. Also names differ a lot and due to lack of official documentation (for most BMSs) it's hard to make it consistent.

closed 2024-10-26 by patman15