patman15/BMS_BLE-HA #57: Integrate the Supervolt batteries delivered until 23.10.2023 from the Batmon HA integration

Karatzie, 2024-09-28:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I use the Batmon HA integration to connect my two Supervolt batteries to HA.
The Bluetooth connection with my RPI 3 is running but stopped after short or long time.
The RPI 3 does have a high CPU load and is not able to build ESPHome firmware files.
The Bluetooth connection with my RPI 4 is very bad and I got not a connection.
My RPI 4 is placed in a metallic case with passive cooling.
I would like to use a Bluetooth Proxy to connect the RPI 4.

### Describe the solution you'd like

Please add the Supervolt BMS from Batmon HA in your integration.
I will perform the test of this connection with my both batteries.
I have a description of the communication protocol. see attachments.

### Additional context

[rs485-uart-rs232 Protocol_BT.zip](https://github.com/user-attachments/files/17175617/rs485-uart-rs232.Protocol_BT.zip)
[bms-metrisun.pdf](https://github.com/user-attachments/files/17175618/bms-metrisun.pdf)

patman15, 2024-09-28:
Please see the README Supervolt batteries are already supported. Do you experience issues?

Karatzie, 2024-09-30:
Hello Patrick.

Thanks for your reply.
I expected that your Supervolt integration for my batteries does not work.
I tested your integration and got the information
"Keine unterstützen Geräte via Bluetooth gefunden."
I used the RPI 3 onboard Bluetooth and not a Bluetooth proxy.

Supervolt sold three different types of LiFePo4 batteries.
**Supervolt Standard V1 2019-2023**
This battery type has a black housing cover. 
This battery type includes a E&J BMS.

**Supervolt Standard V2 2023**
This battery type has a light blue housing cover.
I don't know the BMS type.

**Supervolt Standard V3 2024**
This battery type has a light blue housing cover.
I don't know the BMS type.

I own the **Supervolt Standard V1 2019-2023**.

How can we add the **Supervolt Standard V1 2019-2023** BMS in your integration?

BR
Andreas

patman15, 2024-09-30:
Hi Andreas! Sorry, I missed the fact that there are multiple versions, although I myself wrote only V3 is supported. 😳
So you say it worked with Batmon HA, correct? If so, I can most probably port the protocol rather quick. Thanks also for the description of the protocol, that usually helps a lot to get it working stable.
Nevertheless, I currently can't work on the integration, so I need to ask for a few weeks of patience to get started.

Karatzie, 2024-09-30:
Hi Patrick.

My batteries can be observed with the Battmon HA.
I have some problems with the connection betwenn Battmon HA and my Supervolt battaries.
The Connection is freezing and no data will be collect from the batteries.
A repowering of the Bluetooth device of the RPi3 reconnect sometimes.
If the repowering does not work I must reboot my RPi3 completly.
After reboot of the RPi3 is it very rarely possible that the Bluetooth configuration of the RPi3 is lost.
In this case I have to delete the Bluetooth device in HA and after reboot of the RPi3 the Bluetooth device is discovered from HA.
It is a little bit complicate but I am not under pressure.
It will be a great thing if you add the Supervolt V1 in your integration.
You can perform the changes if you have time for this action.
I am looking forward and will check your solution with my batteries.

I have some remarks for you.
My campervan will be placed in a storage over the winter end of October.
It will be back in April of the next year.
It is not possible to test the integration in this period. 

BR
Andreas

patman15, 2024-10-01:
Ok, got you! I might still have some time till end of October but can't guarantee it. Things are piling up already. 😎 Till April it should be no issue at all to get it done.

BikeAtor, 2024-10-08:
i have a (bad) working version with support for the old black Supervolt batteries in my fork.
You can give it a try.
@patman15: you can use my code to support the old Supervolt batteries.

patman15, 2024-10-08:
@BikeAtor thanks a lot! I might do some minor enhancements and add tests before releasing. B-)
Could you do me a favour and post the output of `bluetoothctl info <MAC address>` here?
I'd like to understand the announced characteristics of the device for proper auto-detection.

Karatzie, 2024-10-08:
Hi Patrick.

here are the infos:
[image]

BikeAtor, 2024-10-08:
My code is based upon the official doku from Bluetoothpotocol. You can get the official doku from Supervolt, but is has some minor glitches.

Karatzie, 2024-10-08:
I have attached the documentation in the first post.

BikeAtor, 2024-10-09:
I have another documentation, but the heatingpad isn't described there. So I think it's an older one.
[App_Protokoll.pdf](https://github.com/user-attachments/files/17311617/App_Protokoll.pdf)

BikeAtor, 2024-10-10:
I have another documentation, but the heatingpad isn't described there. So I think it's an older one.

[App_Protokoll.pdf](https://github.com/user-attachments/files/17331695/App_Protokoll.pdf)

patman15, 2024-11-06:
Started working on an implementation, but noticed that there are two different types:
[App_Protokoll.pdf](https://github.com/user-attachments/files/17311617/App_Protokoll.pdf) and [bms-metrisun.pdf](https://github.com/user-attachments/files/17175618/bms-metrisun.pdf)
The latter has all information in one frame, while the first requires 2 requests.
@BikeAtor your implementation decodes the 2 frames length 128 and 30, while the second has one frame, length 140.
@Karatzie any idea what your battery uses?
Do I need to implement both variants? :confused:

patman15, 2024-11-07:
> I got the document directly from Supervolt for the first black batteries V1. Maybe Karatzie can test my implementation, so we know which one to implement. But I think my document will be the correct one, cause it is already working :-)

Checked against the Android app, it definitely requests both values and combines them. Funnily it ignores the remaining capacity and calculates from SoC and design capacity.

Anyway, I pushed a first working version to the branch linked to this ticket. If you want to follow my progress / help me testing (please! :blush:), you can manually install from there. Should be working fine. Need to clean up tests and I did not yet get how the CRC works. The description is ... :shushing_face:

Karatzie, 2024-11-07:
> Started working on an implementation, but noticed that there are two different types: [App_Protokoll.pdf](https://github.com/user-attachments/files/17311617/App_Protokoll.pdf) and [...]

Hi Patrick.

I got the APP_protokoll.pdf from Supervolt Customer Service.

I searched for the string
[image]
and found the same string in the BMS-Meritsum.pdf in GITHUB.
It is possible that your BMS_BLE-HA  also works with Meritsum.
I belive that the code of the batmon-ha integration is based on the code from @BikeAtor.

I will test your test version.
Thank you for your effort.

BR
Andreas

BikeAtor, 2024-11-07:
First try: didn't work

Changed libatt to libatt* in manifest and ej-bms -> didn't work
Removed maufacturer id and service_uuid in both files -> Working!!

I will try to to see if manufacturer id can be added again.

BikeAtor, 2024-11-07:
Ok, with these changes it is running:

manifest.json (added *):
```
{
      "local_name": "libatt*",
      "service_uuid": "0000fff0-0000-1000-8000-00805f9b34fb",
      "manufacturer_id": 21320
    }
```

ej_bms.py (added *, removed service_uuid)
```
def matcher_dict_list() -> list[dict[str, Any]]:
        """Provide BluetoothMatcher definition."""
        return [
            {
                "local_name": "libatt*",
                "manufacturer_id": 21320,
                "connectable": True,
            }
        ]
```

BikeAtor, 2024-11-07:
[image]

Karatzie, 2024-11-07:
How can I install the BMS_BLE-HA localy?

BikeAtor, 2024-11-07:
I checked out branch "57-integrate-supervolt-batteries-v1" to my local computer and copied the files to my HA-instance.

It should be working in HA in the Advanced Terminal as well.

Maybe (not sure if last cp is right):
```
mkdir tmp
cd tmp
git clone https://github.com/patman15/BMS_BLE-HA.git
cd BMS_BLE-HA
git switch 57-integrate-supervolt-batteries-v1
cp -r custom_components/bms_ble/* ../../config/custom_components/bms_ble/
```

And don't forget to make the changes from my Post. I use "Studio Code Server" to edit files in HA.

BikeAtor, 2024-11-07:
No, there is no really need for it. If some more files are there it doesn't matter. I wanted to keep my own implementation there. So I just did copy the new files over the old ones.

But you can do it if you like. All needed files are checked out from git. And HA only looks for the files. There is no other database.

BikeAtor, 2024-11-07:
Here are my changes to configuration.yaml
```
template:
  # Supervolt cell voltages
  - sensor:
      - name: cell_voltage_1_0
        state: >-
          {{ state_attr('sensor.libatt55b4_delta_voltage', 'cell_voltages')[0] }}
        unit_of_measurement: "V"
        state_class: measurement
        device_class: voltage
        availability: >-
          {{ has_value('sensor.libatt55b4_delta_voltage') }}
  - sensor:
      - name: cell_voltage_1_1
        state: >-
          {{ state_attr('sensor.libatt55b4_delta_voltage', 'cell_voltages')[1] }}
        unit_of_measurement: "V"
        state_class: measurement
        device_class: voltage
        availability: >-
          {{ has_value('sensor.libatt55b4_delta_voltage') }}
  - sensor:
      - name: cell_voltage_1_2
        state: >-
          {{ state_attr('sensor.libatt55b4_delta_voltage', 'cell_voltages')[2] }}
        unit_of_measurement: "V"
        state_class: measurement
        device_class: voltage
        availability: >-
          {{ has_value('sensor.libatt55b4_delta_voltage') }}
  - sensor:
      - name: cell_voltage_1_3
        state: >-
          {{ state_attr('sensor.libatt55b4_delta_voltage', 'cell_voltages')[3] }}
        unit_of_measurement: "V"
        state_class: measurement
        device_class: voltage
        availability: >-
          {{ has_value('sensor.libatt55b4_delta_voltage') }}

  - sensor:
      - name: cell_voltage_2_0
        state: >-
          {{ state_attr('sensor.libatt0ffc_delta_voltage', 'cell_voltages')[0] }}
        unit_of_measurement: "V"
        state_class: measurement
        device_class: voltage
        availability: >-
          {{ has_value('sensor.libatt0ffc_delta_voltage') }}
  - sensor:
      - name: cell_voltage_2_1
        state: >-
          {{ state_attr('sensor.libatt0ffc_delta_voltage', 'cell_voltages')[1] }}
        unit_of_measurement: "V"
        state_class: measurement
        device_class: voltage
        availability: >-
          {{ has_value('sensor.libatt0ffc_delta_voltage') }}
  - sensor:
      - name: cell_voltage_2_2
        state: >-
          {{ state_attr('sensor.libatt0ffc_delta_voltage', 'cell_voltages')[2] }}
        unit_of_measurement: "V"
        state_class: measurement
        device_class: voltage
        availability: >-
          {{ has_value('sensor.libatt0ffc_delta_voltage') }}
  - sensor:
      - name: cell_voltage_2_3
[... 93 lines cut ...]
          - type: history-graph
            title: Zellspannungen 1
            entities:
              - entity: sensor.cell_voltage_1_0
                name: Zelle 0
              - entity: sensor.cell_voltage_1_1
                name: Zelle 1
              - entity: sensor.cell_voltage_1_2
                name: Zelle 2
              - entity: sensor.cell_voltage_1_3
                name: Zelle 3
            hours_to_show: 24
          - type: history-graph
            title: Zellspannungen 2
            entities:
              - entity: sensor.cell_voltage_2_0
                name: Zelle 0
              - entity: sensor.cell_voltage_2_1
                name: Zelle 1
              - entity: sensor.cell_voltage_2_2
                name: Zelle 2
              - entity: sensor.cell_voltage_2_3
                name: Zelle 3
            hours_to_show: 24
```

patman15, 2024-11-07:
> How can I install the BMS_BLE-HA localy?

See the description for [manual installation](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual) in the README.

patman15, 2024-11-07:
> Ok, with these changes it is running:
> manifest.json (added *):
> ej_bms.py (added *, removed service_uuid)

Thanks, these are the things that are impossible to know without having a battery. Just to verify (because I need to make the detection is precise as possible to avoid interference with other types), can you post a screenshot from _nRF connect_ showing the advertisements of the battery? Like this one: https://github.com/patman15/BMS_BLE-HA/issues/59#issuecomment-2440113013 ?
Thanks!

BikeAtor, 2024-11-07:
They look correct. I will try some more Tests.

[image]

---

[image]

---

[image]

patman15, 2024-11-07:
I need the advertisement, see the screenshot, I assume that although the 6e4000... UUID is mostly used, the device advertises the fff0 ... Can you send me the screenshot that is equivalent to the one I linked? Only that one shows what is advertised.

BikeAtor, 2024-11-07:
??? It's working now with  "service_uuid": BMS.uuid_services()[0], ???
So only the * was missing. Or if it's a Regex libatt[0-9A-F]4.

Karatzie, 2024-11-07:
I copied the files manually to my HA server.
[image]

I reboot the HA server completely.
The integration will not find.
[image]

In the past, I deleted the folder bms_ble manually.
Was that ok?
What can I do?

BikeAtor, 2024-11-07:
You should only copy the content of https://github.com/patman15/BMS_BLE-HA/tree/57-integrate-supervolt-batteries-v1/custom_components/**bms_ble** to the directory **config/custom_components/bms_ble**, not the whole repository. Now you have config/custom_components/bms_le/custom_components/bms_ble.

BikeAtor, 2024-11-07:
@patman15 : Is this the information you like?
[image]

patman15, 2024-11-07:
@BikeAtor yes, thanks! So it does not advertise the service at all. :disappointed: I'll remove that, stick to the name and manufacturer ID. Will fix that. Thanks a lot!

BikeAtor, 2024-11-07:
I tried it another time and I had to remove the "service_uuid" again. Maybe something was chached.
So without it always works, as expected with the advertising data.

patman15, 2024-11-07:
:+1: just pushed a commit fixing the detection.

Karatzie, 2024-11-07:
The integration does not find my batteries at the moment.
@patman15 are you going to share the changed files today?

BikeAtor, 2024-11-07:
They are already in the brach 57-integrate-supervolt-batteries-v1

I reinstalled the directory and rebooted the system -> working

[image]

Many thanks to Patrick

Karatzie, 2024-11-07:
[image]
I am also happy.

patman15, 2024-11-07:
> Required test coverage of 100% reached. Total coverage: 100.00%
> 
> Results (18.53s):
>      163 passed
Also happy (for today). Need to figure out how the CRC is calculated.
Thanks to you guys for the support. I will integrate it into the next release 1.9.0 when I'm done with a few other things.

Karatzie, 2024-11-07:
One battery has problems wiht the connection.
[image]

Karatzie, 2024-11-07:
Now have the other battery problems
[image]

Karatzie, 2024-11-07:
Both batteries are comminicate at the moment.
The values will be updated all 300 seconds.
That is a long time

patman15, 2024-11-07:
@Karatzie Update rate should be 30s, so I suspect an issue with Bluetooth. Can you please enable the RSSI and link quality sensor and post the graphs of ~1h?

patman15, 2024-11-07:
- either a bug -> I would need a debug log, or
- issue with the Bluetooth controller, which one do you have? Can you check the [Bluetooth integrations settings](https://my.home-assistant.io/redirect/integration/?domain=bluetooth) send a screenshot of those? It should tell your adapter.

Karatzie, 2024-11-07:
I use an USB Bluetooth adapter with a RB Pi 4.
The adapter is placed 40 cm away from the batteries.
I have also problems with the batmon-HA AddOn.
After a longer runtime, the Bluetooth connection is not stable.
I observe a lot of gaps in the communication.
I this case, I delete the Bluetooth Adapter and reboot the RB Pi4.
I will do it now.
If this not work, I will try a Bluetooth Proxy tomorrow.
This should be better run.

patman15, 2024-11-07:
Please try a Bluetooth proxy or an [external quality dongle](https://www.home-assistant.io/integrations/bluetooth#known-working-high-performance-adapters). I read that the [RPi has issues with Bluetooth](https://www.home-assistant.io/integrations/bluetooth#cypress-based-adapters). You can also try disabling one battery and see if the other then works well. (I would guess so)

Karatzie, 2024-11-07:
The doungle works fine with my Bluetti power station.
The bluetooth device of the both supervolt batteries are very bad.
I will use the bluetooth proxy tomorrow.

Karatzie, 2024-11-07:
The Bluetooth connection with batmon-ha runs good.
This graph displays the data transfers of the hole day.
[image]

BikeAtor, 2024-11-08:
@patman15 : I don't think it is a problem with your integration. The pauses are from some tests (I removed 0ffc to test the integration).

[image]

[image]

Karatzie, 2024-11-08:
I have at the End four Bluetooth devices in my Camper Van.
How can I manage which device has to use the Bluetooth Proxy and which device has to use the RB Pi Bluteooth connection?

Karatzie, 2024-11-08:
I have a Bluetooth Proxy running in my camper van.
The signal exchange is running.
After 6 minutes the Link Quallity and the Signalstärke from both batteries are offline.
[image]
The Battery signals are staying online.
[image]

patman15, 2024-11-08:
> How can I manage which device has to use the Bluetooth Proxy and which device has to use the RB Pi Bluteooth connection?

I'm not expert on the proxy, but I think you can't. It's automatically selected by signal strength.

> After 6 minutes the Link Quallity and the Signalstärke from both batteries are offline.

That's weired. Do you have a debug log?
Did you set active to true?
```
bluetooth_proxy:
  active: true
```

Karatzie, 2024-11-09:
I set active true.
I saw an entry in the Bluetooth Proxy Log
[image]
The entry scan duration 300s sems to be the root cause for this behavior.
[image]

Karatzie, 2024-11-09:
I see a little problem with the symbols for battery voltage and cussent.
[image]
Why is used for the voltage a sinus symbol? The voltage of a battery is always DC without a sinus.
For the current is also used a current AC symbol. We have a special case here. The current can be positive during charging and negative during decharging.
In HA is a current-dc symbol.
That can be used for voltage and current.
Where are the symbols assigned?
In your integration or automatically in HA?

Karatzie, 2024-11-09:
I used this symbol.
[image]

mdi:current-dc

Karatzie, 2024-11-09:
My first battery has following texts:
[image]

The second has additional texts.
[image]

I activate the battery 1 first.
The second baddery was activate later.

patman15, 2024-11-09:
> I used this symbol. mdi:current-dc
>
Ok, look good. 

What about this one for the voltage?
[image] mdi:triangle-wave

Karatzie, 2024-11-09:
A triangle wave has also a amplitude from minus to plus.
[image]

Karatzie, 2024-11-09:
Here are some multimeter symbols
[image]

patman15, 2024-11-09:
> My first battery has following texts: 
> The second has additional texts. 
> I activate the battery 1 first. The second baddery was activate later.

I'm looking into it, but not sure yet if/what is the issue. I can replicate it, but it makes no sense to me. :disappointed:

Karatzie, 2024-11-09:
The texts in @BikeAtor picture are the same for both battery.
[image]
From my point of view should the texts for each battery different.

patman15, 2024-11-09:
No worries, I'm just seeing ghost already. :face_with_head_bandage: 
About the symbol: can't find a good one for DC voltage in the [library](https://pictogrammers.com/library/mdi/). Also NUT uses the sine wave for the voltage (although not a super good argument :wink:).

[2 more replies]

closed 2024-11-11 by patman15