patman15/BMS_BLE-HA #59: Failing to add support for Creabest/ CBT-Power-BMS

snipah, 2024-09-30:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Unfortunately I'm failing to add a support for Creabest/ CBT-Power-BMS. I followed the steps described in die contributions guidelines but I fail to amend to contents of basebms.py in my newly created creabest_bms.py. I did the required changes to all the other files and even amended the seplos-file with my identifier: The battery was discovered in Home Assistant, but of course communication failed.

### Describe the solution you'd like

I would like to be able to use your great plugin with my battery in my camper.

UUIDs are the same all other manufacturers. By batteries name is found starting with "140*" - no clue about other Creabest batteries. Maybe other owners can help here and provide the names under which their batteries are advertised via Bluetooth. This can be easily seen in the app.

`Device 38:AB:41:1F:40:CE (public)
        Name: 140R00036D
        Alias: 140R00036D
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Unknown                   (0000ffb0-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000fff0-0000-1000-8000-00805f9b34fb)
        ManufacturerData.Key: 0x0000 (0)
        ManufacturerData.Value:
  ff ff ff ff 64 00 ff 00 00 00 00 00 00 00 00 00  ....d...........
        RSSI: 0xffffffc7 (-57)
        TxPower: 0x0000 (0)
        AdvertisingFlags:
`

How to proceed? Can someone please help me? If any info is missing, I will provide it gladly!

snipah, 2024-09-30:
...connecting via bluetoothctl works...

> Connection successful
➜  ~ bluetoothctl info 38:AB:41:1F:40:CE    

Device 38:AB:41:1F:40:CE (public)
        Name: 140R00036D
        Alias: 140R00036D
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000fc60-0000-1000-8000-00805f9b34fb)
        UUID: Amazon.com Services, Inc. (0000fe00-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ff90-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ffa0-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ffb0-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ffc0-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ffd0-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ffe0-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ffe5-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000fff0-0000-1000-8000-00805f9b34fb)
        ManufacturerData.Key: 0x0000 (0)
        ManufacturerData.Value:
  ff ff ff ff 64 00 ff 00 00 00 00 00 00 00 00 00  ....d...........
        RSSI: 0xffffffc7 (-57)
        TxPower: 0x0000 (0)
        AdvertisingFlags:
  06                                               .               
        Battery Percentage: 0x64 (100)

patman15, 2024-09-30:
The information looks useful. Where did you get stuck? What is your question? You say 'of course communication failed'. Do you have a description of the protocol used by the battery, what's the issue?

Gruni22, 2024-09-30:
I found the Dataseheet. 
https://cloud.loichis.de/s/yMSctC7nXP9YCCM

Seems like it needs notification on FFE4 and write to FFE9. 
That is also what the App does. But I am unable to receive data via FFE4, no matter what I send to FFE9. And after 5 seconds of interaction the connection is closed. 
Possibly there is a Anti Hijack Password, which I could not find in the App. 

By the way, I do not want to use HA. I want to create my own Web Based App for a Caravan based on ESP32.

snipah, 2024-09-30:
- Connection Password Acquisition
Input the following string "TTM:PWD-?" to RX to acquire connection password.
“TTM:PWD-XXXXXX": It means the successful operation and XXXXXX is the password in 6-bit.
"TTM:ERP\r\n\0": It means the incorrect command format

[image]

snipah, 2024-10-01:
@Gruni22  Just out of curiosity: How does your battery identify via bluetooth? What are name and/ or alias. In my case it is 140R00036D in both cases for a Creabest VB037 which is the "CREABEST LiFePO4 Battery 12V 135Ah for RV Solar Caravan Leisure Camping Marine Boat with Bluetooth". This would be helpful for the BT-identifiert, I assume

Gruni22, 2024-10-01:
It is a VB029
Device Service UUID: 000003c1-0000-1000-8000-00805f9b34fb
Device Name: 100R0002D1
But I think you have to ferget about the TTM commands. They are only for real serial access, as far as I know.
Or dou you think, one has to send the AT (TTM...) commands via FFF9?

snipah, 2024-10-01:
Thanks... a possible pattern to recognize the Creabest-batteries might be 1**R000***. We will hopefully see.

> Strings starting with "TTM" will be regarded as AT commands to be parsed and executed, and will return exactly the 
same from the serial port. Afterwards the execution result will be output (“TTM:OK\r\n\0" or "TTM:ERP\r\n\0", etc.). Data 
packets which do not start with “TTM” will be regarded as transparent transmission data. CRLF cannot be added 
in any command.

I do not see, how you would transfer "strings" over the serial console, right?

patman15, 2024-10-01:
> Am stuck at filling in my creabest_bms.py-file... especially points 2 & 3. Currently only have the dummy file in place. BMS is detected and reports the dummy values... but now I want real values :)

Well, that is the hardest part. 😄 You need to figure out which commands to send to which characteristic and then parse the reply. You can have a look at the other implementations, e.g. daly_bms.py for reference.

Best way to identify is using a UUID of a service (if really unique) or combined with a name pattern.

@Gruni22 
> By the way, I do not want to use HA. I want to create my own Web Based App for a Caravan based on ESP32.

In principle you can use the libraries also stand-alone. The interface is rather simple and can be seen from the base_bms.py. In the libraries there shouldn't be anything HA specific, that was the design goal. I thought about making it a separate pypi library but couldn't find the time yet.

As mentioned in other issues, I currently need to pause on developing, but in a few weeks I can support more actively.

snipah, 2024-10-01:
> As mentioned in other issues, I currently need to pause on developing, but in a few weeks I can support more actively.

Take your time, as there's loads of more important things to do. Maybe we can get this implemented anytime in the future. Would appreciate your help as I have hardly any clue about coding.

Gruni22, 2024-10-01:
I think the main spot is to find out, why the batteries does always disconnect. 
@snipah perhaps you want to get in touch with me for some investigation (I am German).

snipah, 2024-10-01:
@Gruni22 Sure, if I can be of any help. How to get in touch?

Gruni22, 2024-10-01:
@snipah I added my Email to bio.

Alex-DeIt, 2024-10-01:
Hello everybody in this discussion, pls. let me know if I can support somehow. I have no experience related to BMS transmission protocols and I am a survivor related to Home Assistant, but if I can help, it would be a pleasure for me. Thank you again, patman15 for your efforts.

snipah, 2024-10-01:
[Kommunikationsprotokoll_der_VB-Serie-DE.pdf](https://github.com/user-attachments/files/17215062/Kommunikationsprotokoll_der_VB-Serie-DE.pdf)

Gruni22, 2024-10-03:
I prepared bunch of data of a working (Java) setup.
But I do not know, how to push a branch.
I uploaded my work here.
[tools.zip](https://github.com/user-attachments/files/17246529/tools.zip)

snipah, 2024-10-03:
Wow! Great work! Patrick, up to you to amend to running Python code :)

Gruni22, 2024-10-03:
Thanks, the python cracks should be able to modify the templates accordingly.
If not - ChatGPT can help :-)
But for me, python is really useless for my case. I need C++ for Arduino.

patman15, 2024-10-03:
> Wow! Great work! Patrick, up to you to amend to running Python code :)

Great, thanks, will do. Out of curiosity, does the implementation match the [specification posted above](https://github.com/patman15/BMS_BLE-HA/issues/59#issuecomment-2386746573)?

snipah, 2024-10-03:
Which "specification" are you referring to? Gruni22 seems to have identified all relevant parameters but coded the parsing-logic in java. I am not capable of coding in python on this level unfortunately. Hope you have everything you need to have Creabest/ CBT Powers batteries supported in your great plugin.  If you need help, just let me know.

patman15, 2024-10-03:
> Which "specification" are you referring to?

The one you [posted.](https://github.com/patman15/BMS_BLE-HA/issues/59#issuecomment-2386746573)

Gruni22, 2024-10-04:
Perhaps I can assist.
@patman15 we could also get in closer touch. 
The language Java or Python is not that different. I have to understand your logic for the notification and write subscriptions.

patman15, 2024-10-04:
@Gruni22, sure, thanks for the offer, I'll ping you as soon as I can get started. Thanks a lot for the offer!

snipah, 2024-10-06:
> > Am stuck at filling in my creabest_bms.py-file... especially points 2 & 3. Currently only have the dummy file in place. BMS is detected and reports the dummy values... but now I want real values :)
> 
> Well, that is the hardest part. 😄 You need to figure out which commands to send to which characteristic and then parse the reply. You can have a look at the other implementations, e.g. daly_bms.py for reference.
> 
> Best way to identify is using a UUID of a service (if really unique) or combined with a name pattern.
> 
> As mentioned in other issues, I currently need to pause on developing, but in a few weeks I can support more actively.

I tried and failed miserably. I have to admit that my coding skills too bad. I understood the logic but failed to amend it to the requirements for Creabest. I will patiently wait until Patrick finds the time to enhance the add on.

patman15, 2024-10-06:
@snipah heads up: will do my best still this month

snipah, 2024-10-26:
> @snipah heads up: will do my best still this month

@patman15 Any new regarding the implementation, Patrick?

patman15, 2024-10-26:
Preparing the ground for it with a maintenance release covering all issues of the last month and a [branch](https://github.com/patman15/BMS_BLE-HA/tree/code-redundancy-reduction) to make adding new BMSs even simpler. Can start within the next days.

snipah, 2024-10-27:
Sounds great. Looking forward to seeing the results 👍

patman15, 2024-10-27:
@snipah can you send me the information the battery advertises (content of the advertisement) using _nRF connect_? It's on the _scanner_ tab if you just tap the device and then click _raw_. Trying to figure out the best way to identify the BMS. Thanks!

snipah, 2024-10-27:
[image]

snipah, 2024-10-27:
[image]

patman15, 2024-10-27:
one more question: what does it say under "Incomplete list of 16-bit Service UUIDs"? "`0xFF0F`, `0xFFB0`"?

snipah, 2024-10-27:
[image]

patman15, 2024-10-27:
@Gruni22 the frame format is as follows:
for TX:
`AA 55 03 00 03  0A 0D`
`HEAD  CMD   CRC TAIL`
`CMD` is 16 bit low byte first
`CRC` is 8-bit sum of everything before the CRC, except `HEAD`
for RX:
`AA 55 03  02  0204 0B  0D 0A`
`HEAD  CMD LEN DATA CRC TAIL`
At least this is my current best knowledge.

Gruni22, 2024-10-27:
TX:
CMD can be one of:
Version of Battery: 03
Cell voltage 00-07: 05
Cell Voltage 08-15: 06
Cell Voltage 16-23: 07
Cell Voltage 24-31: 08
Temperature: 09
Total Capacity and Warnings: 0A
Voltage and Current: 0B
Load and remaining Time: 0C
Number of cycles: 15
Warnings: 21

Head.., CMD,alwaysZero,CMD,Tail...;
AA, 55, CMD,    00,    CMD, 0A, 0D;

RX = message:
Head.., CMD, message[i], Tail..
AA, 55, CMD, message[i], OD, OA

Decoding for 03:
softwareVer = message[4];
hardwareVer = message[5];

Decoding for 09:
temperature = (short)(message[5] << 8 | message[4] & 255); //Celsius

Decoding for 0A:
capacity =  message[4]; // Total Ah

Decoding for 0B:
floatVar1 = (float)((message[4] & 255) + (message[5] & 255) * 256);
shortVar1 = message[6];
shortVar2 = message[7];
voltage = (floatVar1 + (float)((shortVar1 & 255) * 256 * 256) + (float)((shortVar2 & 255) * 256 * 256 * 256)) / 1000.0F; // Volt
current = (float)((message[8] & 255) + (message[9] & 255) * 256 + (message[10] & 255) * 256 * 256 + (message[11] & 255) * 256 * 256 * 256) / 1000.0F; //Amperes

Decoding for 05-08:
int j = (message[2] - 5) * 5;
for (int i = 0; i <= 8; i += 2) {
  short s = message[i + 4];
  short t = message[i + 5];
  int cellVoltage = (s & 255) | (t & 255) * 256;
  if (cellVoltage > 0) {
    cell[j] = cellVoltage;
  }
  ++j;
}

I ommited the error Handling because it is quite complicated. 

Please let me know if you need it, or any other Information. 

Perhaps we can meet in a video call. 

I do not understand anything about CRC Checksum low and high byte. 
This is only reverse engineered from the android App. 

How i implemented Error handling:
This is my BMS Handling Code:

String errList[16] = {
  "MOS Ladezustand",
  "MOS Entladezustand",
  "Entladetemperatur Hoch",
  "Entladetemperatur Niedrig",
  "Ladetemperatur Hoch",
  "Ladetemperatur Niedrig",
  "Spannung Hoch",
[... 221 lines cut ...]
        char k = paramString.charAt(m);
        char n = paramString.charAt(m + 1);
        
        if (k >= '0' && k <= '9') {
            k -= '0';
        } else if (k >= 'A' && k <= 'F') {
            k = k - 'A' + 10;
        } else {
            return nullptr;
        }
        
        if (n >= '0' && n <= '9') {
            n -= '0';
        } else if (n >= 'A' && n <= 'F') {
            n = n - 'A' + 10;
        } else {
            return nullptr;
        }
        
        arrayOfShort[i] = (short)((k << 4) + n);
    }
    
    return arrayOfShort;

}

Gruni22, 2024-10-27:
LEN and CRC might be correct. 
I do Not Check/Calculate it. 
I am just looking if Head and Tail is correct.

Gruni22, 2024-10-28:
`int hardV;
int softV;
int temperature;
int capacity;
float voltage;
float current;
int chargecycle;
int cell1;
int cell2;
int cell3;
int cell4;
String chargestate;
int duration;
String alarmmessage;
bool receivedData = false;
String errList[16] = {"MOS Ladezustand", "MOS Entladezustand", "Entladetemperatur Hoch", "Entladetemperatur Niedrig", "Ladetemperatur Hoch", "Ladetemperatur Niedrig", "Spannung Hoch", "Ladestrom Hoch", "Entladestrom Hoch", "Batterie schwach", "Spannung Niedrig", "Kurzschluss", "Kommunikation", "MOS Temperatur Hoch", "AFE Einstellungen", "AFE Anschließen"};

void showData(String answer)
{
    receivedData = true;
    short *message = hexToShorts(answer);
    int messageLength = sizeof(answer);
    float floatVar1;
    float floatVar2;
    short shortVar1;
    short shortVar2;
    short shortVar3;
    
    if (message[2] == 3)
    {
        if (messageLength >= 6)
        {
            softV = message[4];
            hardV = message[5];
        }
    }
    if (message[2] == 9)
    {
        if (messageLength >= 5)
        {
            temperature = (short)(message[5] << 8 | message[4] & 255);
        }
    }

    // I only need the first 4 cells.
    if (message[2] >= 5 && message[2] <= 8)
    {
        int j = (message[2] - 5) * 5;
        for (int i = 0; i <= 8; i += 2)
        {
            short s = message[i + 4];
            short t = message[i + 5];
            int cellVoltage = (s & 255) | (t & 255) * 256;
            if (cellVoltage > 0)
            {
                int currentCell = j + 1;
                if (currentCell == 1)
                {
                    cell1 = cellVoltage;
                }
                else if (currentCell == 2)
                {
                    cell2 = cellVoltage;
                }
                else if (currentCell == 3)
                {
                    cell3 = cellVoltage;
                }
                else if (currentCell == 4)
                {
[... 239 lines cut ...]
            k = k - 'A' + 10;
        }
        else
        {
            return nullptr;
        }

        if (n >= '0' && n <= '9')
        {
            n -= '0';
        }
        else if (n >= 'A' && n <= 'F')
        {
            n = n - 'A' + 10;
        }
        else
        {
            return nullptr;
        }

        arrayOfShort[i] = (short)((k << 4) + n);
    }

    return arrayOfShort;
}`

patman15, 2024-10-28:
@snipah getting there:
[image]

@Gruni22 
> Please let me know if you need it, or any other Information.
>
Did not look too deep yet (will continue tomorrow), but I of course looked at your code. What I'm still missing is
- remaining capacity in Ah
- seems like there is a second temperature sensor ...
- number of cells that are in the battery as number
- delta voltage

> Perhaps we can meet in a video call.
>
I have quite irregular hacking times, but you can try to contact me on Discord. My details are in the github profile.

snipah, 2024-10-28:
> @snipah getting there: ![foto](https://private-user-images.githubusercontent.com/14628713/380834115-f5882550-428b-4b51-835a-35c0da789247.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAxMzk2MjMsIm5iZiI6MTczMDEzOTMyMywicGF0aCI6Ii8xNDYyODcxMy8zODA4MzQxMTUtZjU4ODI1NTAtNDI4Yi00YjUxLTgzNWEtMzVjMGRhNzg5MjQ3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI4VDE4MTUyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFjOWMxM2Y1ZWE4ZTU2MDA4ZDFlYzI5YWY0YmY0NDMyMGUxYTRkZjBhNDJjNTJmNWRiMWZjZWY3YWQ3YWMxMzEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.rCMf-s1p0mF1vfvMZuJgSDfNGQPpYvH9qu11564KbgE)

Great news! Please let me know if I can be of any help oder you need to test something. Thanks a lot!

snipah, 2024-10-29:
> delta voltage - what do mean?

[image]

Gruni22, 2024-10-29:
@snipah if it is required it must be calculated.
@patman15 By the way, you asked before, if the protocol matches to the uploaded PDF. It does not.
"Kommunikationsprotokoll_der_VB-Serie-DE.pdf]"

patman15, 2024-10-29:
> if it is required it must be calculated.

I can do so, no problem, just preferring reported values over calculated ones.

> @patman15 By the way, you asked before, if the protocol matches to the uploaded PDF. It does not.
"Kommunikationsprotokoll_der_VB-Serie-DE.pdf]"

Thanks for confirming, noticed yesterday when implementing as well.

> seems like there is a second temperature sensor ... - where?

`uint8_t temp_frame[] = {0xAA, 0x55, 0x09, 0x0C, 0x15, 0x00, 0x15, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x3F, 0x0D, 0x0A};`
I would guess pos 5/6 and 7/8 are temperature values (21°C)?

> remaining capacity in Ah can be calculated from total capacity in Ah and remaining capacity in %.

Hmm, strange BMS, but ok. Where do we get the total capacity from? From the name of the device?! :fearful:

Gruni22, 2024-10-29:
Regarding the temperature, the original app code does only analyze pos 5 and 6. So in fact, I have no idea what pos. 7/8 is.
I have another reading:
uint8_t temp_frame[] = {0xaa, 0x55, 0x09, 0x0c, 0x0f, 0x00, 0x11, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x35, 0x0d, 0x0a};
which is 16.0 °C.
and:
uint8_t temp_frame[] = {0xaa, 0x55, 0x09, 0x0c, 0x14, 0x00, 0x14, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x3d, 0x0d, 0x0a};
which is 20.0 °C

From the name of the device?! Yes :-D it is really strange.

Gruni22, 2024-10-29:
I have the VB029 with 40Ah.
It is advertising as 100R0002E3_7E

Gruni22, 2024-10-29:
This is extracted from the powerQuarry App.
The VB037 was not yet existing there. But I really think the total capacity is hardcoded.

`
if (var1 != null) {
         if (var1.startsWith("VB101")) {
            var1 = "VB101 12.8V 172Ah";
         } else if (this.dev_name.startsWith("VB007")) {
            var1 = "VB007 12.8V 120Ah";
         } else if (this.dev_name.startsWith("VB102")) {
            var1 = "VB102 12.8V 258Ah";
         } else if (this.dev_name.startsWith("VB201")) {
            var1 = "VB201 12.8V 172Ah";
         } else if (this.dev_name.startsWith("VB016")) {
            var1 = "VB016 12.8V 200Ah";
         } else if (this.dev_name.startsWith("VB103")) {
            var1 = "VB103 12.8V 240Ah";
         } else if (this.dev_name.startsWith("VB104")) {
            var1 = "VB104 12.8V 480Ah";
         } else if (this.dev_name.startsWith("CBT001")) {
            var1 = "CBT001 12.8V 480Ah";
         } else if (this.dev_name.startsWith("VB106")) {
            var1 = "VB106 12.8V 42Ah";
         } else if (this.dev_name.startsWith("VB029")) {
            var1 = "VB029 12.8V 40Ah";
         } else if (this.dev_name.startsWith("VB030")) {
            var1 = "VB030 12.8V 100Ah";
         } else if (this.dev_name.startsWith("VB031")) {
            var1 = "VB031 12.8V 750Ah";
         } else if (this.dev_name.startsWith("VB032")) {
            var1 = "VB032 12.8V 100Ah";
         } else if (this.dev_name.startsWith("VB037")) {
            var1 = "VB037 12.8V";
         } else if (this.dev_name.startsWith("VB038")) {
            var1 = "VB038 12.8V 50Ah";
         } else if (this.dev_name.startsWith("VB035")) {
            var1 = "VB035 12.8V";
         } else if (this.dev_name.startsWith("VB028")) {
            var1 = "VB028 12.8V 300Ah";
         } else if (this.dev_name.startsWith("VB024")) {
            var1 = "VB024 12.8V";
         } else if (this.dev_name.startsWith("VB001")) {
            var1 = "VB001 12.8V 172Ah";
         } else if (this.dev_name.startsWith("VB002")) {
            var1 = "VB002 12.8V 258Ah";
         } else if (this.dev_name.startsWith("VB107")) {
            var1 = "VB107 12.8V 135Ah";
         } else if (this.dev_name.startsWith("VB017")) {
            var1 = "VB017 24V 135Ah";
         } else if (this.dev_name.startsWith("VB033")) {
            var1 = "VB033 24V 200Ah";
         } else if (this.dev_name.startsWith("VB042")) {
            var1 = "VB042 12.8V 60Ah";
         } else {
            var1 = "";
         }
`
Perhaps you can extract it from one of those responses, which is my VB029 with 40Ah:
Received data: AA 55 03 02 02 04 0B 0D 0A 
Received data: AA 55 05 0A 0B 0D 0B 0D 0B 0D 0A 0D 00 00 6E 0D 0A 
Received data: AA 55 06 0A 00 00 00 00 00 00 00 00 00 00 10 0D 0A 
Received data: AA 55 07 0A 00 00 00 00 00 00 00 00 00 00 11 0D 0A 
Received data: AA 55 08 0A 00 00 00 00 00 00 00 00 00 00 12 0D 0A 
Received data: AA 55 09 0C 15 00 15 00 00 00 00 00 00 00 00 00 3F 0D 0A 
Received data: AA 55 0A 06 64 13 0D 00 00 00 94 0D 0A 
Received data: AA 55 0B 08 58 34 00 00 00 00 00 00 9F 0D 0A 
Received data: AA 55 0C 0C 00 00 00 00 5B 06 00 00 03 00 00 00 7C 0D 0A 
Received data: AA 55 15 04 28 00 03 00 44 0D 0A 
Received data: AA 55 21 04 00 00 00 00 25 0D 0A 
Received data: AA 55 03 02 02 04 0B 0D 0A

snipah, 2024-10-29:
[image]

Gruni22, 2024-10-29:
I think here:
Received data: AA 55 15 04 **28 00** 03 00 44 0D 0A

Gruni22, 2024-10-29:
I have also another battery. Same Type.
0 cycles, 40 ah:
aa 55 15 04 **28 00 0000** 410d0a

patman15, 2024-10-29:
> I have also another battery. Same Type. 0 cycles, 40 ah: aa 55 15 04 **28 00 00 00** 41 0d 0a
> 
> For @snipah 's Battery it must be: aa 55 15 04 **87 00 42 00** 41 0d 0a

Thanks makes sense. I'll prepare a draft version and then will ask @snipah to verify with debug logs.

Gruni22, 2024-10-29:
@patman15 I sent you a discord invite.

snipah, 2024-10-29:
> > I have also another battery. Same Type. 0 cycles, 40 ah: aa 55 15 04 **28 00 00 00** 41 0d 0a
> > For @snipah 's Battery it must be: aa 55 15 04 **87 00 42 00** 41 0d 0a
> 
> Thanks makes sense. I'll prepare a draft version and then will ask @snipah to verify with debug logs.

Ready when you are ;)

patman15, 2024-10-29:
> Ready when you are ;)

@snipah ready! Are you able to check the branch? (see linked one top right of this issue) Basically you would need to follow the [instructions for manual installation](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual).
Would be cool if you could enable debug logging and then run a short charging and a discharging test to see if all values react accordingly.

snipah, 2024-10-29:
Unfortunately: "Keine unterstützen Geräte via Bluetooth gefunden."

patman15, 2024-10-29:
Stupid mistake, can you update and try again?

snipah, 2024-10-29:
Tried and the battery was recognized... however setup failed:

`2024-10-29 19:46:57.239 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request temperature info
2024-10-29 19:46:57.289 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=9 size=2 sign=False
2024-10-29 19:46:58.290 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request battery_level info
2024-10-29 19:46:58.339 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=10 size=1 sign=False
2024-10-29 19:46:58.369 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) Rx BLE data: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00\xd3\xfc\xff\xff\x07\r\n')
2024-10-29 19:46:58.369 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) Rx data CRC is invalid: 0x7 != 0x507
2024-10-29 19:46:59.340 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request design_capacity info
2024-10-29 19:46:59.389 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=21 size=2 sign=False
2024-10-29 19:47:02.490 DEBUG (MainThread) [custom_components.bms_ble] 140R00036D: no valid data received
`

Gruni22, 2024-10-29:
There is a typo:

BLE data: bytearray(b'\**xaaU**\x0b\x08\xf43\x00\x00\xd3\xfc\xff\xff\x07\r\n')

snipah, 2024-10-29:
`2024-10-29 19:50:33.727 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for 140R00036D (38:AB:41:1F:40:CE) as CBT Power Smart BMS
2024-10-29 19:50:33.727 DEBUG (MainThread) [custom_components.bms_ble] device data: {'name': '140R00036D', 'address': '38:AB:41:1F:40:CE', 'rssi': -57, 'manufacturer_data': {0: b'\xff\xff\xff\xffd\x00\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, 'service_data': {}, 'service_uuids': ['0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fc60-0000-1000-8000-00805f9b34fb', '0000fe00-0000-1000-8000-00805f9b34fb', '0000ff90-0000-1000-8000-00805f9b34fb', '0000ffa0-0000-1000-8000-00805f9b34fb', '0000ffb0-0000-1000-8000-00805f9b34fb', '0000ffc0-0000-1000-8000-00805f9b34fb', '0000ffd0-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', '0000ffe5-0000-1000-8000-00805f9b34fb', '0000fff0-0000-1000-8000-00805f9b34fb'], 'source': 'DC:A6:32:81:DA:6D', 'advertisement': AdvertisementData(local_name='140R00036D', manufacturer_data={0: b'\xff\xff\xff\xffd\x00\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fc60-0000-1000-8000-00805f9b34fb', '0000fe00-0000-1000-8000-00805f9b34fb', '0000ff90-0000-1000-8000-00805f9b34fb', '0000ffa0-0000-1000-8000-00805f9b34fb', '0000ffb0-0000-1000-8000-00805f9b34fb', '0000ffc0-0000-1000-8000-00805f9b34fb', '0000ffd0-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', '0000ffe5-0000-1000-8000-00805f9b34fb', '0000fff0-0000-1000-8000-00805f9b34fb'], tx_power=0, rssi=-57), 'device': BLEDevice(38:AB:41:1F:40:CE, 140R00036D), 'connectable': True, 'time': 82222.16297646, 'tx_power': 0}
2024-10-29 19:50:33.727 DEBUG (MainThread) [custom_components.bms_ble] 140R00036D: BMS data update
2024-10-29 19:50:33.728 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] Connecting BMS (140R00036D)
2024-10-29 19:50:34.767 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request voltage info
2024-10-29 19:50:34.826 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=11 size=4 sign=False
2024-10-29 19:50:35.827 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request current info
2024-10-29 19:50:35.876 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=11 size=4 sign=True
2024-10-29 19:50:36.878 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request temperature info
2024-10-29 19:50:36.926 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=9 size=2 sign=False
2024-10-29 19:50:37.927 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request battery_level info
2024-10-29 19:50:37.978 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=10 size=1 sign=False
2024-10-29 19:50:38.006 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) Rx BLE data: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00~\xfe\xff\xff\xb4\r\n')
2024-10-29 19:50:38.007 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) Rx data CRC is invalid: 0xb4 != 0x4b4
2024-10-29 19:50:38.979 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request design_capacity info
2024-10-29 19:50:39.026 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=21 size=2 sign=False
2024-10-29 19:50:40.028 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request cycles info
2024-10-29 19:50:40.076 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=21 size=2 sign=False
2024-10-29 19:50:41.078 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request runtime info
2024-10-29 19:50:41.126 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=12 size=2 sign=False
2024-10-29 19:50:42.128 DEBUG (MainThread) [custom_components.bms_ble] 140R00036D: no valid data received
2024-10-29 19:50:42.128 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching 140R00036D data in 8.401 seconds (success: False)
2024-10-29 19:50:43.978 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] Disconnected from BMS (140R00036D)
2024-10-29 19:50:56.405 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request voltage info
2024-10-29 19:50:56.464 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=11 size=4 sign=False
2024-10-29 19:50:57.465 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request current info
2024-10-29 19:50:57.514 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=11 size=4 sign=True
2024-10-29 19:50:58.516 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request temperature info
2024-10-29 19:50:58.564 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=9 size=2 sign=False
2024-10-29 19:50:59.565 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request battery_level info
2024-10-29 19:50:59.614 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=10 size=1 sign=False
[... cut ...]
2024-10-29 19:51:01.666 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request cycles info
2024-10-29 19:51:01.714 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=21 size=2 sign=False
2024-10-29 19:51:02.717 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request runtime info
2024-10-29 19:51:02.764 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=12 size=2 sign=False
2024-10-29 19:51:03.775 DEBUG (MainThread) [custom_components.bms_ble] 140R00036D: no valid data received
`

patman15, 2024-10-29:
> There is a typo:
> BLE data: bytearray(b'*_xaaU_*\x0b\x08\xf43\x00\x00\xd3\xfc\xff\xff\x07\r\n')

No, this is just a stupid representation. I think I can fix it, the CRC calculation is wrong. Minor issue again. One minute.

snipah, 2024-10-29:
Now setup ran through, but nearly no data... an 244% SOC

Batterie
244 %
 
Cycles
Unbekannt
 
Gespeicherte Energiemenge
Unbekannt
 
Ladestatus
Wird nicht geladen
 
Leistung
Unbekannt
 
Runtime
Unbekannt
 
Spannung
Unbekannt
 
Stromstärke
Unbekannt
 
Temperatur
Unbekannt

`2024-10-29 20:02:34.630 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JBCSJKKCFYXYMMC4BQ5RXWY2 version=1 domain=bms_ble title=140R00036D state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=38:AB:41:1F:40:CE>
2024-10-29 20:02:34.634 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for 140R00036D (38:AB:41:1F:40:CE) as CBT Power Smart BMS
2024-10-29 20:02:34.634 DEBUG (MainThread) [custom_components.bms_ble] device data: {'name': '140R00036D', 'address': '38:AB:41:1F:40:CE', 'rssi': -65, 'manufacturer_data': {0: b'\xff\xff\xff\xffd\x00\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, 'service_data': {}, 'service_uuids': ['0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fc60-0000-1000-8000-00805f9b34fb', '0000fe00-0000-1000-8000-00805f9b34fb', '0000ff90-0000-1000-8000-00805f9b34fb', '0000ffa0-0000-1000-8000-00805f9b34fb', '0000ffb0-0000-1000-8000-00805f9b34fb', '0000ffc0-0000-1000-8000-00805f9b34fb', '0000ffd0-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', '0000ffe5-0000-1000-8000-00805f9b34fb', '0000fff0-0000-1000-8000-00805f9b34fb'], 'source': 'DC:A6:32:81:DA:6D', 'advertisement': AdvertisementData(local_name='140R00036D', manufacturer_data={0: b'\xff\xff\xff\xffd\x00\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00'}, service_uuids=['0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000fc60-0000-1000-8000-00805f9b34fb', '0000fe00-0000-1000-8000-00805f9b34fb', '0000ff90-0000-1000-8000-00805f9b34fb', '0000ffa0-0000-1000-8000-00805f9b34fb', '0000ffb0-0000-1000-8000-00805f9b34fb', '0000ffc0-0000-1000-8000-00805f9b34fb', '0000ffd0-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', '0000ffe5-0000-1000-8000-00805f9b34fb', '0000fff0-0000-1000-8000-00805f9b34fb'], tx_power=0, rssi=-65), 'device': BLEDevice(38:AB:41:1F:40:CE, 140R00036D), 'connectable': True, 'time': 82943.18039508, 'tx_power': 0}
2024-10-29 20:02:34.635 DEBUG (MainThread) [custom_components.bms_ble] 140R00036D: BMS data update
2024-10-29 20:02:34.635 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] Connecting BMS (140R00036D)
2024-10-29 20:02:35.766 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request voltage info
2024-10-29 20:02:35.826 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=11 size=4 sign=False
2024-10-29 20:02:36.838 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request current info
2024-10-29 20:02:36.876 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=11 size=4 sign=True
2024-10-29 20:02:37.877 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request temperature info
2024-10-29 20:02:37.926 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=9 size=2 sign=False
2024-10-29 20:02:38.928 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request battery_level info
2024-10-29 20:02:38.979 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=10 size=1 sign=False
2024-10-29 20:02:39.011 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) Rx BLE data: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00\xa6\xfe\xff\xff\xdc\r\n')
2024-10-29 20:02:39.014 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) {10: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00\xa6\xfe\xff\xff\xdc\r\n')}
2024-10-29 20:02:40.067 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request design_capacity info
2024-10-29 20:02:40.116 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=21 size=2 sign=False
2024-10-29 20:02:41.118 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request cycles info
2024-10-29 20:02:41.166 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=21 size=2 sign=False
2024-10-29 20:02:42.168 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) request runtime info
2024-10-29 20:02:42.216 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (140R00036D) cmd=12 size=2 sign=False
2024-10-29 20:02:43.219 DEBUG (MainThread) [custom_components.bms_ble] BMS data sample {'battery_level': 244}
2024-10-29 20:02:43.219 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching 140R00036D data in 8.584 seconds (success: True)
2024-10-29 20:02:44.978 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] Disconnected from BMS (140R00036D)`

patman15, 2024-10-29:
Ok, now it is getting weird. Seems like the BMS is not answering to the requests except for one. Need to check tomorrow.

patman15, 2024-10-29:
@snipah just pushed a last fix for today. Some genius designed the protocol with different ending on send and receive, which I missed when programming. Sorry for the mess, was no (yet) caught by my tests.

[9 more replies]

closed 2024-11-05 by patman15