patman15/BMS_BLE-HA #60: Adding Creabest LiFePO4 battery

Alex-DeIt, 2024-09-30:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I can't see the BMS data of my Creabest batteries in my home assistant.

### Describe the solution you'd like

Would you please consider adding this battery ?

### Additional context

Unfortunately I do not have this links. I only know the app used on my smartphone for reading the BMS

snipah, 2024-09-30:
Duplicate to #59

patman15, 2024-10-01:
@Alex-DeIt this request will be tracked via #59 where all information about the battery is collected. If you have know-how to contribute, please join the discussion there.

closed 2024-10-01 by patman15

Alex-DeIt, 2024-10-01:
Ok. it's fine. I did not see that another discusion was already opened. Thank you patman15.