patman15/BMS_BLE-HA #61: Add support for Victron Shunt via BLE.

RoySalisbury, 2024-10-02:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

No problem, just a feature request.

### Describe the solution you'd like

Currently using BATMON with Victron Shunt.  However, I would like to us it via the BLE proxy and not have to have an actual bluetooth device on HA for this.

patman15, 2024-10-02:
The Victron shunt is not an BMS so I'm hesitant to add it. Can you check if https://github.com/keshavdv/victron-hacs/ works for you? It is the right device to be covered there. Let me know if this is fine. Thanks!

RoySalisbury, 2024-10-03:
Thats fine.  Was just trying to get BLE Proxy support.  I can continue to use BatMon for direct BLE connection.

patman15, 2024-10-03:
Sorry, but the Victron uses advertisements to provide the information, that is a different method and does not fit the architecture of this integration. Nevertheless, I'm pretty sure that https://github.com/keshavdv/victron-hacs/ does support ESPproxy as well although not explicitly stated.
Anyways, Batmon is fine, used it myself a lot before.

closed 2024-11-28 by github-actions[bot]