patman15/BMS_BLE-HA #63: Voltage for each cell.

herijaya, 2024-10-04:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

No.

### Describe the solution you'd like

I can't read the battery cell voltage, right?
It seems even better to read the battery voltage for each cell. Thank you my brother

herijaya, 2024-10-04:
my device : BMS Protection Home Energy Storage Smart Bms 16S 100A with 1A Active Balance

patman15, 2024-10-04:
> I can't read the battery cell voltage, right?

You can, please see the [README](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#provided-information):

Individual cell voltage are available as attribute to the `delta voltage` sensor. If you want to have them as individual sensors, please check the [FAQ](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#how-do-i-get-the-cell-voltages-as-individual-sensor-for-tracking).

danbi, 2024-10-28:
The debug log shows Daly returns voltages with three fractional digits, the interface displays only two. Maybe the full resolution should be processed.

2024-10-28 21:28:06.211 DEBUG (MainThread) [custom_components.bms_ble] BMS data 
sample {'temp#0': -40.0, 'voltage': 52.4, 'current': -6.1, 'battery_level': 63.6
, 'cycle_charge': 197.1, 'cell_count': 16, 'temp_sensors': 1, 'cycles': 329, 'de
lta_voltage': 0.004, 'temp#1': 24.0, 'cell#0': 3.278, 'cell#1': 3.279, 'cell#2':
 3.28, 'cell#3': 3.279, 'cell#4': 3.28, 'cell#5': 3.279, 'cell#6': 3.28, 'cell#7
': 3.279, 'cell#8': 3.281, 'cell#9': 3.28, 'cell#10': 3.28, 'cell#11': 3.279, 'c
ell#12': 3.28, 'cell#13': 3.28, 'cell#14': 3.28, 'cell#15': 3.279, 'cycle_capaci
ty': 10328.04, 'power': -319.64, 'battery_charging': False, 'runtime': 116321, '
temperature': -8.0}

patman15, 2024-10-29:
Full resolution is available, but HA for some reason does not show it.
If you have a look at the [developer tools](https://my.home-assistant.io/redirect/developer_states/) it's there:
[image]
You can also create individual sensors, see the FAQ on how to do that.

closed 2024-12-10 by github-actions[bot]