patman15/BMS_BLE-HA #64: Is it possible to switch off auto discovering of new batteries?

Karatzie, 2024-10-07:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

Hi Patrick.

I traveled with my camper van at the last weekend.
I had installed the BMS_BLE_HA on my Home Assistant system.
On the camping ground there was a lot of trafic with different camper vans.
Your integration generate always a new device is discover message if a new battery is scaned over Bluetooth.
I do not like to get this messages if a camper van with his batteries comes in the Bluetooth range of my RPI.
Is it possible to add a switch for activate and deactivate the discovery function for new batteries in your integration.

BR
Andreas

patman15, 2024-10-08:
You mean the notifications in the identically named entry on the sidebar that a _new device has been discovered_?
Hmm, this is a Home Assistant feature, I did a small research, it seems the configuration option to disable it has been removed.
Would this [automation](https://community.home-assistant.io/t/how-can-i-get-rid-of-new-devices-discovered-notifications/619675) probably solve your issue? Unfortunately, I can't test it at the moment.

Karatzie, 2024-10-16:
Hi Patrick.
I observed a new problem with the autodiscovering for new batteries.
I have a Bluetti AC200Max and an own integration for this device.
The integration is running and connected to the Bluetti.
The BMS_BLE_HA discovers the Bluetti as new device.
Is it possible to make a configuration witch battery should be used with BMS_BLE_HA?

BR
Andreas

patman15, 2024-10-16:
Hi Andreas!
Why is this behaviour a problem? It is the main purpose of auto-discovery to propose compatible devices. Simply click on _ignore_ in the device manager and it should not propose it again.
The discovered device is not used until you explicitly acknowledge it.
Otherwise please explain your issue with the behaviour, probably my thoughts go in the wrong direction.
Kind regards,
Patrick

Karatzie, 2024-10-17:
Hi Patrick.
Thank you for your quick response.
I want to explain my problem and the behavior of the HA system.

There is also a Home Assistant HACS Bluetti integration.
[https://github.com/Patrick762/hassio-bluetti-bt](url)
I would like to use this HACS integration in the future.
The Bluetti HACS integration does not work in the moment.
The old Bluetti ADDON is still running at the moment.
My old Bluetti ADDON dislays an error that a future running is not possible.
I got this error in the log.
```
**WARNING  filtered_messages() is deprecated and will be removed in a future version. Use messages() together with Topic.matches() instead.**
``` 
The old ADDON is not supported in the moment.
We got a lot off new bugs and the developer does not answer to the new bugs.
I would like to move to the HACS Bluetti integration if the integration is running again.

I am not realy shure what is happen if I ingnore the Bluetti for your integration.
Does the HACS  Bluetti integration find this ignored device in the future?
What is happen if I add my Bluetti to the HACS Bluetti integration.
Does your integration find the Buetti again?

From my point of view I would like to switch off the auto discovery in HA completely.
My HA sytem is installed in a mobile home.
If I visit a camping ground I got a lot of discovering messeges from HA.
HA discovers a lot of bluetooth sensors, electrical tooth-brushes, batteris, light controllers, s*x toys and so on.
I do not want to see this messages.

BR
Andreas

patman15, 2024-10-17:
After a bit of playing around I found the following. Have you tried, going to the Bluetooth integration and then
[image]
select _system options_ and then
[image]
deactivated _enable newly added devices_?
That should solve all your problems I think.

**In the unlikely case it does not**, you can also completely deactivate the _Bluetooth_ integration, previously added devices should still work. (At least on my system it worked)

And just for completeness, the answers to your questions:

> Does the HACS Bluetti integration find this ignored device in the future?

Yes, a deactivated "device" is an instance per integration. In case there are issues, you can still activate it again.

> What is happen if I add my Bluetti to the HACS Bluetti integration.
> Does your integration find the Buetti again?

Shouldn't be the case.

Maybe I add Bluetti to my integration as well. 😉

Karatzie, 2024-10-18:
Hi Patrick.
I will check this.

BR
Andreas

patman15, 2024-10-18:
Hi Andreas, thanks! I will next week fix that the integration wrongly detects the Bluetti as supported. Found a way for that.

closed 2024-11-10 by Karatzie