patman15/BMS_BLE-HA #70: Daly BMS displays temperature -8 degrees since 1.7.0

danbi, 2024-10-28:
### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).

### Describe the issue

After upgrading to 1.7.0 all my Daly BMS show temperature is -8 degrees.

### Reproduction steps

1. Upgrade to version 1.7.0
2.
3.
...

### Debug logs

```text
2024-10-28 21:19:30.852 DEBUG (MainThread) [custom_components.bms_ble] DL-401606012122: BMS data update
2024-10-28 21:19:30.853 DEBUG (MainThread) [custom_components.bms_ble.plugins.daly_bms] Connecting BMS (DL-401606012122)
2024-10-28 21:19:32.162 DEBUG (MainThread) [custom_components.bms_ble.plugins.daly_bms] Received BLE data: bytearray(b'\xd2\x03\x12\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02,')
2024-10-28 21:19:32.168 DEBUG (MainThread) [custom_components.bms_ble.plugins.daly_bms] DL-401606012012: MOS info: bytearray(b'\xd2\x03\x12\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02,')
2024-10-28 21:19:32.409 DEBUG (MainThread) [custom_components.bms_ble.plugins.daly_bms] Received BLE data: bytearray(b'\xd2\x03|\x0c\xd4\x0c\xd5\x0c\xd3\x0c\xd4\x0c\xd5\x0c\xd7\x0c\xd4\x0c\xd7\x0c\xd6\x0c\xd9\x0c\xd4\x0c\xd7\x0c\xd4\x0c\xd8\x0c\xd4\x0c\xd6\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\rt\xf1\x02x\x0c\xd8\x0c\xd1\x00@\x00@\x00\x02\x07\xe6\x00\x10\x00\x01\x01T\x00\x00\x00\x01\x00\x01\x0c\xd1\x00\x07\x01J\x00\x00\x00\x00\x00\x00\x00\x008\xef')
2024-10-28 21:19:32.411 DEBUG (MainThread) [custom_components.bms_ble] BMS data sample {'temp#0': -40.0, 'voltage': 52.5, 'current': -6.3, 'battery_level': 63.2, 'cycle_charge': 202.2, 'cell_count': 16, 'temp_sensors': 1, 'cycles': 340, 'delta_voltage': 0.007, 'temp#1': 24.0, 'cell#0': 3.284, 'cell#1': 3.285, 'cell#2': 3.283, 'cell#3': 3.284, 'cell#4': 3.285, 'cell#5': 3.287, 'cell#6': 3.284, 'cell#7': 3.287, 'cell#8': 3.286, 'cell#9': 3.289, 'cell#10': 3.284, 'cell#11': 3.287, 'cell#12': 3.284, 'cell#13': 3.288, 'cell#14': 3.284, 'cell#15': 3.286, 'cycle_capacity': 10615.5, 'power': -330.75, 'battery_charging': False, 'runtime': 115542, 'temperature': -8.0}
2024-10-28 21:19:32.412 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching DL-401606012012 data in 1.565 seconds (success: True)
2024-10-28 21:19:32.624 DEBUG (MainThread) [custom_components.bms_ble.plugins.daly_bms] Received BLE data: bytearray(b'\xd2\x03\x12\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02,')
2024-10-28 21:19:32.625 DEBUG (MainThread) [custom_components.bms_ble.plugins.daly_bms] DL-4016060121E1: MOS info: bytearray(b'\xd2\x03\x12\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02,')
2024-10-28 21:19:32.911 DEBUG (MainThread) [custom_components.bms_ble.plugins.daly_bms] Received BLE data: bytearray(b'\xd2\x03|\x0c\xd5\x0c\xd2\x0c\xd5\x0c\xd4\x0c\xd6\x0c\xd4\x0c\xd5\x0c\xd5\x0c\xd5\x0c\xd2\x0c\xd5\x0c\xd2\x0c\xd5\x0c\xd4\x0c\xd5\x0c\xd4\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00@\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\rt\xdd\x02q\x0c\xd6\x0c\xd3\x00@\x00@\x00\x02\x07\xd0\x00\x10\x00\x01\x01,\x00\x00\x00\x01\x00\x01\x0c\xd1\x00\x03\x01\xb3\x00\x00\x00\x00\x00\x00\x00\x00u\x93')
2024-10-28 21:19:32.913 DEBUG (MainThread) [custom_components.bms_ble] BMS data sample {'temp#0': -40.0, 'voltage': 52.5, 'current': -8.3, 'battery_level': 62.5, 'cycle_charge': 200.0, 'cell_count': 16, 'temp_sensors': 1, 'cycles': 300, 'delta_voltage': 0.003, 'temp#1': 24.0, 'cell#0': 3.285, 'cell#1': 3.282, 'cell#2': 3.285, 'cell#3': 3.284, 'cell#4': 3.286, 'cell#5': 3.284, 'cell#6': 3.285, 'cell#7': 3.285, 'cell#8': 3.285, 'cell#9': 3.282, 'cell#10': 3.285, 'cell#11': 3.282, 'cell#12': 3.285, 'cell#13': 3.284, 'cell#14': 3.285, 'cell#15': 3.284, 'cycle_capacity': 10500.0, 'power': -435.75, 'battery_charging': False, 'runtime': 86746, 'temperature': -8.0}
2024-10-28 21:19:32.914 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching DL-4016060121E1 data in 2.064 seconds (success: True)
```

patman15, 2024-10-29:
Thanks for reporting. Seems to be caused by then enhancement #55 because your BMS does not seem to have this sensor. Will fix that soon.

patman15, 2024-10-29:
@danbi are you able to check the fix on the [branch](https://github.com/patman15/BMS_BLE-HA/tree/70-daly-bms-displays-temperature-8-degrees-since-170)? Basically you would need to follow the [instructions for manual installation](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual) except for step 7.

closed 2024-10-31 by patman15

danbi, 2024-11-03:
Worked fine for me.

patman15, 2024-11-03:
Thanks for confirming!

commit 5b2e1430 (2024-10-29): fix #70
PR #72 (2024-10-31): 70 daly bms displays wrong temperature since v1.7.0
fixes #70
commit 34b50591 (2024-10-31): 70 daly bms displays wrong temperature since v1.7.0 (#72)

* fix #70
* adapted tests for BMS w, w/o MOS temp