patman15/BMS_BLE-HA #77: Pairing with one battery is fails continously

Alex-DeIt, 2024-11-05:
### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).

### Describe the issue

The pairing process with one of my two batteries is not working properly. Pls. see here below the extract from the HA terminal:

~ $ bluetoothctl pair BA:03:EC:11:0C:F8
Attempting to pair with BA:03:EC:11:0C:F8
hci0 type 6 discovering off
hci0 BA:03:EC:11:0C:F8 type LE Public connected eir_len 25
[CHG] Device BA:03:EC:11:0C:F8 Connected: yes
hci0 BA:03:EC:11:0C:F8 type LE Public disconnected with reason 1
Failed to pair: org.bluez.Error.AuthenticationTimeout
~ $ H

Attention: in the log file below is indicated the battery which could be paired (170R000121), not the battery which I cannot pair with Hassio.

### Reproduction steps

1.I tried to pair one battery inserting in the HA terminal "bluetoothctl pair BA:03:EC:11:0C:F8"
2.
3.
...

### Debug logs

```text
~ $ tail -f /config/home-assistant.log
2024-11-05 18:41:53.787 INFO (MainThread) [homeassistant.components.automation.nuova_automazione] Initialized trigger Avviso movimento interno TZE 2
2024-11-05 18:41:53.790 INFO (MainThread) [homeassistant.components.automation.avviso_movimento_interno_tze_3] Initialized trigger Avviso movimento interno TZE 3
2024-11-05 18:41:53.829 INFO (SyncWorker_5) [homeassistant.loader] Loaded kegtron from homeassistant.components.kegtron
2024-11-05 18:41:54.326 INFO (SyncWorker_1) [homeassistant.loader] Loaded zha from homeassistant.components.zha
2024-11-05 18:41:59.124 INFO (MainThread) [homeassistant.components.mqtt.client] MQTT client initialized, birth message sent
2024-11-05 18:42:03.620 INFO (ImportExecutor_0) [zigpy.serial] Using pyserial-asyncio-fast in place of pyserial-asyncio
2024-11-05 18:42:26.987 ERROR (MainThread) [custom_components.bms_ble] Error fetching 170R000121 data: device communicating failed: failed to discover services, device disconnected (BleakError)
2024-11-05 18:43:42.171 INFO (MainThread) [custom_components.bms_ble] Fetching 170R000121 data recovered
2024-11-05 18:46:32.699 ERROR (MainThread) [custom_components.bms_ble] Error fetching 170R000121 data: device communicating failed: failed to discover services, device disconnected (BleakError)
2024-11-05 18:47:12.596 INFO (MainThread) [custom_components.bms_ble] Fetching 170R000121 data recovered
```

Alex-DeIt, 2024-11-05:
I apologize, but I am not an expert with Hassio. Please find below the log (tried to restart Hassio and pair the battery like explained before):

2024-11-05 19:01:33.697 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration bms_ble which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-05 19:01:33.701 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-11-05 19:01:36.245 INFO (MainThread) [homeassistant.setup] Setup of domain system_log took 0.51 seconds
2024-11-05 19:01:36.246 INFO (MainThread) [homeassistant.setup] Setup of domain logger took 0.51 seconds
2024-11-05 19:01:36.246 INFO (MainThread) [homeassistant.bootstrap] Setting up frontend: {'frontend'}
2024-11-05 19:01:36.248 INFO (MainThread) [homeassistant.setup] Setting up http
2024-11-05 19:01:36.249 INFO (MainThread) [homeassistant.setup] Setting up device_automation
2024-11-05 19:01:36.250 INFO (MainThread) [homeassistant.setup] Setup of domain device_automation took 0.00 seconds
2024-11-05 19:01:36.288 INFO (MainThread) [homeassistant.setup] Setup of domain http took 0.04 seconds
2024-11-05 19:01:36.290 INFO (MainThread) [homeassistant.setup] Setting up auth
2024-11-05 19:01:36.296 INFO (MainThread) [homeassistant.setup] Setup of domain auth took 0.01 seconds
2024-11-05 19:01:36.297 INFO (MainThread) [homeassistant.setup] Setting up config
2024-11-05 19:01:36.310 INFO (MainThread) [homeassistant.setup] Setup of domain config took 0.01 seconds
2024-11-05 19:01:36.311 INFO (MainThread) [homeassistant.setup] Setting up diagnostics
2024-11-05 19:01:36.315 INFO (MainThread) [homeassistant.setup] Setup of domain diagnostics took 0.00 seconds
2024-11-05 19:01:36.316 INFO (MainThread) [homeassistant.setup] Setting up file_upload
2024-11-05 19:01:36.317 INFO (MainThread) [homeassistant.setup] Setup of domain file_upload took 0.00 seconds
2024-11-05 19:01:36.318 INFO (MainThread) [homeassistant.setup] Setting up image_upload
2024-11-05 19:01:36.324 INFO (MainThread) [homeassistant.setup] Setup of domain image_upload took 0.01 seconds
2024-11-05 19:01:36.325 INFO (MainThread) [homeassistant.setup] Setting up repairs
2024-11-05 19:01:36.327 INFO (MainThread) [homeassistant.setup] Setup of domain repairs took 0.00 seconds
2024-11-05 19:01:36.328 INFO (MainThread) [homeassistant.setup] Setting up websocket_api
2024-11-05 19:01:36.330 INFO (MainThread) [homeassistant.setup] Setup of domain websocket_api took 0.00 seconds
2024-11-05 19:01:36.331 INFO (MainThread) [homeassistant.setup] Setting up api
2024-11-05 19:01:36.337 INFO (MainThread) [homeassistant.setup] Setup of domain api took 0.01 seconds
2024-11-05 19:01:36.338 INFO (MainThread) [homeassistant.setup] Setting up person
2024-11-05 19:01:36.340 INFO (MainThread) [homeassistant.setup] Setting up search
2024-11-05 19:01:36.341 INFO (MainThread) [homeassistant.setup] Setup of domain search took 0.00 seconds
2024-11-05 19:01:36.346 INFO (MainThread) [homeassistant.setup] Setup of domain person took 0.01 seconds
2024-11-05 19:01:36.348 INFO (MainThread) [homeassistant.setup] Setting up onboarding
2024-11-05 19:01:36.348 INFO (MainThread) [homeassistant.setup] Setup of domain onboarding took 0.00 seconds
2024-11-05 19:01:36.349 INFO (MainThread) [homeassistant.setup] Setting up lovelace
2024-11-05 19:01:36.356 INFO (MainThread) [homeassistant.setup] Setup of domain lovelace took 0.01 seconds
2024-11-05 19:01:36.357 INFO (MainThread) [homeassistant.setup] Setting up frontend
2024-11-05 19:01:36.392 INFO (MainThread) [homeassistant.setup] Setup of domain frontend took 0.03 seconds
2024-11-05 19:01:36.394 INFO (MainThread) [homeassistant.bootstrap] Setting up recorder: {'recorder'}
2024-11-05 19:01:36.396 INFO (MainThread) [homeassistant.setup] Setting up recorder
2024-11-05 19:01:36.412 INFO (MainThread) [homeassistant.components.http] Now listening on port 8123
2024-11-05 19:01:36.416 INFO (MainThread) [homeassistant.setup] Setup of domain http took 0.02 seconds
2024-11-05 19:01:36.539 INFO (MainThread) [homeassistant.setup] Setup of domain recorder took 0.14 seconds
2024-11-05 19:01:36.540 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 1: {'usb', 'hassio', 'http', 'dhcp', 'cloud', 'websocket_api', 'bluetooth', 'zeroconf', 'ssdp', 'api', 'webhook', 'auth', 'repairs', 'network'}
2024-11-05 19:01:36.544 INFO (MainThread) [homeassistant.setup] Setting up hassio
2024-11-05 19:01:36.553 INFO (MainThread) [homeassistant.setup] Setting up webhook
2024-11-05 19:01:36.556 INFO (MainThread) [homeassistant.setup] Setup of domain webhook took 0.00 seconds
2024-11-05 19:01:36.559 INFO (MainThread) [homeassistant.setup] Setting up network
2024-11-05 19:01:36.568 INFO (MainThread) [homeassistant.setup] Setup of domain network took 0.01 seconds
2024-11-05 19:01:36.652 INFO (MainThread) [homeassistant.setup] Setting up usb
2024-11-05 19:01:36.653 INFO (MainThread) [homeassistant.setup] Setup of domain usb took 0.00 seconds
2024-11-05 19:01:36.708 INFO (MainThread) [homeassistant.setup] Setup of domain hassio took 0.16 seconds
2024-11-05 19:01:36.824 INFO (MainThread) [homeassistant.setup] Setting up binary_sensor
[... cut ...]
2024-11-05 19:02:58.348 INFO (SyncWorker_1) [homeassistant.loader] Loaded led_ble from homeassistant.components.led_ble
2024-11-05 19:02:58.354 INFO (SyncWorker_1) [homeassistant.loader] Loaded xiaomi_ble from homeassistant.components.xiaomi_ble
2024-11-05 19:02:58.357 INFO (SyncWorker_1) [homeassistant.loader] Loaded ruuvitag_ble from homeassistant.components.ruuvitag_ble
2024-11-05 19:03:10.732 ERROR (MainThread) [custom_components.bms_ble] Timeout fetching 170R000121 data
2024-11-05 19:03:48.712 INFO (MainThread) [custom_components.bms_ble] Fetching 170R000121 data recovered

patman15, 2024-11-05:
Issues with pairing are issues with the Bluetooth stack, not necessarily this integration. Can you please provide the output of `bluetoothctl devices` and additionally `bluetoothctl info <MAC>` for both batteries?
Are you using an ESPproxy and if not, which BT adapter do you use?

patman15, 2024-11-06:
Hi Alexander!

> Device BA:03:EC:11:0C:B4 (public)
> Name: 170R000121
> Alias: 170R000121
> Paired: no

Are you sure we are talking about pairing? Even the battery that you say **is** paired, is reported as **not**. From the indications I see, I would guess that you mean it is not detected (showing up as new device in HA)?

- Are you using ESPHome BT proxy?
- Are you sure, you closed all connections (== apps) on your mobile? Some BMSs do not support parallel connections.
- Does the device show up in the [integration settings](https://my.home-assistant.io/redirect/integration/?domain=bms_ble)?
  - If yes, is there an error reported? Which?
- Can you remove all devices from the [integration settings](https://my.home-assistant.io/redirect/integration/?domain=bms_ble), enable debug logging, then re-add them, and stop logging and **attach** the log file to your comment?
- Did you try `bluetoothctl connect BA:03:EC:11:0C:F8` and see what happens?

Sorry for the many questions, but there is some Bluetooth issue ongoing, which is hard to find out remotely. To me it looks like, Bluetooth is not able to query the services from the device, which is indicated by the second device not showing any UUIDs. The latter are required for the device to be detected.

Alex-DeIt, 2024-11-06:
Hello Patrick,

pls. find here below my answers to your questions:

_Are you sure we are talking about pairing?_ --> I would say yes. This is what i get when I start the pairing command:
[image]
The output is "Failed to pair...", so I woul say that there is a pairing issiue, isn't it?

Even the battery that you say is paired, is reported as not. --> I do not know, Patrick. As I sarted the pairing comman with the other battery I did not receive this error message and the battery was recognized by Hassio and I have the data on its dashbord (see below, pls.)
[image]

_From the indications I see, I would guess that you mean it is not detected (showing up as new device in HA)?_ --> I thought that a detection was only possible after a pairing. But, you are right, it is not detected as well.

_Are you using ESPHome BT proxy? Which BT adapter are you using (check dmesg | grep Bluetooth)?_ --> i do not know. When I start te suggested command I get following:
[image]

_Are you sure, you closed all connections (== apps) on your mobile? Some BMSs do not support parallel connections._ --> yes, absolutely. My cellular phone was far away from the batteries and no conection was running.

_Does the device show up in the [integration settings](https://my.home-assistant.io/redirect/integration/?domain=bms_ble)?_ --> I only have one device in integrations (see below, pls.):
[image]
[image]
This is the battery which could be paired.

_Can you remove all devices from the [integration settings](https://my.home-assistant.io/redirect/integration/?domain=bms_ble), enable debug logging, then re-add them, and stop logging and attach the log file to your comment?_ --> pls. find the logfile in attachment
[home-assistant_bms_ble_2024-11-06T15-11-48.876Z.log](https://github.com/user-attachments/files/17649075/home-assistant_bms_ble_2024-11-06T15-11-48.876Z.log)

_Did you try bluetoothctl connect BA:03:EC:11:0C:F8 and see what happens?_ --> this is the result when I start te command:
[image]

patman15, 2024-11-06:
> Are you using ESPHome BT proxy? Which BT adapter are you using (check dmesg | grep Bluetooth)? --> i do not know. When I start te suggested command I get following:

That unfortunately did not provide the information that I was looking for. Can you check the [Bluetooth integrations settings](https://my.home-assistant.io/redirect/integration/?domain=bluetooth) send a screenshot of those? It should tell your adapter.

> find the logfile in attachment

I looked at it and I can see a lot of error messages "[org.bluez.Error.Failed] Software caused connection abort (BleakDBusError)" from the underlying Bluetooth system. Which is similar to the output you provided last (which tries to connect to your 2nd battery without using the integration). Looks like there is an issue with your adapter.

Just to be sure, could you please check the output of `bluetoothctl devices Connected` ? (Be sure that the **C** is capital!)

Alex-DeIt, 2024-11-06:
Hello Patrick, thank you for your patience.
Do you think this?
[image]

And this is the result of the suggested command:
[image]

Thank you again :)

Alex-DeIt, 2024-11-06:
So I tried adding the command a few more times:
[image]

patman15, 2024-11-06:
Hi! Ok, from your tries I can see that even the connection to the "working" battery is constantly lost. Did you enable the diagnostic sensors for link quality / RSSI, can you show the graph?
I would guess that you have an issue with your Bluetooth adapter being low quality. If link quality/RSSI looks bad and you are sure that the batteries are close enough to your HA system, you should consider buying an adapter listed as [known-working high-performance adapters](https://www.home-assistant.io/integrations/bluetooth#known-working-high-performance-adapters). I noticed that I have similar issues (only one out of multiple batteries showing up) with my Realtek adapter while my internal Intel one, works flawless. Thus, I ordered another BT dongle based on the CSR8510A10 chipset.

Alex-DeIt, 2024-11-06:
_Did you enable the diagnostic sensors for link quality / RSSI, can you show the graph?_ --> No, it was disabled. I have enabled it just now. I will provide you the graph tomorrow if it is enough for you. Related to the workin battery the quality seems to be approximately 73%.

Thus, I ordered another BT dongle based on the CSR8510A10 chipset. --> Then I will try the same.

Alex-DeIt, 2024-11-07:
Hello Patrick, pls. find here the requested graph.
[image]

patman15, 2024-11-07:
Doesn't look too good. Let's see if the other dongle helps. I'll get mine hopefully tomorrow.

Alex-DeIt, 2024-11-08:
Hello Patrick, I have made some more tests. Pls. find here below a short summary:
1) Another BT dongle based on the CSR8510A10 chipset did not solve the problem. Hassio was still detecting only the 121 battery.
2) Turning off the 121 battery, Hassio was able to detect the 086 battery and I was able to proceed with its configuration,
3) Turning on at this point the 121 again, I was able to have the data related to both batteries (see here below).
[image]
4) The only strange think that I can notice now is that both batteries seem to disconnect and reconnect randomly (saee here below). Based on your experience, is it possible that the batteries transmit the data intermittently?
[image]

patman15, 2024-11-08:
> The only strange think that I can notice now is that both batteries seem to disconnect and reconnect randomly (saee here below). Based on your experience, is it possible that the batteries transmit the data intermittently?

Yes that can happen, I think there can be multiple reasons or combinations that trigger that. The fact that you were able to add the second battery when turning off the first, strongly hints towards an issue with the Bluetooth system.

Did you ever check the RSSI sensor for both batteries (have a look at the graph)? In [this post](https://github.com/patman15/BMS_BLE-HA/issues/77#issuecomment-2458801413) the BT system reports RSSI -84 and -91 which is to my experience really on the edge, see the README about the sensors. I think you should reposition the batteries / receiver so that you can achieve stable RSSI below above -80 better -70. Check that the batteries are not too close together, get the receiver closer, ...

Alex-DeIt, 2024-11-08:
This are the RSSI graphs related to the batteries:
[image]
[image]
Unfortunately the batteries are very close together and I cannot relocate them as they are installed into a Motorhome. I wil try to move the receiver closer. Any way I solved with following graph. So I can see the charging level of them even if they are temporary disconnected.
[image]
I will you keep informed.

I would like to express you once again my gratitude for all the efforts and the support. I owe you one.

patman15, 2024-11-09:
@sp0ngeboobs ok, that is a very good hint, thanks! I'll have a look at what changed. Probably it is an issue that HA allows parallel updates while the Bluetooth library I think has issues with that. I'll do some experiments and come back.

patman15, 2024-11-09:
@sp0ngeboobs do you know which version was the first that changed the behavior for you? I just checked 1.7.0 against 1.6.4 and can't spot anything related. My guess would be 1.7.0 is still fine and 1.7.1 and 1.8.0 have some issue? Can you confirm?

patman15, 2024-11-09:
Ok, I played around with 3 devices, leanings from today:
- distance / RSSI is key. If values are ~75dBm then it works
- 1 device is very stable, going up to 2 or 3 significantly reduces stability of the connections, fixing by going close
- I think the reason why < v1.7.0 worked better is, that is always generated a new connection client if connection was lost. This was due to a bug that had other negative effects. So maybe taking up this strategy will help. Need to think about it.
- So far I do not see a bug in the integration, it is how you need to treat the flaky Bluetooth system that is used in case connection quality is "bad.
- It's really hard to determine the influencing factors when changing the code, because a lot of caching / buffering is involved resulting in fast establishment of connections, sometimes after it worked once.

I'll have more experiments to figure out which strategy improves the situation, aside from moving devices closer to the receiver.

patman15, 2024-11-09:
> I tried to take screenshots of the bluetooth proxy logs. or provide any more logs?

I don't think it's related to the proxy, but in order to be useful, I would need debug logs from the integration in parallel.

patman15, 2024-11-10:
@sp0ngeboobs, @Alex-DeIt can you try the code from this [branch](https://github.com/patman15/BMS_BLE-HA/tree/feature/retry_connector)? At least for me it works significantly better. Happy to hear about any feedback.

Alex-DeIt, 2024-11-10:
Sure Patrick, but how? Must I install the complete branch as new or is it eough to copy paste the files of "custom_components" into my "custom_components/bms_ble/ directory"?

Alex-DeIt, 2024-11-10:
Is it ok to simply exchange the standard basebms.py file with the "new" one of this branch and restar Hassio?

patman15, 2024-11-10:
@Alex-DeIt you need to exchange all files (safety first) of `custom_components/bms_ble/`, see [manual installation](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual) (since you already have the directory, you just need to replace files). Please make sure that you replace both, the `custom_components/bms_ble/*` and `custom_components/bms_ble/plugins/*`.

Alex-DeIt, 2024-11-10:
Hello Patrick, yes, I can confirm. It seems to be much more stable now. Battery 121 has not disconnected during the last 30 minutes and battery 086 only 2 times for 30 to 40 seconds each time. I will observe it until tomorrow an let you know.
[image]

patman15, 2024-11-10:
Thanks! Sounds promising! :smiley: If you are good, please provide a graph of the `link quality` and `RSSI` sensors. Thanks!

Alex-DeIt, 2024-11-10:
So, Patrick, pls. find here the graphs related to the two batteries. the 121 has had no interuptions during the last 2 hours, while it seems that the 086 has had 13 interruptions. Each interruption was between 36 and 50 sec. long.
[image]

Alex-DeIt, 2024-11-10:
It seems that there is a connection interruption (or Hassio is simply not able to show the values) every time when a quality drop happens (see red contious lines in the graph below). There is only one interruption with no quality drop (see red dashed line).
[image]

patman15, 2024-11-10:
> It seems that there is a connection interruption (or Hassio is simply not able to show the values) every time when a quality drop happens

It's the other way round, every time the integration is not able to retrieve data, the sensor degrades. See definition of the sensor. That's why looking at the sensor is much easier, especially for longer periods, and the reason I integrated it.

Alex-DeIt, 2024-11-10:
Does it mean that Hassio is now showing calculated and not actual data during the periods in which it is not able to retrieve them?

patman15, 2024-11-10:
No, why do you think so?

Alex-DeIt, 2024-11-10:
May be I understood the word "integrated" in the wrong way, Patrick.
But why is Hassio now able tho show the values with less and shorter interruptions?

patman15, 2024-11-10:
> May be I understood the word "integrated" in the wrong way, Patrick.

Aaah! :smile: 

> But why is Hassio now able tho show the values with less and shorter interruptions?

I added the bleak-retry-connector library, which (in case the Bluetooth connection is bad), each update interval retries a few times before giving up. It's used also by other HA components.

patman15, 2024-11-11:
@Alex-DeIt thanks for the support, I will prepare a release, which will close this issue. Feel free to open a new one in case you have troubles!

Alex-DeIt, 2024-11-11:
Hallo Patrick, I have to say thank you for all your efforts.
Unfortunately I had to notice other issues. Hassio stopped the registration of the data related to the 121 battery at 02:30 AM and is stil not working (see below pls.)
[image]
while the data of the 086 battery were registered, even if with some interruptions (see here below).
[image]
Here below is the RSSI of the not working battery.
[image]
It dropped down at 02:30 AM and since then no registration is available.
And here is the link quality.
[image]

Alex-DeIt, 2024-11-11:
So, restarting Hassio it seems to work again for both batteries.
[image]

Alex-DeIt, 2024-11-11:
Do you mean this as Kernel log?

2024-11-11 02:00:29.335 ERROR (MainThread) [custom_components.bms_ble] Error fetching 170R000086 data: device communicating failed: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Unlikely Error) (BleakDBusError)
2024-11-11 02:01:22.073 INFO (MainThread) [custom_components.bms_ble] Fetching 170R000086 data recovered
2024-11-11 02:03:06.582 ERROR (MainThread) [custom_components.bms_ble] Error fetching 170R000086 data: device communicating failed: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Unlikely Error) (BleakDBusError)
2024-11-11 02:03:54.891 INFO (MainThread) [custom_components.bms_ble] Fetching 170R000086 data recovered
2024-11-11 02:14:12.033 INFO (MainThread) [telegram.ext.Updater] Flood control exceeded. Retry in 5 seconds
2024-11-11 02:33:33.024 WARNING (MainThread) [bleak.backends.bluezdbus.client] Failed to cancel connection (/org/bluez/hci0/dev_BA_03_EC_11_0C_F8): 
2024-11-11 02:33:33.031 ERROR (MainThread) [custom_components.bms_ble] Error fetching 170R000121 data: device communicating failed: [org.bluez.Error.Failed] Not connected (BleakDBusError)
2024-11-11 02:49:10.840 ERROR (MainThread) [custom_components.bms_ble] Error fetching 170R000086 data: device communicating failed: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Unlikely Error) (BleakDBusError)
2024-11-11 02:49:45.356 INFO (MainThread) [custom_components.bms_ble] Fetching 170R000086 data recovered
2024-11-11 02:51:31.267 ERROR (MainThread) [custom_components.bms_ble] Error fetching 170R000086 data: device communicating failed: Service Discovery has not been performed yet (BleakError)
2024-11-11 02:53:22.446 INFO (MainThread) [habluetooth.scanner] hci0 (B8:2D:28:06:62:FB): Bluetooth scanner has gone quiet for 90s, restarting
2024-11-11 02:53:22.449 ERROR (MainThread) [habluetooth.scanner] hci0 (B8:2D:28:06:62:FB): Error stopping scanner: [org.bluez.Error.Failed] No discovery started
2024-11-11 02:53:36.441 INFO (MainThread) [custom_components.bms_ble] Fetching 170R000086 data recovered
2024-11-11 02:53:52.338 INFO (MainThread) [custom_components.hacs] Loading known repositories
2024-11-11 04:04:50.708 ERROR (MainThread) [custom_components.bms_ble] Error fetching 170R000086 data: device communicating failed: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Unlikely Error) (BleakDBusError)

Unfortunately I do not have a debug from 02:00 to 03:00. I started it now. Let's see what happens.

I apologize, Patrick, for causing this issues.

patman15, 2024-11-11:
Never saw such an issue before, so I'm not sure what happened. But the message hints to a rare error in the Linux Bluetooth stack.
Did you check the Kernel logs?  `dmesg -H` ?

Alex-DeIt, 2024-11-11:
Here the Kernel log

2024-11-11 02:00:03.226 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:00:03.594 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:00:03.834 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:00:04.060 Khadas systemd[1]: Starting system activity accounting tool...
2024-11-11 02:00:04.086 Khadas systemd[1]: sysstat-collect.service: Succeeded.
2024-11-11 02:00:04.089 Khadas systemd[1]: Finished system activity accounting tool.
2024-11-11 02:00:15.082 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:00:17.434 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:00:20.790 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:00:22.778 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:00:31.614 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:00:32.994 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:00:43.222 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:00:44.206 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:00:44.642 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:01:01.186 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:01:01.762 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:01:11.814 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:01:14.210 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:01:22.578 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:01:22.622 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:01:38.750 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:01:42.182 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:01:49.790 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:01:50.550 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:01:59.958 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:04.254 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:09.802 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:12.046 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:13.430 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:22.614 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:22.674 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:33.114 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:34.426 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:44.614 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:44.678 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:55.850 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:02:56.114 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:06.674 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:09.374 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:09.750 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:20.762 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:21.594 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:31.758 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:33.834 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:42.610 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:42.774 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:53.794 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:55.714 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:59.790 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:03:59.954 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:04:10.610 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:04:11.010 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:04:21.010 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:04:21.626 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:04:31.310 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:04:33.554 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:04:34.870 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:04:43.650 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:04:46.206 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:04:54.370 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:04:54.858 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:05:05.682 Khadas kernel: Bluetooth: hci0: advertising data len corrected
[... cut ...]
2024-11-11 02:59:25.534 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:59:35.150 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:59:36.390 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:59:46.786 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:59:49.114 Khadas kernel: Bluetooth: hci0: advertising data len corrected
2024-11-11 02:59:57.590 Khadas kernel: Bluetooth: hci0: advertising data len corrected

I'm not able to recognize any special event.

patman15, 2024-11-11:
> I'm not able to recognize any special event.

True, neither can I.

Still looking at your log from the integration:
```
2024-11-11 02:33:33.024 WARNING (MainThread) [bleak.backends.bluezdbus.client] Failed to cancel connection (/org/bluez/hci0/dev_BA_03_EC_11_0C_F8):
2024-11-11 02:33:33.031 ERROR (MainThread) [custom_components.bms_ble] Error fetching 170R000121 data: device communicating failed: [org.bluez.Error.Failed] Not connected (BleakDBusError)
2024-11-11 02:49:10.840 ERROR (MainThread) [custom_components.bms_ble] Error fetching 170R000086 data: device communicating failed: [org.bluez.Error.Failed] Operation failed with ATT error: 0x0e (Unlikely Error) (BleakDBusError)
```
it seems that the Bluetooth stack crashed. This is outside the control of the integration.
I can see you still using the Khadas Bluetooth controller. Did you consider switching to the new one based on the CSR8510A10 chipset? Maybe that one is more stable?
I fear I can't do much about this issue ...

Alex-DeIt, 2024-11-11:
I tried it 2 days ago with a CSR8510A10 based bluetooth dongle, but it was not better. Even with this I had a lot of interruptions. I am thinking about trying with a Raspberry based Hassio instead of the Khadas based. What do you think?

patman15, 2024-11-11:
As far as I understood you, the new version significantly improved the availability, correct?
The issue last night looks to me like a crash of the Bluetooth stack/dongle. So switching that one could help.
With your RSSI values I think it's as good as it can get. You will always have drop outs. They last at least 30 seconds (or multiple of them) since this is the polling interval.
RPi is not known for its good Bluetooth performance, see https://www.home-assistant.io/integrations/bluetooth/#cypress-based-adapters so I would not recommend switching to that.

Alex-DeIt, 2024-11-11:
As far as I understood you, the new version significantly improved the availability, correct? --> yes, much better than before.

OK, Patrick, then I will try again with the CSR8510A10 based dongle.
Did you receive yours? Is your system performing better with it?

patman15, 2024-11-11:
> Did you receive yours? Is your system performing better with it?

Yes, I did receive mine, but I had no issues even before. My development system had a Realtek device which seems to be ok. I have tried the CSR8510A10 dongle but the change in 1.9.0 made much more impact. I would need to do more intensive tests to really see a difference.
But, I never had the BT system being stuck like you experienced it.

Alex-DeIt, 2024-11-12:
Hello Patrick, can I disturb you a last time?
I had again a longer interruption related to one of the two batteties which started at 14:05:11.
[image]
This time a tried not to restart the whole Hassio but only the integration related to the blocked battery switching the button indicated below with the blue arrow.
[image]
After that the data were shown again correctly.
Pls. find here below some lines of the debug related to a time slot containing the moment of connection loss.
Would be great if you could give a look one time more.

2024-11-12 14:03:43.627 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) Rx BLE data: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00\x00\x00\x00\x00:\r\n')
2024-11-12 14:03:43.629 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) {11: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00\x00\x00\x00\x00:\r\n')}
2024-11-12 14:03:43.629 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) request current info
2024-11-12 14:03:43.629 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) {11: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00\x00\x00\x00\x00:\r\n')}
2024-11-12 14:03:43.630 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) request temperature info
2024-11-12 14:03:43.750 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] Disconnected from BMS (170R000086)
2024-11-12 14:03:43.872 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) Rx BLE data: bytearray(b'\xaaU\t\x0c\x03\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1b\r\n')
2024-11-12 14:03:43.872 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) {11: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00\x00\x00\x00\x00:\r\n'), 9: bytearray(b'\xaaU\t\x0c\x03\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1b\r\n')}
2024-11-12 14:03:43.873 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) request battery_level info
2024-11-12 14:03:44.164 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) Rx BLE data: bytearray(b'\xaaU\n\x06d\x13\r\x00\x00\x00\x94\r\n')
2024-11-12 14:03:44.165 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) {11: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00\x00\x00\x00\x00:\r\n'), 9: bytearray(b'\xaaU\t\x0c\x03\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1b\r\n'), 10: bytearray(b'\xaaU\n\x06d\x13\r\x00\x00\x00\x94\r\n')}
2024-11-12 14:03:44.165 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) request design_capacity info
2024-11-12 14:03:44.262 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) Rx BLE data: bytearray(b'\xaaU\x15\x04\xc8\x00\x05\x00\xe6\r\n')
2024-11-12 14:03:44.263 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) {11: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00\x00\x00\x00\x00:\r\n'), 9: bytearray(b'\xaaU\t\x0c\x03\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1b\r\n'), 10: bytearray(b'\xaaU\n\x06d\x13\r\x00\x00\x00\x94\r\n'), 21: bytearray(b'\xaaU\x15\x04\xc8\x00\x05\x00\xe6\r\n')}
2024-11-12 14:03:44.263 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) request cycles info
2024-11-12 14:03:44.263 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) {11: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00\x00\x00\x00\x00:\r\n'), 9: bytearray(b'\xaaU\t\x0c\x03\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1b\r\n'), 10: bytearray(b'\xaaU\n\x06d\x13\r\x00\x00\x00\x94\r\n'), 21: bytearray(b'\xaaU\x15\x04\xc8\x00\x05\x00\xe6\r\n')}
2024-11-12 14:03:44.264 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) request runtime info
2024-11-12 14:03:44.505 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) Rx BLE data: bytearray(b'\xaaU\x0c\x0c&\x00\x00\x00\xf1H\x00\x00\x05\x00\x00\x00|\r\n')
2024-11-12 14:03:44.507 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) {11: bytearray(b'\xaaU\x0b\x08\xf43\x00\x00\x00\x00\x00\x00:\r\n'), 9: bytearray(b'\xaaU\t\x0c\x03\x00\x03\x00\x00\x00\x00\x00\x00\x00\x00\x00\x1b\r\n'), 10: bytearray(b'\xaaU\n\x06d\x13\r\x00\x00\x00\x94\r\n'), 21: bytearray(b'\xaaU\x15\x04\xc8\x00\x05\x00\xe6\r\n'), 12: bytearray(b'\xaaU\x0c\x0c&\x00\x00\x00\xf1H\x00\x00\x05\x00\x00\x00|\r\n')}
2024-11-12 14:03:44.651 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] (170R000121) Rx BLE data: bytearray(b'\xaaU\x05\n\xfe\x0c\xfe\x0c\xfd\x0c\xfd\x0c\x00\x005\r\n')
2024-11-12 14:03:44.653 DEBUG (MainThread) [custom_components.bms_ble] BMS data sample {'voltage': 13.3, 'current': 0.0, 'temperature': 3, 'battery_level': 100, 'design_capacity': 200, 'cycles': 5, 'cell#0': 3.326, 'cell#1': 3.326, 'cell#2': 3.325, 'cell#3': 3.325, 'cycle_charge': 200.0, 'cycle_capacity': 2660.0, 'power': 0.0, 'battery_charging': False, 'delta_voltage': 0.001}
2024-11-12 14:03:44.654 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching 170R000121 data in 3.808 seconds (success: True)
2024-11-12 14:03:47.227 DEBUG (MainThread) [custom_components.bms_ble] 170R000121: RSSI value: -92 dBm
2024-11-12 14:03:47.229 DEBUG (MainThread) [custom_components.bms_ble] 170R000121: Link quality: 100 %
2024-11-12 14:03:49.869 DEBUG (MainThread) [custom_components.bms_ble.plugins.cbtpwr_bms] Disconnected from BMS (170R000121)
[... cut ...]
2024-11-12 14:06:43.615 DEBUG (MainThread) [custom_components.bms_ble] 170R000086: RSSI value: -127 dBm
2024-11-12 14:06:43.616 DEBUG (MainThread) [custom_components.bms_ble] 170R000086: Link quality: 97 %
2024-11-12 14:06:47.234 DEBUG (MainThread) [custom_components.bms_ble] 170R000121: RSSI value: -127 dBm
2024-11-12 14:06:47.235 DEBUG (MainThread) [custom_components.bms_ble] 170R000121: Link quality: 96 %
2024-11-12 14:06:52.544 DEBUG (MainThread) [custom_components.bms_ble] 170R000086: BMS data update

Alex-DeIt, 2024-11-12:
Now I am testing an automation which forces HA to reload the device if it becomes unavailable (one automation for each battery like this here below)

```
alias: Ricarica dati 170R000086
  description: ''
  triggers:
  - trigger: state
    entity_id:
    - sensor.170r000086_batteria
    to: unavailable
  conditions: []
  actions:
  - action: homeassistant.reload_config_entry
    target:
      device_id: 5950357ac0cd252b967d1685fd87af3a
    data: {}
  mode: single
```
The interruptions seem to be much more shorter than before now (between 3 and 18 sec.).
[image]
[image]

patman15, 2024-11-12:
Hmm, just had a look at your logs. It looks like the underlying BT stack gets into a wrong state. It reports the device to be connected but when data is requested it raises an error that the device is not connected. I'll have look into the Bluetooth library how this can happen. Either I need to work around or patch the library.

Alex-DeIt, 2024-11-12:
> Which dongle did you use?

I was and I am  using the build in bluetooth. I had to give back the dongle with which I was performing the trials some days ago. Now I am waiting for receiving a new one.

patman15, 2024-11-12:
I would add
```YAML
...
      for:
        hours: 0
        minutes: 3
        seconds: 0
```
to your trigger condition to only catch long outages.
Can you tell me your exact Home Assistant version. I need to check the BT library and what could cause the issue on your side.

Alex-DeIt, 2024-11-12:
> Can you tell me your exact Home Assistant version.

Do you mean this?
[image]

Alex-DeIt, 2024-11-12:
> I would add

I have added it. Let's see tomorrow if it will continue to collect data over the night.

patman15, 2024-11-12:
@Alex-DeIt in order to get a bit closer to the root cause, could you check that your `configuration.yaml` has an entry for the logger and especially the two lines for the integration and bleak? You need to restart home assistant when added. Especially the last line should help to find out, why you run into such a strange state. Still not sure how I could detect/fix this from inside the integration in a reliable way.

```YAML
logger:
  default: info
  logs:
    custom_components.bms_ble: debug
    bleak.backends.bluezdbus.client: debug
```

Alex-DeIt, 2024-11-12:
yes, Patrick, sure.

I have in my configuration.yaml following:

```
logger:
  default: info
  logs:
    homeassistant.components.yamaha: critical
    custom_components.my_integration: critical
```
I have added this two lines to it.

```
custom_components.bms_ble: debug
bleak.backends.bluezdbus.client: debug
```

Alex-DeIt, 2024-11-13:
So, the longest interruption the system had tonigt started at 02:15:58 as it stopped showing the data related to battery 121 for approximately  7 minutes and 31 seconds (until 02:23:29). Strange, shouldn't it have restarted after approximately 3 minutes with the automation?
[image]
[image]

[home-assistant.log](https://github.com/user-attachments/files/17729182/home-assistant.log)

patman15, 2024-11-13:
> Where can I check this? In the history I can only see if the automation was activated or not, not if it forced a reload of the entity.

You can find it in the traces:
[image]
You can see what was executed.

> Even so, I am happy to have this integration. :)

Nice to hear! So far, I cannot find any issue with the integration (anymore), these are all problems related to an unreliable connection and misbehavior of the underlying Bluetooth stack. :sunglasses:

Alex-DeIt, 2024-11-13:
Patrick, your suspicion was correct. Neither automation has triggered a restart since 9pm yesterday.

Alex-DeIt, 2024-11-18:
Hallo Patrick,
after the core update of a couple of days ago Hassio is showing the data with almost no interruptions. The few interruprions which still happen are approximately 10 seconds long (even with the CSR8510A10 based donge as with the build in bluetooth) . So, I would say that you can close the ticket. Thank you again for your support and the great integration. :+1:

patman15, 2024-11-18:
Did you upgrade from something earlier than HA 2024.6? I guess no, but then I didn't see any important changes, but anyways, thanks for letting me know, seems that it was an underlying Bluetooth (stack) issue.

closed 2024-11-19 by patman15