patman15/BMS_BLE-HA #78: Add support Redodo battery BMS

azisto, 2024-11-05:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Hi, is it possible to add Redodo Smart BMS support to your integration ?

### Describe the solution you'd like

What information should I provide to do this?

### Additional context

I try add to manifest.json ` {
      "local_name": "R-24*"
    },`

and 
[image]

In Debug: 

`DEBUG (MainThread) [custom_components.bms_ble] Bluetooth device detected: <BluetoothServiceInfoBleak name=R-24100BNN160-A00643 address=C8:47:80:15:D8:34 rssi=-88 manufacturer_data={22618: b'\xc8G\x80\x15\xd84'} service_data={} service_uuids=['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', 'f000ffc0-0451-4000-b000-000000000000'] source=BC:2B:02:10:E2:B4 connectable=True time=19205.862500036 tx_power=None>`

patman15, 2024-11-05:
Hi! The information you already provided is helpful! Next, details about the used protocol, e.g. specification, would be needed. If you do not have that and can't find it, a BT debug trace from an Android/iOS app communicating with the BMS and the link to the app itself are necessary.
If you know Python, you can also have a look at the [contribution guidelines](https://github.com/patman15/BMS_BLE-HA/blob/main/CONTRIBUTING.md) and give it a try yourself.

azisto, 2024-11-06:
Hello, here is the link to the program: https://play.google.com/store/apps/details?id=com.redodopower.ble
My python knowledge is very weak, but I was able to get a bluetooth dump  from my smartphone while using the app. 
[btsnoop_hci.log](https://github.com/user-attachments/files/17648283/btsnoop_hci.log)
What else can I do?

patman15, 2024-11-06:
> What else can I do?

Hmm, hope for bad weather in my area, so I'm bored. :laughing: 
Basically I think, I have everything I need, so just need to find some time to implement, the read-out, the emulation, and the tests. So it takes a bit.
When (almost) done, I would ask you to test. Thanks for providing comprehensive information about the battery!

patman15, 2024-11-07:
@azisto is it really the case that I need to register with Redodo in order to use the app?! :disappointed:

azisto, 2024-11-07:
@patman15 Yes, you need to create an account to use the app. I couldn't find any information about the Bluetooth protocol.

azisto, 2024-11-07:
Maybe I could write to the manufacturer, but I don't know how to write a proper request (what exactly do I need from them and why)

patman15, 2024-11-07:
Well the ask would be a specification of the protocol between the battery and the app via Bluetooth, especially the frame format being used.
I'll have another look tomorrow to see if I can proceed otherwise.

patman15, 2024-11-08:
@azisto ok, I had a look into the data that you provided and could come up with a first version that should work partially. Can you check the version from the linked branch in this ticket? Please follow the steps for manual installation in the README.

What I could not identify (but I think must be there) is SoC, remaining capacity, cycles, temperature. In you [message](https://github.com/patman15/BMS_BLE-HA/issues/78#issuecomment-2459874828) you linked a btsnoop_hci.log that was extremely helpful to progress. Could you do this again, but also note down the time and the values from the app? Basically I would need the log and something like
`14:32:16 -> SoC 56%, remaining capacity 34.8 Ah, 22 cycles, temp 38.1°C, current: 2.8A` or a screenshot for simplicity.
Ideal would be to change a bit the battery conditions, e.g. charge/discharge etc.

If you could provide that as a .zip attached (especially if you have a number of screenshots) that would be great, I think we can solve that together!
Also any feedback to the first version is appreciated!

azisto, 2024-11-08:
@patman15  Here you go.  Now I will try the first version and give you my feedback soon.
[hci_logs.zip](https://github.com/user-attachments/files/17677969/hci_logs.zip)

azisto, 2024-11-08:
@patman15, I just tried to add the previous version from this test branch to HA manually (according to your readme instructions). This is the problem I'm having. I must be doing something wrong.
`2024-11-08 16:28:23.908 DEBUG (MainThread) [custom_components.bms_ble] Bluetooth device detected: <BluetoothServiceInfoBleak name=R-24100BNN160-A00643 address=C8:47:80:15:D8:34 rssi=-78 manufacturer_data={22618: b'\xc8G\x80\x15\xd84'} service_data={} service_uuids=['0000ffe0-0000-1000-8000-00805f9b34fb'] source=BC:2B:02:10:E2:B4 connectable=True time=54.93475358 tx_power=None>`
`2024-11-08 16:30:24.770 DEBUG (MainThread) [custom_components.bms_ble] user step`
[image]

azisto, 2024-11-08:
Update: I've uploaded the latest version. Everything is working now (commit 2e47676). 
[image]

patman15, 2024-11-08:
Except for one useless debug message, it looks good to me. Let me know if you are happy or spot some strange or implausible values. I'll release it with 1.9.0 with you confirm it's good.

patman15, 2024-11-09:
> The only thing that doesn't work is the runtime, maybe it should be the remaining time?

Good catch! I actually mixed up cycle charge (Ah) and cycle capacity (Wh). It showed the number of the first as the second which also lead to missing runtime value. Fixed in newest commit. Thanks!

closed 2024-11-11 by patman15