patman15/BMS_BLE-HA #82: Add Ective LiFePo4 batteries to your integration

Karatzie, 2024-11-11:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

I would like to add a new battery in your integration

### Describe the solution you'd like

You will find my documents below.

### Additional context

Hi Patrick,

The Ective LiFePo4 batteries have also a signal exchange.
I have not a signal list with the description of the byte order but I have reverse enginered the signals.
I have a nRF connect list and know 95% of the signals.
Do you need additional information for an integration?

This is the log-file:
[Log 2024-11-11 12_50_52.txt](https://github.com/user-attachments/files/17702750/Log.2024-11-11.12_50_52.txt)

This are my investigations.
[Daten.xlsx](https://github.com/user-attachments/files/17702841/Daten.xlsx)

Please check the documents and let me know if it possible to add the Ective battery.

Thanks

Andreas (Karatzie)

patman15, 2024-11-11:
Hi Andreas! Shouldn't be too hard. Are you using this app: https://play.google.com/store/apps/details?id=com.topband.smartpower.ective ? Because there are multiple and the store.
- From the log it looks like there is no command to request data from the battery.
- The frame always starts with `^` and seems to have a fixed length
- in your analysis you missed one line, probably because it is not shown as string, e.g. in the Excel row 77 (before it repeats)

Need to do a release and I'm a bit busy this week, but will have a look!

Karatzie, 2024-11-11:
Hi Patrick.
Thank you for your fast response.
I am not the owner of this battery.
My friend is the owner.
Yes, he uses this APP.
My Excel sheet is only for find out the words for the different values.
I still miss at the moment the words for the battery capacity and the current.
The battery was loading today. The values are not stable.
I will find out this tomorrow.

I want to understand to add a new type of battery in your integration.
I hope you can explain me this action.

I think there are many battery brands that use this type of BMS.
Please take look at this APP. This APP looks like the Ective APP.
https://play.google.com/store/apps/details?id=com.blelith.lithiumbattery
My friend have tried this APP but he got no data exchange.
It is possible that the different APPs are filtered the batteries by name or MAC address.
It is possible if the Ective battery is running, that a lot of brands are working with this integration too.
It is the same is with my Supervolt battery. 
E&J produces also for different brands.

BR
Andreas

patman15, 2024-11-11:
> I want to understand to add a new type of battery in your integration.

If you want to try yourself, have a look at the [contribution guidelines](https://github.com/patman15/BMS_BLE-HA/blob/main/CONTRIBUTING.md) for details.
BR, Patrick

patman15, 2024-11-23:
@Karatzie Hi Andreas,
do you need support adding this battery from my side? I assigned you the ticket because you said you want to try an implementation. Let me know if / when I should take it or if you are good!

Karatzie, 2024-11-23:
@patman15 Hi Patrick.

I have to try to analyze your code. 
I take a look in the different BMS-files.
I believe, I am not able to implement this type of battery. 
I am sorry.
I will be faster and better if you implement this type of battery.

I have analyzed the last missing values of the communication log.

[image]

Please add this type of battery in your code.

BR
Andreas

patman15, 2024-11-23:
> I believe, I am not able to implement this type of battery.
> I am sorry.

No worries, with your great prework it shouldn't be too difficult.

patman15, 2024-11-24:
@Karatzie Hi Andreas! I'm almost done, but I need info on how to identify the battery. Can you send me a screenshot of the advertisement of the battery from nRF connect? Or at least whether the batteries have a name schema? From the screenshots of the app, I would guess "NWJ20*" would be a good name identifier, correct?

Karatzie, 2024-11-24:
Hi Patrick.

this is from the log attached to the first post.

`nRF Connect, 2024-11-11
$PFLAC,R,RADIOID  (10:55:44:33:E8:B4)
V	12:48:48.166	Connecting to 10:55:44:33:E8:B4...
D	12:48:48.166	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	12:48:49.078	[Callback] Connection state changed with status: 133 and new state: DISCONNECTED (0)
E	12:48:49.078	Error 133 (0x85): GATT ERROR
I	12:48:49.078	Disconnected
D	12:49:11.673	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	12:49:11.740	gatt.close()
D	12:49:11.745	wait(200)
V	12:49:11.951	Connecting to 10:55:44:33:E8:B4...
D	12:49:11.951	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	12:49:11.980	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	12:49:11.980	Connected to 10:55:44:33:E8:B4
V	12:49:12.055	Discovering services...
D	12:49:12.055	gatt.discoverServices()
D	12:49:12.330	[Callback] Services discovered with status: 0
I	12:49:12.330	Services discovered
V	12:49:12.375	Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R W] (0x2A01)
- Peripheral Privacy Flag [R] (0x2A02)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
- Service Changed [I R] (0x2A05)
   Client Characteristic Configuration (0x2902)
Unknown Service (0000ffe0-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R WNR] (0000ffe1-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R WNR] (0000ffe2-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R WNR] (0000ffe3-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N] (0000ffe4-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [R] (0000ffe5-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R W WNR] (0000ffe6-0000-1000-8000-00805f9b34fb)
Unknown Service (f000ffc0-0451-4000-b000-000000000000)
- Unknown Characteristic [N W WNR] (f000ffc1-0451-4000-b000-000000000000)
   Client Characteristic Configuration (0x2902)
   Characteristic User Description (0x2901)
- Unknown Characteristic [N W WNR] (f000ffc2-0451-4000-b000-000000000000)
   Client Characteristic Configuration (0x2902)
   Characteristic User Description (0x2901)
   `
   
   Is that what you need?

Karatzie, 2024-11-24:
> _EDIT:_ theoretically working version is available on the branch, could you check/test?

I am not the owner of the battery.
I have sent him your message.
He will answer in the evening.

Karatzie, 2024-11-24:
we have to wait for the evening.

Karatzie, 2024-11-24:
My friend established a telegram group with 98 members at the moment.
This group talk over Home Assistand in a camper or camper van.
The members use a lot of different LiFePo4 battery types.
Are you interested for this group?

patman15, 2024-11-25:
Hi Andreas, that is nice of you, but it sounds a bit overwhelming for me. Working on the integration already consumes a lot of my spare time. :wink: Happy to assist if someone needs support, though.

Karatzie, 2024-11-25:
Hi Patrick,

here are the pictures from the Ective App scanning.

[image]

[image]
[image]
[image]

Karatzie, 2024-11-25:
The Ective battery was not found by the bms-ble-ha.

patman15, 2024-11-25:
Assumed so after seeing your picture, please try the updated branch.

Karatzie, 2024-11-26:
Hi Patrick.
we have a data exchange with the Ective battery.
[image]
But my friend said that the sensors changed from a value to not available after 15 seconds.
He will create a debug log for you but he has to work at the moment.

Karatzie, 2024-11-26:
I am not able to send you the log directly.
I crated an accound at [https://community.home-assistant.io](url).
But I didn't find a button to send you a personal message.
It is possible that a new user has not the permission to do that.
This is not my log and I do not know if I send secrets or is it ok.

patman15, 2024-11-26:
> He will try this settings

This won't help, I do not recommend changing it!

I sent you a PM from my HA account so I hope you are able to reply to it.

patman15, 2024-11-26:
@Karatzie I pushed an update, I think this will fix the issue (hopefully).

Karatzie, 2024-11-27:
Hi Patrick.

I was not at home over night.

You will find an error log as attachment.
[error_log-1(1).log](https://github.com/user-attachments/files/17939356/error_log-1.1.log)
I saw a lot of bytearrays twice.
[image]

We will che your new update.

BR Andreas

patman15, 2024-11-28:
> I saw a lot of bytearrays twice.

That was just a duplicate log message. I fixed that together with another patch to make sure it works.

Karatzie, 2024-11-28:
Hi Patrick,

here are the Logs of the last update.

[home-assistant_bms_ble_2024-11-28T06-19-38.629Z.log](https://github.com/user-attachments/files/17952120/home-assistant_bms_ble_2024-11-28T06-19-38.629Z.log)

[home-assistant_bms_ble_2024-11-28T19-12-29.291Z.log](https://github.com/user-attachments/files/17952122/home-assistant_bms_ble_2024-11-28T19-12-29.291Z.log)

Please take a look into the logs.
The values are valid in HA.

BR Andreas

patman15, 2024-11-29:
Hi Andreas! Does your message mean it works and you are happy? I can see still some errors in the log, but all of them are before 2024-11-28 20:11:19, after that point in time, everything is fine.

Karatzie, 2024-11-29:
Hi Patrick.
My friend had setup the RB Pi 4 in this time.
He sow that the Runtime value showes the value unbekannt.
He had restart HA very often to get Runtime with a time value.
The Runtime value has the state Unbekannt during loading of the battery.
Today in the morning he send a message that everything is ok.
He is happy with your Ective integration.

Thank you for your support for the Ective BMS.

BR
Andreas

patman15, 2024-11-30:
Thanks for the feedback, I'll add it to the next release.
`runtime` is unavailable when not discharging, since it cannot be calculated then.

PR #99 (2024-11-30): support Ective LiFePO4 batteries
fixes #82
closed 2024-11-30 by patman15