patman15/BMS_BLE-HA #84: Li Time 12.8V 100Ah and 51.4V 100Ah working

garylankford8, 2024-11-12:
I installed the integration last week, but with no success for integrating these Li Time batteries. 

After today's update v.1.9.0, HA found the new devices and configured them.

Entities created are:
Battery (SOC)
Charging
Current
Cycles
Power
Runtime
Shared Energy
Temperature
Voltage
Delta Voltage

patman15, 2024-11-12:
Cool, thanks for the information. Can you let me know which type of BMS is detected?

garylankford8, 2024-11-13:
I hate to have to ask this, but where can I find that?

patman15, 2024-11-13:
> Looks like it is using redodo.bms.

Ok, that makes sense, as the support was added in 1.9.0. Thanks!

> Only problem I see so far is that the 12.8V battery is returning an incorrect temperature. Temp for the 51.4V looks OK.

If you could attach me a debug log to this ticket, see README, chapter [_troubleshooting_](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#troubleshooting), I can have a look and probably fix it.

patman15, 2024-11-14:
Thanks a lot, I will have a look. Unfortunately, all the apps require an account so I can't verify the integration in real life.
Do you, by chance, know from, e.g. the app, what the battery temperatures were at that time so I can check for the right value?
Thanks, Patrick

garylankford8, 2024-11-14:
No, but I do believe the 51.4V is showing the correct voltage, which doesn't really make sense, right?

At the time of the log, the 12.8V showed 32.0°F (never has showed anything else) and the 51.4V showed 70°Fish. 

From the log, the centigrade temps of  0° (32°F) and 22° (71.6°F) seem to compare.

The other thing I am noting is that the 51.4V often displays signal strength as unavailable, even when link quality is high, even 100%.

garylankford8, 2024-11-14:
There is a GetHub project,

https://github.com/chadj/litime-bluetooth-battery

that uses Java to display Li Time data in a Web app.

It displays data correctly.

The address for the app is:

https://chadj.github.io/litime-bluetooth-battery/

The html for the Web app is:

[image]

Hope this helps...

patman15, 2024-11-14:
Will check and fix over the weekend. I already have some changes in mind. I will ping you, when I have a draft fix to test. Thanks again for reporting!

patman15, 2024-11-15:
@garylankford8, @azisto can you checkout the 1.9.1 pre-release regarding correct temperature presentation? You should be able to see the values of the individual sensors in the sensor attributes, in case you are interested.

azisto, 2024-11-15:
Hi @patman15. The accuracy of the temperature sensor has improved after installing 1.9.1. 
`DEBUG (MainThread) [custom_components.bms_ble] BMS data sample {'voltage': 26.688, 'current': 0.0, 'battery_level': 69, 'cycle_charge': 73.49, 'cycles': 3, 'cell#0': 3.336, 'cell#1': 3.336, 'cell#2': 3.336, 'cell#3': 3.336, 'cell#4': 3.336, 'cell#5': 3.336, 'cell#6': 3.336, 'cell#7': 3.336, 'temp#0': 23, 'temp#1': 22, 'temp#2': 23, 'delta_voltage': 0.0, 'cycle_capacity': 1961.301, 'power': 0.0, 'battery_charging': False, 'temperature': 22.667}`

azisto, 2024-11-15:
> Have a look at the FAQ regarding the max voltage. That can be adapted also to average. Let me know if you get stuck!

I did it, thank you.

garylankford8, 2024-11-15:
> @garylankford8, @azisto can you checkout the 1.9.1 pre-release regarding correct temperature presentation? You should be able to see the values of the individual sensors in the sensor attributes, in case you are interested.

I go to HACS and Redownload 1.9.1RC, from 'Need a different version'. Under 'Settings', '1 update' appears and 'Restart required'. I click on '1 update', and it shows 'Latest version' v1.9.0 and 'Installed  version' v1.9.1RC, with 'Update available'. If I 'Update' it installs v1.9.0, if I 'Skip' it shows v1.9.0 installed.

How can I get v1.9.1RC to install?

patman15, 2024-11-15:
@garylankford8 it should work if you just do your first step. Go to HACS, redownload 1.9.1RC, wait till it is finished, restart. Just tried, worked for me. If not, you can also do a [manual](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual) installation, by replacing the files.

garylankford8, 2024-11-15:
> @garylankford8 it should work if you just do your first step. Go to HACS, redownload 1.9.1RC, wait till it is finished, restart. Just tried, worked for me. If not, you can also do a [manual](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual) installation, by replacing the files.

OK, that seems to have solved my 12.8V temperature and my 51.4V signal strength issues. 

Thanks Patman!

patman15, 2024-11-15:
> Spoke too soon on the 51.4V signal strength; it still comes and goes unavailable, but not really very important...

This is an issue with the BT system. Depending on the Bluetooth adapter, not sure which one you use, it might indicate the it cannot handle the number of update events of the devices in range. As long as the `link quality` is good (high), this is not an issue.

garylankford8, 2024-11-15:
> This is an issue with the BT system. Depending on the Bluetooth adapter, not sure which one you use, it might indicate that it cannot handle the number of update events of the devices in range. As long as the `link quality` is good (high), this is not an issue.

Yeah, link quality is high for both batteries. 12.8V shows signal strength continuously. I use a mini x386 to run HAOS, which shows the BT adapter to be Intel Corporate...

I also have multiple bt-proxys and home assistant cloud, so I can see this battery status from almost anywhere, assuming my golf cart stays in wifi-range and I have internet, I guess...

Where can I contribute?

patman15, 2024-11-15:
> Where can I contribute?

Depends on what you want to contribute.

garylankford8, 2024-11-15:
> > Where can I contribute?
> 
> Depends on what you want to contribute.

A.smallish donation... I'm not good for much else, or even much more, lol!

This coding stuff is mostly beyond me, though I'm fairly good at following instructions...

patman15, 2024-11-15:
Thanks a lot. There is a [buy me a coffee](https://buymeacoffee.com/patman15) on the front page if you feel like, appreciated. Any form of feedback is also a great contribution.
Regards, Patrick

garylankford8, 2024-11-16:
Do you have a PayPal account?

patman15, 2024-11-16:
No, sorry, but never mind. :blush: