patman15/BMS_BLE-HA #85: Bulltron battery setup error

PG248, 2024-11-14:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

I try to read entities from a Bulltron LiFePo battery with HA. The battery is automatically recognised but when I confirm to proceed with the automatic setup I get the following error:

device communicating failed: [org.bluez.Error.Failed] 
Operation failed with ATT error: 0x0e (Unlikely Error) (BleakDBusError)

In terminal I can connect to the battery (sensitive data changed to XXX) :

```
bluetoothctl connect B4:XX:XX:XX:XX:XX
Attempting to connect to B4:XX:XX:XX:XX:XX
[CHG] Device E1:XX:XX:XX:XX:XX RSSI: 0xffffffbb (-69)
AdvertisementMonitor path registered
[CHG] Device B4:XX:XX:XX:XX:XX RSSI: 0xffffffc7 (-57)
hci0 B4:XX:XX:XX:XX:XX type LE Public connected eir_len 17
[CHG] Device B4:XX:XX:XX:XX:XX Connected: yes
Connection successful

bluetoothctl info B4:XX:XX:XX:XX:XX   
Device B4:XX:XX:XX:XX:XX (public)
        Name: DL-FB4XXXXXXXXXX
        Alias: DL-FB4XXXXXXXXXX
        Appearance: 0x0XXX (XXX)
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-XXXXXXXXXXXX)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-XXXXXXXXXXXX)
        UUID: Unknown                   (0000ff00-0000-1000-8000-XXXXXXXXXXXX)
        UUID: Unknown                   (0000fff0-0000-1000-8000-XXXXXXXXXXXX)
        ManufacturerData.Key: 0x0XXX (XXX)
        ManufacturerData.Value:
  00 b4 ...                    .......DL.      
        RSSI: 0xffffffc9 (-55)
        AdvertisingFlags:
  06 
```                                              .  

I tried to pair the battery with bluetoothctl pair MAC but the pairing process stops after some timeout probably waiting for a pin.

I think pairing with a pin is not necessary unless I want to change parameters of the battery. At least that's the case with the Bulltron app. I do not want to change parameters via HA only read actual charging level, voltage, current.

Any help appreciated.

patman15, 2024-11-14:
Hi! I would need a debug log to tell you more / analyze the issue. Do you know which BMS is used by the battery? (If not, logs will tell anyway.) Thanks.

PG248, 2024-11-14:
Thanks for looking.  I think it is Daly BMS though I have no official confirmation from manufacturer.

I'll send debug log tomorrow. Can I send it directly to  you? Don't like to post it publicly  because it might contain sensitive data.

patman15, 2024-11-15:
> Can I send it directly to you?

Sure, there are links in my github profile you can use, e.g. my user on HA community which allows DMs.

PG248, 2024-11-17:
Sent the debug files to your Discord account as I already have an account in Discord.

patman15, 2024-11-17:
Thanks for sending the logs. I had a look at them and there are two possible issues.
1. Please try to pair the device before using the integrations. See the [FAQ](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#my-bms-needs-a-pin-how-can-i-enter-it) on how to do that.
2. if (1) does not help, it could be related to #89 as well. See my last post there on how to update.

A feedback with a log is highly appreciated so I can nail down the issue. Thanks for raising the issue!

PG248, 2024-11-17:
Thanks for looking.

I've already tried to pair the device with bluetoothctrl pair MAC but I never get a prompt to enter a PIN and the pairing process just times out with an error.

When using the Atom bluetooth proxy pairing is even more problematic as bluetoothctl doesn't work with proxy.

If you can guide me I'd give it another try.

patman15, 2024-11-19:
@PG248 I looked at the logs again and can see a number of Bluetooth errors (indicating that the connection does not work). Have you tried rebooting once? Can you let me know, which HA version you use, what OS, what Bluetooth adapter? Did you ensure that your phone / the app or any other software that could access the battery is disabled?

PG248, 2024-12-09:
Just updated everything:
Core 2024.12.1
Supervisor 2024.11.4
Operating System 14.0
Frontend 20241127.6

BLE BMS 1.10.0

Bluetooth is via Atom Stack M5 proxy.
Hardware is RPI4.
Battery is a [Bulltron Polar 185Ah LiFePO4 battery](https://www.bulltron.de/produkte/lifepo4-12-v-185ah-polar).  

Still get the following error every few seconds:
device communicating failed: Bluetooth GATT Error address= B4:XX:XX:XX:XX handle=21 error=14 description=Unlikely error (BleakError)

Can connect to the battery with bluetoothctl connect B4:XX:XX:XX:XX but cannot pair with bluetoothctl pair B4:XX:XX:XX:XX probably because pairing needs a PIN that I cannot enter during the pairing process (no request).

Maybe an option to set the PIN in the config file for BLE BMS might be helpful?

If you want me to do further testing tell me what you need.

PG248, 2024-12-30:
I use mainly iOS and Mac stuff.

I've enabled sysdiagnose on my iPhone according to this document: https://download.developer.apple.com/iOS/iOS_Logs/Bluetooth_Logging_Instructions.pdf

Played around with the Bulltron app via Bluetooth triggered the sysdiagnose and transferred the sysdiagnose.tar.gz to my Mac. There are a ton of files and folders inside this archive. Would it be helpful to send you the archive?

If not, I have access to an older Android smartphone (Samsung) and could try the Wireshark capture outlined in your link above. Don't know whether I can gain root-level access without changing the system though.

patman15, 2024-12-31:
Hi!
Have a look at #134 there a user took https://novelbits.io/debugging-sniffing-secure-ble-ios/ to generate a trace.
Meanwhile I faked the Bulltron battery and it seems it is a Daly BMS, but the app first writes a time code to the BMS. Maybe that is required to get the flow running? Not sure. If you could generate a trace with you battery that would be most helpfull. I will see if I can make a test version that also does this one command that you could test.

patman15, 2024-12-31:
@PG248 please try the version on [this branch](https://github.com/patman15/BMS_BLE-HA/tree/85-bulltron-battery-setup-error) and send me a debug log so I can finalize the code in case it works or figure out the issue if not.

PG248, 2024-12-31:
Will do but will take a day or two because I'm currently traveling with just an iPhone and I assume I need to manually install the files from the bulltron branch, which is a bit cumbersome to do on iOS.

PG248, 2025-01-01:
Sent the debug log to your Discord account .

patman15, 2025-01-01:
Checked, but the main issue seems to be the PIN code. I need either a [BT trace](https://github.com/patman15/BMS_BLE-HA/issues/85#issuecomment-2566223563) or you could try removing the PIN if that is an option. I have no clue how they do the PIN checking, so I cannot do much without a recording.

PG248, 2025-01-02:
Sent you a Bluetooth traffic capture according to #134 via Discord.

patman15, 2025-01-05:
Key for solving the detection issue was this entry from the `VERY_VERBOSE` log of the BT proxy:
```
[13:24:48][VV][esp32_ble_tracker:453]:   Address: B4:C2:E0:XX:XX:XX (PUBLIC)
[13:24:48][VV][esp32_ble_tracker:456]:   RSSI: -45
[13:24:48][VV][esp32_ble_tracker:457]:   Name: 'DL-FB4C2E0C68F04'
[13:24:48][VV][esp32_ble_tracker:465]:   Ad Flag: 6
[13:24:48][VV][esp32_ble_tracker:479]:   Manufacturer ID: 0x0303, data: 00.B4.C2.E0.XX.XX.XX.44.4C.00 (10)
[13:24:48][VV][esp32_ble_tracker:489]:   Adv data: 02.01.06.
                                                   0D.FF.03.03.00.B4.C2.E0.XX.XX.XX.44.4C.00.
                                                   11.09.44.4C.2D.46.42.34.43.32.45.30.43.36.38.46.30.34 (35)
```
The Bulltron battery only announces name and manufacturer ID / data. No services are announced.
I updated the branch to fix the detection in the first run.

PG248, 2025-02-14:
Please have a look at our Discord chat.

patman15, 2025-05-16:
Issue found, Bulltron characteristics claim to support write-with-response, which is incorrect. Device commands require to use "write-no-response" despite the characteristics properties.

to0b, 2025-05-29:
Fyi: the branch works perfectly 👍🏼