patman15/BMS_BLE-HA #88: ACCURAT Battery is working with BMS_BLE

Karatzie, 2024-11-16:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

```text
Hi Patrick.

The ACCURAT battery is working with BMS_BLE.
```

### Additional context

[image]
[image]

Karatzie, 2024-11-16:
[Log 2024-11-15 07_25_10.txt](https://github.com/user-attachments/files/17784801/Log.2024-11-15.07_25_10.txt)

patman15, 2024-11-16:
Oh cool, thanks for the info! I will update the documentation. So no adaptations needed, simply working out of the box?

Karatzie, 2024-11-16:
That is not my battery.
It belongs to my friend with the Ective battery.
He renamed the battery from the name noted on the sticker to batt1.
[image]
The communication works out of the box.

Karatzie, 2024-11-16:
I moved my camper van to the winter storrage.
The BMS_BLE-HA found six of new devices.
I can try all batteries but I do not know the type of the battery.

closed 2024-11-19 by patman15

Karatzie, 2024-11-21:
Hi Patrick,

your battery uses the sam BMS like the accurate battery.
The [APP](https://play.google.com/store/apps/details?id=com.xiaoxiang.accurat.bms&hl=de) for the accurate battery offers commands for switching on/off the charge switch and dicharge swith.
Did you check the accurate APP with your supervolt battery?
Is it possible to use the accrate APP with your supervolt battery?
If yes, can you also switch the charge / dicharge switch of your battery?

What is your concept for these kind of switches in your integration.
Will you implement these commands in your integration?

I would like to switch off my charge switch if the battery is loaded up to 95%.
The supervolt v1 does not offer controling of the switches over APP.

BR
Andreas

patman15, 2024-11-23:
> Did you check the accurate APP with your supervolt battery?

I do not own any of those batteries. I have a small Offgridtec one, with which I started this project.

> What is your concept for these kind of switches in your integration.
> Will you implement these commands in your integration?

The concept of this integration is to be a pure monitoring one. I actually do not plan to implement active commands as this could interfere with safety critical functions of the BMS, e.g. overload, overheat. Since only a few vendors provide specifications of their protocol it is quite risky to set some flags just by trying it out or relying on information from the Internet. It could damage the battery or worse. Just recently found a protocol error in one type of BMS that is handled incorrect in all other implementations I know. Not dangerous in this case, but supports my point.
I would change my opinion only if I get official support from some vendor.

Best regards,
Patrick

Karatzie, 2024-11-23:
Hi Patrick.
Thanks for your response.
I also prefer an official support from the ventor.
That is the secure way.

Best regards,
Andreas