patman15/BMS_BLE-HA #89: Daly BMS "Failed setup, will retry: device communication timed out" since v1.7.0

flibble2000, 2024-11-17:
### Checklist

- [X] I have tried the latest version of BMS_BLE-HA.
- [X] I have enabled debug logging for my installation.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [X] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

Daly BMS 

### Describe the issue

Hi,

Thank you for the integration, it has been incredibly useful.

Since v1.7 I am unable to connect to my Dalys BMS. Rolling back to v1.6.4 fixes the problem. 

I am using the BT proxy on ESPHome 2024.10.3 and BT signal strength is -48db. The same BT proxy is used for 2 Victron devices, I've disabled the Victron integration, but it's not helped.

Personally, I'm not bothered as I'll continue with v1.6.4. But hopefully this'll help for future builds if it's a fault.

### Reproduction steps

```text
1. Upgrade to any version v1.7.0 onwards
2. 
3.
...
```

### Debug logs

[esphome.log](https://github.com/user-attachments/files/17789778/esphome.log)
[home-assistant_bms_ble_2024-11-17T09-13-35.360Z.log](https://github.com/user-attachments/files/17789779/home-assistant_bms_ble_2024-11-17T09-13-35.360Z.log)

patman15, 2024-11-17:
@flibble2000 thanks for raising this and providing a log! I suspect it has (again) to do with a previous request to add the MOS temperature value which is not present on all BMS of this type. Could you please try the modified code on the [branch](https://github.com/patman15/BMS_BLE-HA/tree/89-daly-bms-failed-setup-will-retry-device-communication-timed-out-since-v170) (see column on the right "Development") by following the instructions for [manual installation](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual)? If you could provide a log again that would be awesome!

flibble2000, 2024-11-17:
Nice one! Thanks, give me a day or two and I'll report back.

patman15, 2024-11-18:
Great, thanks for the feedback! I'm sorry for the regression, but there seem to be 3 types of Daly BMS
1. with MOS temperature sensor
2. without but responding to the request
3. without and not responding to the request

On request, I added the sensor and now you unfortunately found the 3rd type. I will fix it with the next release in a clean way. Thanks for being patient and reporting the issue! Without having such a battery there is no chance to figure that out in advance since specifications are practically non-existent.

flibble2000, 2024-11-18:
Not a problem, I'm glad I could assist. Thank you for all you hard work.