patman15/BMS_BLE-HA #90: JiKong BMS timout during initialization

SkeLLLa, 2024-11-17:
### Checklist

- [X] I have tried the latest version of BMS_BLE-HA.
- [X] I have enabled debug logging for my installation.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [X] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

Jikong Smart BMS

### Describe the issue

I've installed this integration in addition to batmon to see if it works. Initially it wasn't picking up battery at all, but eventually (maybe after upgrading to 1.9.1 and upgrading HA to latest version) I've received a notification that the battery was discovered.

So I've added the battery, however it failed during initialization step. I suspect that the problem is somewhere on my end, since batmon addon didn't work either. However if my debug info will help to make this integration better or fix some issues for other users, I'll be glad to help.

I also suspect that my BMS might not satisfy the requirement "(HW version >=11 required)", but didn't know how to check that.

I see that there's a timeout issue there, so might be it's possible to configure bigger timeout values somehow?

### Reproduction steps

```text
1. Add integration to HA, and enable debug logs
2. Wait until battery is discovered
3. Add the battery to integration
4. Battery starts initialization but fails with timeout error.
```

### Debug logs

```
2024-11-17 12:24:36.248 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JCWNPAZRY0AZ2JZ2HE9BZJ7K version=1 domain=bms_ble title=12v420a state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=C8:47:80:14:D0:C2>
2024-11-17 12:24:36.249 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for 12v420a (C8:47:80:14:D0:C2) as Jikong Smart BMS
2024-11-17 12:24:36.249 DEBUG (MainThread) [custom_components.bms_ble] device data: {'name': '12v420a', 'address': 'C8:47:80:14:D0:C2', 'rssi': -72, 'manufacturer_data': {2917: b'\x88\xa0\xc8G\x80\x14\xd0\xc2'}, 'service_data': {}, 'service_uuids': ['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', 'f000ffc0-0451-4000-b000-000000000000'], 'source': 'DC:A6:32:DA:44:EC', 'advertisement': AdvertisementData(local_name='12v420a', manufacturer_data={2917: b'\x88\xa0\xc8G\x80\x14\xd0\xc2'}, service_uuids=['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000180f-0000-1000-8000-00805f9b34fb', '0000ffe0-0000-1000-8000-00805f9b34fb', 'f000ffc0-0451-4000-b000-000000000000'], rssi=-72), 'device': BLEDevice(C8:47:80:14:D0:C2, 12v420a), 'connectable': True, 'time': 4465821.940952145, 'tx_power': None}
2024-11-17 12:24:36.250 DEBUG (MainThread) [custom_components.bms_ble] 12v420a: BMS data update
2024-11-17 12:24:36.250 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] Connecting BMS (12v420a)
2024-11-17 12:24:36.251 DEBUG (MainThread) [bleak_retry_connector] C8:47:80:14:D0:C2 - C8:47:80:14:D0:C2: Connection attempt: 1
2024-11-17 12:24:39.971 DEBUG (MainThread) [bleak_retry_connector] C8:47:80:14:D0:C2 - C8:47:80:14:D0:C2: Connected after 1 attempts
2024-11-17 12:24:39.971 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a19-0000-1000-8000-00805f9b34fb (#40): ['read', 'notify']
2024-11-17 12:24:39.971 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a27-0000-1000-8000-00805f9b34fb (#27): ['read']
2024-11-17 12:24:39.972 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a50-0000-1000-8000-00805f9b34fb (#37): ['read']
2024-11-17 12:24:39.972 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a28-0000-1000-8000-00805f9b34fb (#31): ['read']
2024-11-17 12:24:39.972 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a2a-0000-1000-8000-00805f9b34fb (#35): ['read']
2024-11-17 12:24:39.973 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a25-0000-1000-8000-00805f9b34fb (#25): ['read']
2024-11-17 12:24:39.973 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a26-0000-1000-8000-00805f9b34fb (#29): ['read']
2024-11-17 12:24:39.973 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a23-0000-1000-8000-00805f9b34fb (#33): ['read']
2024-11-17 12:24:39.973 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a24-0000-1000-8000-00805f9b34fb (#23): ['read']
2024-11-17 12:24:39.973 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a29-0000-1000-8000-00805f9b34fb (#21): ['read']
2024-11-17 12:24:39.974 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 0000ffe1-0000-1000-8000-00805f9b34fb (#17): ['write-without-response', 'write', 'notify']
2024-11-17 12:24:39.974 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 0000ffe2-0000-1000-8000-00805f9b34fb (#15): ['write-without-response']
2024-11-17 12:24:39.974 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a05-0000-1000-8000-00805f9b34fb (#11): ['read', 'indicate']
2024-11-17 12:24:39.974 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered f000ffc2-0451-4000-b000-000000000000 (#48): ['write-without-response', 'write', 'notify']
2024-11-17 12:24:39.975 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered f000ffc1-0451-4000-b000-000000000000 (#44): ['write-without-response', 'write', 'notify']
2024-11-17 12:24:39.975 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a04-0000-1000-8000-00805f9b34fb (#8): ['read']
2024-11-17 12:24:39.975 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a01-0000-1000-8000-00805f9b34fb (#4): ['read', 'write']
2024-11-17 12:24:39.975 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a00-0000-1000-8000-00805f9b34fb (#2): ['read', 'write']
2024-11-17 12:24:39.976 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Discovered 00002a02-0000-1000-8000-00805f9b34fb (#6): ['read']
2024-11-17 12:24:39.976 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Using characteristics handle #17 (notify), #17 (write)
2024-11-17 12:24:50.578 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) request cell info
2024-11-17 12:24:51.478 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Rx BLE data (start): bytearray(b'U\xaa\xeb\x90\x03\x98JK_B2A8S20P\x00\x00\x00\x00\x0011.XA\x00\x00\x0011.48\x00\x00\x00\xe4\xa7F\x00\x07\x00\x00\x0012v420a\x00\x00\x00\x00\x00\x00\x00\x00\x001234\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00240704\x00\x00404092C2262\x00000\x00Input Userdata\x00\x00140703\x00\x00\x00\x00')
2024-11-17 12:24:51.480 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Rx BLE data (cnt.): bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-11-17 12:24:51.873 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Rx BLE data (cnt.): bytearray(b'\xfe\xf9\xff\xff\x1f-\x00\x02\x00\x00\x00\x00\x90\x1f\x00\x00\x00\x00\xc0\xd8\xe72\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x07\x04\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00A\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\t\x00\x00\x00d\x00\x00\x00_\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-11-17 12:24:52.265 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Rx BLE data (cnt.): bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\xfe\xbf!\x06\x00\x00\x00\x00\x00\x00\x00\x00\xcd')
[... cut ...]
2024-11-17 12:26:23.573 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (12v420a) Rx BLE data (cnt.): bytearray(b'\x00\x00\xa0h\x06\x00\xdc\x05\x00\x00\xe4\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-11-17 12:26:29.604 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] Disconnected from BMS (12v420a)
2024-11-17 12:26:32.185 DEBUG (MainThread) [custom_components.bms_ble] 12v420a: device communication timed out
2024-11-17 12:26:32.185 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching 12v420a data in 31.500 seconds (success: False)
2024-11-17 12:26:32.186 DEBUG (MainThread) [custom_components.bms_ble] 12v420a: shuting down BMS device
```

patman15, 2024-11-17:
Hi! Thanks for the report. I can see the BMS react in different way than expected, so most probably there is a mismatch in the protocol version, or similar. Need to check more details. Have you installed the latest firmware?
You can try to change the timeout (if you know how to edit the source on your installation), although I doubt it will help, here: https://github.com/patman15/BMS_BLE-HA/blob/adc63b3eeff069de50098760ccf023b61bb024a6/custom_components/bms_ble/plugins/jikong_bms.py#L30

SkeLLLa, 2024-11-18:
@patman15 I've already tried to change the timeouts, but likely forgot to send the message with updated and logs. It didn't help a lot. Same error.

I'll try to connect with mobile app and see what firmware is there and also will try updating it to see if it helps, thanks.

patman15, 2024-11-18:
I need to check the protocol details that I can see from the log. I know there are different versions out there. Most probably it's a different one. I'll check other information on the Internet to find solutions.

SkeLLLa, 2024-11-19:
Hi.

First answering your questions:
1. Doesn't think so, since the app (mobile) actually was never connected before I tried it today. I could think only about crappy bluetooth module on my PI4.
2. As I said it was never connected before. I suspect that battery manufacturer did connections to test it, but it's definitely not in range.
3. Yes, I've did non full reset, but what usually helps with stuck bluetooth on bms - turned off the battery for ~5-10 minutes and turned on again. When I've tried BatMon addon it helped, since after some time I couldn't see battery bluetooth, but after that it helped.
4. Unfortunately currently I'm not at that place where battery is, so just asked another person to take some app screenshots. But might be I'll figure it out somehow. 

Meanwhile I've installed the app to verify firmware. Here are some readings (if it helps you).
Also the bluetooth password (that app's is asking) is standard as I assume: `1234`

[image]
[image]

patman15, 2024-11-20:
Hi!
If it has worked with batmon-ha then my current best guess is an issue with the Bluetooth connection / adapter. My protocol implementation is practically the same as the one there and the one from syssi, _JK02_32S_. At least I wouldn't know why the BMS does not answer at this stage. Sorry, I have a lot of further questions in order to narrow down this strange behaviour:

- Which [HA installation method](https://www.home-assistant.io/installation/#advanced-installation-methods) did you use
- Did you fully deactivate/uninstall _batmon-ha_ before trying?
- Do you have a lot of other devices in BT range? Can you get them out of range?
- Do you have another BT dongle that you could try, or use an [ESPHome Bluetooth proxy](https://esphome.io/components/bluetooth_proxy)?
- Probably [deactivating WiFi](https://thelinuxcode.com/disable-wifi-raspberry-pi-through-terminal/) (in case you have a wired connection) on the RPi helps. Be sure to keep BT on!

In summary, I still do not see a reason why it wouldn't work as with other users. :disappointed: Hopefully, some of the possibilities above bring hints.

patman15, 2024-11-20:
Ok, given your answers, given that the app on the phone works, given that I see replies from the BMS, but just not to the request, I think our best guess is to see what the app does. Can you follow the steps from this [guide](https://medium.com/@charlie.d.anderson/how-to-get-the-bluetooth-host-controller-interface-logs-from-a-modern-android-phone-d23bde00b9fa)? Between step 2 and 3, please run the JK app until you see the voltage level, etc., then stop Bluetooth and follow the further steps to get the zip. Out of the zip, I would only need `FS/data/log/bt/btsnoop_hci.log` to check why the app works and integration doesn't.

SkeLLLa, 2024-11-20:
Gotcha, will do that when I'll be in location with battery or just ask other person to do that. When I'll have the logs I'll upload them here.

romanshypovskyi, 2024-11-20:
Hi, I have the same BMS, jk_b2a8s20p, I found that it stopped working since version 1.6.3
Commit [6efa1e414b7f8b1c25e94719a969cc24f43a31a6](https://github.com/patman15/BMS_BLE-HA/commit/6efa1e414b7f8b1c25e94719a969cc24f43a31a6), after reverting this commit everything works again
Process hangs during initialization but works after timeout

Debug logs with and without commit [6efa1e414b7f8b1c25e94719a969cc24f43a31a6](https://github.com/patman15/BMS_BLE-HA/commit/6efa1e414b7f8b1c25e94719a969cc24f43a31a6):
[home-assistant_bms_ble_1.9.1.log](https://github.com/user-attachments/files/17836192/home-assistant_bms_ble_1.9.1.log)
[home-assistant_bms_ble_1.9.1_revert_6efa1e414b7f8b1c25e94719a969cc24f43a31a6.log](https://github.com/user-attachments/files/17836194/home-assistant_bms_ble_1.9.1_revert_6efa1e414b7f8b1c25e94719a969cc24f43a31a6.log)

romanshypovskyi, 2024-11-20:
I just tried this branch and now everything works immediately without errors:
[home-assistant_bms_ble_1.9.1+sleep1.log](https://github.com/user-attachments/files/17836990/home-assistant_bms_ble_1.9.1%2Bsleep1.log)

patman15, 2024-11-20:
Ok, I will do a bit cleaner version. I hope I can ask you for a retest before I merge. Unfortunately, I cannot check such behaviour myself. Thanks again for that amazing hint!

romanshypovskyi, 2024-11-20:
No problem, I will test the new version. Thanks for your work!

patman15, 2024-11-22:
Thanks a lot, I will take a look why. I'm not happy with keeping the delay, it just is a new source for potential floating issues. That's why I keep fixing it.

patman15, 2024-11-22:
@romanshypovskyi learned something new about the protocol which is implemented wrong in all the other implementations ...
If you could test again, it would be great. Hope I this time hit the right spot. As always, log would be great! Thanks a lot!

patman15, 2024-11-22:
:sweat_smile: Finally! Thanks for your patience and the feedback! I hope you can also see continuous updates when you just leave it running. I'll clean it up for a release now. A spec would make things so much easier. :no_mouth:

romanshypovskyi, 2024-11-22:
Everything works fine for almost an hour now. Thanks for your work!

SkeLLLa, 2024-12-16:
Hey, @patman15 . I was able to gather some logs with esp32 bt proxy.

With bt proxy it finally start working and receiving data. It's working not ideally, but since it receives data, Sometimes with following ble proxy config battery sensors become unavailable and then available again (see screenshot and logs attached).

So I blame RPi bluetooth adapter. Probably it doesn't work well either because of drivers or just because of RPi enclosure that blocks signal.

```
[22:40:27][C][bluetooth_proxy:091]: Bluetooth Proxy:
[22:40:27][C][bluetooth_proxy:092]:   Active: YES
[22:40:27][C][bluetooth_proxy:093]:   Connections: 3
[22:40:27][C][bluetooth_proxy:094]:   Raw advertisements: YES
[22:40:27][C][esp32_ble:391]: ESP32 BLE:
[22:40:27][C][esp32_ble:393]:   MAC address: 12:34:56:78:90:12
[22:40:27][C][esp32_ble:394]:   IO Capability: none
[22:40:27][C][esp32_ble_tracker:658]: BLE Tracker:
[22:40:27][C][esp32_ble_tracker:659]:   Scan Duration: 300 s
[22:40:27][C][esp32_ble_tracker:660]:   Scan Interval: 320.0 ms
[22:40:27][C][esp32_ble_tracker:661]:   Scan Window: 30.0 ms
[22:40:27][C][esp32_ble_tracker:662]:   Scan Type: ACTIVE
[22:40:27][C][esp32_ble_tracker:663]:   Continuous Scanning: True
```
[home-assistant_bms_ble_2024-12-16T20-44-41.587Z.log](https://github.com/user-attachments/files/18156924/home-assistant_bms_ble_2024-12-16T20-44-41.587Z.log)
[home-assistant_bms_ble_2024-12-16T19-57-16.134Z.log](https://github.com/user-attachments/files/18156930/home-assistant_bms_ble_2024-12-16T19-57-16.134Z.log)
[image]

Macrisu, 2025-06-08:
Same problem here, with the latest JK BMS v19 and firmware 19.05

[image]

patman15, 2025-06-08:
Ok, which version of the integration did you use? 1.7.x or 1.8.x?

Macrisu, 2025-06-08:
Batmon 1.82 - working
BMS_BLE-HA - 1.17.0 not working
I'll try 1.18.0 beta 3 - not working, log attached

[home-assistant_bms_ble_2025-06-08T20-52-09.183Z.log](https://github.com/user-attachments/files/20646788/home-assistant_bms_ble_2025-06-08T20-52-09.183Z.log)