patman15/BMS_BLE-HA #94: No supported devices found via Bluetooth.

jeremyjoslin, 2024-11-19:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

Installed add-on in HACS and when selecting the integration to add, I get this error immediately. No scanning can happen instantly so I assume this thinks I have no BLE devices which is not correct.

I have an rPi with BT scanning and several Shelly devices in active or passive mode. Other BTE integrations are using my bluetooth sensors fine.

I am trying to connect to my golf cart battery so I know when it needs to be charged. No luck so far on my second add-on to achieve this. Any assistance is appreciated.

patman15, 2024-11-19:
Hi! The integration checks the devices discovered by Home Assistant for compatibility. If it brings the error message this means it does not recognise the battery type.
Which BMS is used in your golf cart and are you sure it's in range?

patman15, 2024-11-20:
Hi Jeremy, does the _another popular add-on_ work with the battery? If so, as far as I know you need to configure the BMS type, which one did you set and did you deactivate it before trying this integration?
If it does not work, please follow the steps below:
- Open a [terminal to Home Assistant](https://www.home-assistant.io/common-tasks/supervised/#installing-and-using-the-ssh-add-on).
- Use the command `bluetoothctl devices` to check that your devices is detected and
- run `bluetoothctl info <MAC_of_BMS>` and paste the output here, so I probably can have an idea which type of BMS is used.

Best regards, Patrick

jeremyjoslin, 2024-11-20:
Thanks for the direction. This is good. 
- regarding the other add-on, I can see the MAC of the BMS in the logs but I cannot connect to it. I tried each of the different types (jk, sok, etc) to no avail. 
- I wasn't aware of these terminal commands which are very insightful. Using bluetoothctl devices I do not see the MAC of my BMS though. It appears that this command is only showing BLE devices visible to the HA device itself and not the proxies.
- Interestingly, I am also trying the Bermuda add-on for triangulation, and this one finds the beacon of the BMS and gives me a pretty good location for it - no connection, just passive. 
- I wonder if my problem may be with the Shelly BLE proxy and how I have it set up. I toggle between and active ad passive modes as part of my troubleshooting. Bermuda finds it with passive. 

I'll continue to troubleshoot on this end... **based on results of bluetoothctl it doesn't look like the issue is with your add-on.**

andreas-bulling, 2024-11-27:
I think I have the same problem. Upon installing the add-on it directly detected the two BMS (Seplos v3) but these cannot be configured - I assume because I haven't paired with these yet.

However, `bluetoothctl devices` doesn't show them.

The HomeAssistant box with the bluetooth dongle is not in the same room as the BMS so it seems Bluetooth Proxy is not working with `bluetoothctl`?

patman15, 2024-11-28:
> I think I have the same problem.

Can I assume that the BMS type detected is correct? If so, have you checked that you enabled _active connections_ on the Bluetooth Proxy?

patman15, 2024-11-28:
There is dynamic allocation between different proxies based on signal strength as far as I understood. So in theory, the closest should be ok.

andreas-bulling, 2024-11-28:
I just checked - "active" was/is set. Still, no devices can be found. :/

patman15, 2024-11-28:
> Upon installing the add-on it directly detected the two BMS (Seplos v3) but these cannot be configured

Can you elaborate on what you mean by "cannot be configured"? Any logs or error message you can share?

andreas-bulling, 2024-11-28:
Sorry - this is all I can see

[image]

I couldn't find anything in the logs for HA core

patman15, 2024-11-28:
Please enable debug logging (on the left) wait ~3 minutes and provide me the log so that I can check what's the issue.

andreas-bulling, 2024-11-28:
```
2024-11-28 19:16:56.994 DEBUG (MainThread) [custom_components.bms_ble] BP01: device communication timed out
2024-11-28 19:16:56.995 DEBUG (MainThread) [custom_components.bms_ble] BP01: shuting down BMS device
2024-11-28 19:16:56.995 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] Disconnecting BMS (BP01)
2024-11-28 19:16:57.089 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] Disconnected from BMS (BP01)
2024-11-28 19:17:02.507 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JCE7SHKBH432H57EW9VPGZJT version=1 domain=bms_ble title=BP00 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=60:6E:41:0F:3D:E6>
2024-11-28 19:17:02.507 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for BP00 (60:6E:41:0F:3D:E6) as Jiabaida Smart BMS
2024-11-28 19:17:02.507 DEBUG (MainThread) [custom_components.bms_ble] device data: {'name': 'BP00', 'address': '60:6E:41:0F:3D:E6', 'rssi': -76, 'manufacturer_data': {}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'EC:62:60:91:56:F0', 'advertisement': AdvertisementData(local_name='BP00', service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-76), 'device': BLEDevice(60:6E:41:0F:3D:E6, BP00), 'connectable': True, 'time': 2360648.255006528, 'tx_power': None}
2024-11-28 19:17:02.507 DEBUG (MainThread) [custom_components.bms_ble] BP00: BMS data update
2024-11-28 19:17:02.508 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] Connecting BMS (BP00)
2024-11-28 19:17:02.508 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:0F:3D:E6 - 60:6E:41:0F:3D:E6: Connection attempt: 1
2024-11-28 19:17:02.934 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:0F:3D:E6 - 60:6E:41:0F:3D:E6: Connected after 1 attempts
2024-11-28 19:17:13.848 DEBUG (MainThread) [custom_components.bms_ble] BP00: device communication timed out
2024-11-28 19:17:13.848 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching BP00 data in 11.340 seconds (success: False)
2024-11-28 19:17:13.848 DEBUG (MainThread) [custom_components.bms_ble] BP00: shuting down BMS device
2024-11-28 19:17:13.848 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] Disconnecting BMS (BP00)
2024-11-28 19:17:13.909 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] Disconnected from BMS (BP00)
2024-11-28 19:17:19.343 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JCE7SHKBH432H57EW9VPGZJT version=1 domain=bms_ble title=BP00 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=60:6E:41:0F:3D:E6>
2024-11-28 19:17:19.343 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for BP00 (60:6E:41:0F:3D:E6) as Jiabaida Smart BMS
2024-11-28 19:17:19.343 DEBUG (MainThread) [custom_components.bms_ble] device data: {'name': 'BP00', 'address': '60:6E:41:0F:3D:E6', 'rssi': -76, 'manufacturer_data': {}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'EC:62:60:91:56:F0', 'advertisement': AdvertisementData(local_name='BP00', service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-76), 'device': BLEDevice(60:6E:41:0F:3D:E6, BP00), 'connectable': True, 'time': 2360665.063156786, 'tx_power': None}
2024-11-28 19:17:19.344 DEBUG (MainThread) [custom_components.bms_ble] BP00: BMS data update
2024-11-28 19:17:19.344 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] Connecting BMS (BP00)
2024-11-28 19:17:19.344 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:0F:3D:E6 - 60:6E:41:0F:3D:E6: Connection attempt: 1
2024-11-28 19:17:19.809 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:0F:3D:E6 - 60:6E:41:0F:3D:E6: Connected after 1 attempts
2024-11-28 19:17:30.321 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JCE7T5PRX4BJ1EKXK7G3VQ44 version=1 domain=bms_ble title=BP01 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=60:6E:41:0F:3C:24>
2024-11-28 19:17:30.321 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for BP01 (60:6E:41:0F:3C:24) as Jiabaida Smart BMS
2024-11-28 19:17:30.321 DEBUG (MainThread) [custom_components.bms_ble] device data: {'name': 'BP01', 'address': '60:6E:41:0F:3C:24', 'rssi': -84, 'manufacturer_data': {}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'EC:62:60:91:56:F0', 'advertisement': AdvertisementData(local_name='BP01', service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-84), 'device': BLEDevice(60:6E:41:0F:3C:24, BP01), 'connectable': True, 'time': 2360675.890253576, 'tx_power': None}
2024-11-28 19:17:30.321 DEBUG (MainThread) [custom_components.bms_ble] BP01: BMS data update
2024-11-28 19:17:30.321 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] Connecting BMS (BP01)
2024-11-28 19:17:30.321 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:0F:3C:24 - 60:6E:41:0F:3C:24: Connection attempt: 1
2024-11-28 19:17:30.835 DEBUG (MainThread) [custom_components.bms_ble] BP00: device communication timed out
2024-11-28 19:17:30.835 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching BP00 data in 11.491 seconds (success: False)
2024-11-28 19:17:30.835 DEBUG (MainThread) [custom_components.bms_ble] BP00: shuting down BMS device
2024-11-28 19:17:30.835 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] Disconnecting BMS (BP00)
[... cut ...]
2024-11-28 19:18:29.105 DEBUG (MainThread) [custom_components.bms_ble] BP01: device communication timed out
2024-11-28 19:18:29.105 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching BP01 data in 11.796 seconds (success: False)
2024-11-28 19:18:29.106 DEBUG (MainThread) [custom_components.bms_ble] BP01: shuting down BMS device
2024-11-28 19:18:29.106 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] Disconnecting BMS (BP01)
2024-11-28 19:18:29.227 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] Disconnected from BMS (BP01)
```

patman15, 2024-11-28:
This is not a Seplos V3 BMS (which is currently the only supported Seplos version) but something else, it's (most likely) wrongly detected as JBD BMS. Do you have any further information about this BMS? If so, please open a new [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=enhancement&projects=&template=feature_request.yml) and add it there so we can track it separately from this topic. Thanks!

andreas-bulling, 2024-11-28:
Indeed, they are Seplos v2... Thanks for your help!

andreas-bulling, 2024-11-28:
https://github.com/patman15/BMS_BLE-HA/issues/101

closed 2025-01-08 by github-actions[bot]