patman15/BMS_BLE-HA #95: Allow setting poling frequency

msevestre31, 2024-11-20:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

```text
Hello

polling is every 30s.

Could this poll recurrence be user-defined ?

Thanks
Matt
```

patman15, 2024-11-20:
Hi Matt! The reason I did not expose it is that debugging communication issues is pretty hard and such a parameter would increase potential error sources. If you want, you can tweak it in this line:
https://github.com/patman15/BMS_BLE-HA/blob/33bc73a104e8d1af53fe944a5ba1c7224585c9a3/custom_components/bms_ble/const.py#L26
I wouldn't recommend to not set it lower though.
Out of curiosity,  why do you want to change it?

msevestre31, 2024-11-20:
Hi 

great, thanks for pointing out this possible setting ! This suits my need :) 

I am about to start some specific work on my battery and need a sharper monitoring of my Seplos BMS for a limited time (I guess I will use around 10s period).

closed 2024-11-20 by msevestre31