patman15/BMS_BLE-HA #100: Separate Calculation for Charging/Discharging Capacity for Home Assistant Battery Management

DAVEfromCZE, 2024-11-28:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

```text
Hello 
First, thank you for your work on this project—it's been incredibly helpful in integrating my Seplos BMS with Home Assistant.

I would like to suggest a feature enhancement that I believe could benefit many users:
Could you add support for calculating separate capacities for charging and discharging energy in Home Assistant's battery management?

Currently, my Seplos BMS provides only the remaining capacity, which makes it challenging to track charging and discharging energy separately. Having this feature would allow better monitoring and management of energy flow in Home Assistant.

Thank you for considering this request, and please let me know if you need any additional details or testing support.
```

patman15, 2025-04-30:
@DAVEfromCZE, @rpardini, I now finally followed up with this issue. I had a look on how to implement. My first idea was to use the change in the stored energy, but:
[image]
* There are those spikes at the end of a charge cycle (caused by voltage increase) which would lead to a quite large amount of `energy_in` and shortly afterwards `energy_out`. -> in my opinion to inaccurate, thus **dead end**.
* The next option is to continuously integrate `power=voltage*current` over time. This would solve the spike, because current is very low at the time it goes down.. Nevertheless, it is also quite inaccurate due to the fact that the sample time is only `30s` and lost samples would further reduce accuracy. -> **not satisfying** / very user dependent.
* Take the values from the BMS. Well there are **hardly any BMSs** that provide a value for the currently stored energy.
* Keep the current solution to integrate the reported power yourself using [HA helpers](https://my.home-assistant.io/redirect/helpers/). -> **Currently still preferred by me**

Let me know what you think / what solution proposal you have in mind.

Best regards,
Patrick

rpardini, 2025-05-02:
> * There are those spikes at the end of a charge cycle

PS: I don't have those spikes in my stored energy sensor. As mentioned above, the number (~1500wh) doesn't match anything though. The "shape" of the graph does match though.

[image]

patman15, 2025-05-02:
> I wonder what are the semantics behind ATTR_CYCLE_CAP (eg the source of the energy delta that is then accumulated)

That comes from the BMS, actually it is the `Ah` multiplied with the current voltage.

 > I guess. Still, it seems to me using a helper would be better resistant to restarts/reloads/overflows etc...?

Definitely, unfortunately I couldn't find a way to generate them automatically and I remember that there was a discussion in some HA forum that this is not possible. I'll check again, safety first.
I will also check if I can at least reuse the HA code to avoid code issues.

patman15, 2025-05-05:
@rpardini I talked to some (core) developers of HA
> Integrations should only expose what they find, the users should do the rest
> ...
> I don't want every integration trying to solve this on their own terms

Long story short, the integration from power to energy is nothing the integration should do. I encourage to HA developers to maybe allow for blueprints or similar and I will improve the documentation. As I'm targeting core integration mid-term, I won't follow up with this topic and go with 

> Keep the current solution to integrate the reported power yourself using [HA helpers](https://my.home-assistant.io/redirect/helpers/).

Best regards,
Patrick

closed 2025-05-05 by patman15

rpardini, 2025-05-06:
Thanks Patrick.