patman15/BMS_BLE-HA #101: Support for Seplos v2 BMS

andreas-bulling, 2024-11-28:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

```text
It'd be great to support the v2 of the Seplos BMS (10E). The BMS is already detected correctly using the current version of this integration but a connection is not possible.
```

### Additional context

https://drive.google.com/drive/folders/10pxgNLHovcDZRVGrCZsSkfecBrRw-AdW

patman15, 2024-11-30:
Hi @andreas-bulling, I need a bit more information to add the support:

- Did you rename the batteries to _BP00_ and _BP01_ or is this a default name given by the BT module?
- Can you please connect to the battery using [nRF connect](https://play.google.com/store/search?q=nrf+connect&c=apps)? When connected, swipe to the right and please provide me the log.
- Are you able to [capture the BT traffic](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) while running the app?

> adding log of wrong detection: [detect_log.txt](https://github.com/user-attachments/files/17964701/detect_log.txt)

riogrande75, 2024-12-01:
I'm interrested in a seplos v2 support, too.
Seplos bms' can be interconnected - then the main (master) bms where the can/rs485 cable is connected to presents itself as BLE device BP00, slave 1 as BP01 and so on.

I posted some of my findings already [here](https://diysolarforum.com/threads/seplos-bms-simultaneous-can-to-gx-and-rs485-to-pc-for-battery-manager.46951/post-1239399)
This should actually show howto connect to the seplos and query cell data.
Protocol description attached.
[Seplos.bluetooth.protocol.2.pdf](https://github.com/user-attachments/files/17968122/Seplos.bluetooth.protocol.2.pdf)

Off course I can also create a trace, if this will help.

patman15, 2024-12-01:
Hi! A trace is always helpful. Before that, could you please provide the output of `bluetoothctl info <MAC>` and the service explorer script. That would be needed to do the first version. Thanks!

riogrande75, 2024-12-01:
Attached requested trace:
Connect to BP00 with official app. It detects that it's a "multi" setup with 3 devices. Then I connected inside the app to BP00 and checked voltages of all batteries (Device detail), then BP01, device detail again and at last to BP02, Device detail.
So in fact, the voltages of 3x16 batteries should be visible in the trace.

MAC's here:
```
[NEW] Device 60:6E:41:26:E8:6A BP00
[NEW] Device 60:6E:41:16:72:CD BP01
[NEW] Device 60:6E:41:18:1E:29 BP02
```
[btsnooz_hci.7z.txt](https://github.com/user-attachments/files/17968350/btsnooz_hci.7z.txt)

patman15, 2024-12-01:
Thanks, but the file is corrupt and I do not need the MAC addresses but the output of `bluetoothctl info <MAC>` where `<MAC>` is the MAC address of your device(s).

riogrande75, 2024-12-01:
In fact it's not corrupt but a different/new format of a snoop trace from a galaxy s24. I'll try to find an old phone to collect traces that can be directly used in wireshark.

riogrande75, 2024-12-01:
This trace includes scenario with BP00 and BP02 in classic btsnoop_hci.log format.
[btsnoop_hci.7z.txt](https://github.com/user-attachments/files/17968899/btsnoop_hci.7z.txt)

andreas-bulling, 2024-12-01:
Thanks a lot for your help @riogrande75 
I couldn't have provided the necessary information.

Thanks @patman15 for looking into this - would be fantastic to see this BMS, which is rather popular in the DIY community, supported.

ebagnoli, 2024-12-02:
Hi all and thanks for the wonderful job:
I have the following situation: A battery with seplos 2.0 BMS and a JK active balancer. I installed your application, the esp32 Proxy, everithing went fine and I have a contact with the JK balancer. He also finds BP00, but it can't initialise. 

here something from the debug file:

2024-12-02 19:58:41.112 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for BP00 (60:6E:41:26:E9:61) as Jiabaida Smart BMS
2024-12-02 19:58:41.112 DEBUG (MainThread) [custom_components.bms_ble] device data: {'name': 'BP00', 'address': '60:6E:41:26:E9:61', 'rssi': -106, 'manufacturer_data': {}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'FC:B4:67:F4:7F:C8', 'advertisement': AdvertisementData(local_name='BP00', service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-106), 'device': BLEDevice(60:6E:41:26:E9:61, BP00), 'connectable': True, 'time': 105180.364329908, 'tx_power': None}
2024-12-02 19:58:41.113 DEBUG (MainThread) [custom_components.bms_ble] BP00: BMS data update
2024-12-02 19:58:41.114 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] Connecting BMS (BP00)
2024-12-02 19:58:41.117 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:26:E9:61 - 60:6E:41:26:E9:61: Connection attempt: 1
2024-12-02 19:58:41.163 ERROR (Thread-10 (_thread_main)) [tuya_iot] Unexpected disconnection.7
2024-12-02 19:58:41.407 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (JK_B2A16S) Rx BLE data (start): bytearray(b'U\xaa\xeb\x90\x02\x95\xa7\x0c\xa7\x0c\xa6\x0c\xa6\x0c\xa0\x0c\xa0\x0c\xa3\x0c\xa2\x0c\xa6\x0c\xa7\x0c\xa7\x0c\xa7\x0c\xa5\x0c\xa2\x0c\xa3\x0c\xa6\x0c\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xff\x00\x00\xa5\x0c\x05\x00\x00\x043\x003\x007\x008\x009\x00B\x00E\x00L\x00P\x00>\x00:\x00:\x00C\x00?\x003\x00I\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
2024-12-02 19:58:41.408 DEBUG (MainThread) [custom_components.bms_ble.plugins.jikong_bms] (JK_B2A16S) Rx BLE data (cnt.): bytearray(b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')
[home-assistant_bms_ble_2024-12-02T18-58-46.488Z.log](https://github.com/user-attachments/files/17982659/home-assistant_bms_ble_2024-12-02T18-58-46.488Z.log)

patman15, 2024-12-02:
@ebagnoli the Seplos V2 is not yet supported, I still need to do the implementation.

ebagnoli, 2024-12-03:
Thank for your replay. Is It something i can do tò help you in this implementation?

ebagnoli, 2024-12-03:
> @ebagnoli the Seplos V2 is not yet supported, I still need to do the implementation.

If this can help you I installed nrf connect and thi is the log

00001800-0000-1000-8000-00805f9b34fb
nRF Connect, 2024-12-03
BP00 (60:6E:41:26:E9:61)
V	10:55:51.216	Connecting to 60:6E:41:26:E9:61...
D	10:55:51.216	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	10:55:51.657	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	10:55:51.657	Connected to 60:6E:41:26:E9:61
V	10:55:51.677	Discovering services...
D	10:55:51.677	gatt.discoverServices()
D	10:55:51.689	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
I	10:55:51.954	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
D	10:55:52.167	[Callback] Services discovered with status: 0
I	10:55:52.167	Services discovered
V	10:55:52.177	Generic Attribute (0x1801)
- Service Changed [B I N R SW W] (0x2A05)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Privacy Flag [R] (0x2A02)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
- Central Address Resolution [R] (0x2AA6)
Unknown Service (0000ff00-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N] (0000ff01-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [W WNR] (0000ff02-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N W WNR] (0000ff03-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N R W WNR] (0000ff04-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
D	10:55:52.177	gatt.setCharacteristicNotification(0000ff01-0000-1000-8000-00805f9b34fb, true)
D	10:55:52.180	gatt.setCharacteristicNotification(0000ff03-0000-1000-8000-00805f9b34fb, true)
D	10:55:52.183	gatt.setCharacteristicNotification(0000ff04-0000-1000-8000-00805f9b34fb, true)
I	10:55:52.219	Connection parameters updated (interval: 11.25ms, latency: 0, timeout: 4000ms)

riogrande75, 2024-12-04:
If you have a first test version ready, feel free to contact me. I'm already curious and I guess I would be able to deliver useful diagnostic data too.

patman15, 2024-12-07:
Just pushed the first version. The protocol description is a bit weak, so I had to make some assumption from the V3 logic. @riogrande75, @ebagnoli, @andreas-bulling if some of you could test and provide me feedback with an attached debug log that would be very helpful. Thanks!

riogrande75, 2024-12-08:
@patman15 Thx for adressing this bms' too.

I just gave it a shot, BP0x get detected but fail after adding them:
[image]
Logfile attached.
[home-assistant_bms_ble_2024-12-08T06-38-31.625Z.log.7z.txt](https://github.com/user-attachments/files/18050995/home-assistant_bms_ble_2024-12-08T06-38-31.625Z.log.7z.txt)

patman15, 2024-12-08:
Seems that you already had (some) entries for the BMSs. Can you please delete all devices, restart and try again. It still detects JBD version although it shouldn't according to the code.

ebagnoli, 2024-12-08:
Hi, I am out of my installation for one week, Aboard, I will. Do all the testing starting from December 16. Thanks a lot for your work

andreas-bulling, 2024-12-08:
Will do - but I first have to figure out an efficient way of downloading all necessary files to custom_components...

riogrande75, 2024-12-08:
Did what you said, but after removing all BMS' and restarting HA, same result.
BP01 gets detected as Jiabaida BMS:

2024-12-08 12:16:19.738 DEBUG (MainThread) [custom_components.bms_ble] confirm step for BP01
2024-12-08 12:16:19.738 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JEJZR3MTZAMBYNQK7782ZNZ9 version=1 domain=bms_ble title=BP01 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=60:6E:41:16:72:CD>
2024-12-08 12:16:19.739 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for **BP01 (60:6E:41:16:72:CD) as Jiabaida Smart BMS**

History:
11:56 restart
12:01 JK-BMS-B added (JK)
12:02 JK-BMS-A added (JK)
12:03 48V_Block added (JBD)
12:16 BP01 added (SeplosV2) - failed 

Logs attached.
[home-assistant_bms_ble_2024-12-08T11-19-10.618Z.log.7z.txt](https://github.com/user-attachments/files/18051826/home-assistant_bms_ble_2024-12-08T11-19-10.618Z.log.7z.txt)

andreas-bulling, 2024-12-08:
For me it says the following

[image]

riogrande75, 2024-12-08:
Give it some time.
For some reason my seplos' get detected much later (~15min). My JK and JBD bms are found right after ha's start.

andreas-bulling, 2024-12-08:
Will do. So far no luck. With the old version (without Seplos v2 support) BL00 and BL01 were detected right away, though...

patman15, 2024-12-08:
@riogrande75 are you sure, you really replaced _all_ files in the folder `/custom_components/bms_ble` and subfolders? I just tried it in my test environment and it immediately changed the detection to Seplos_v2. In your last log BP00 does not even appear. Are you using a BT proxy? If so, please disable (for the test) all other BMS (except the BP00) to avoid running into the 3 device limit ...

patman15, 2024-12-08:
> Will do. So far no luck. With the old version (without Seplos v2 support) BL00 and BL01 were detected right away, though...

Sure you replaced all files (see above) and restarted? Can you provide a log? As mentioned I tried it in my test setup, it should work.

andreas-bulling, 2024-12-08:
What I did: Go to the console within HA, clone the repository, switch to the seplos v2 branch. Move everything (the while) `bms_ble` subdirectory to `/config/custom_components`.  Restarted

riogrande75, 2024-12-08:
In fact I removed the previous installation, created the directory manually at the HA server and then placed files with SFTP in it.
My HA server is currently in my office room at the 2nd floor, where it reaches JBD device directly, all other BMS's are in my tech-room in the basement. These are connecting via a BT proxy.
I can turn off JK's so the proxy should see seplos' only.

patman15, 2024-12-08:
> BL00 and BL01

Wait your devices are called B**L**00 and 01? Not B**P**00 and 01?

patman15, 2024-12-08:
@riogrande75 
> My HA server is currently in my office room at the 2nd floor, where it reaches JBD device directly, all other BMS's are in my tech-room in the basement. These are connecting via a BT proxy.

I'm confused. Which device is really a JBD? And what type is the BP00 one? Note: BMS_BLE by default detects Seplos V2 as JBD but that's not correct...

> I can turn off JK's so the proxy should see seplos' only.

Yes please, just to avoid to run into limitations while testing.

riogrande75, 2024-12-08:
OK, once again I removed everything - restartet HA und copied branch 101 via SFTP to mentioned directory.
Guess I have the right version as it shows a seplos_v2_bms.py file:
[image]
I also turned off both JK's, JBD (named 48V_Block) cannot be turned off afaik. It can be seen but I did not add it.
BP01 is visible to the app and I tried to add it:
[image]
Too bad, app still recognizes it as a JBD and not seplos...
```
2024-12-08 14:57:00.783 DEBUG (MainThread) [custom_components.bms_ble] Bluetooth device detected: <BluetoothServiceInfoBleak name=BP01 address=60:6E:41:16:72:CD rssi=-74 manufacturer_data={} service_data={} service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'] source=7C:2C:67:C9:6B:24 connectable=True time=266822.066851192 tx_power=None>
2024-12-08 14:57:00.785 DEBUG (MainThread) [custom_components.bms_ble] Device BP01 (60:6e:41:16:72:cd) detected as 'Jiabaida Smart BMS'
2024-12-08 14:57:00.785 DEBUG (MainThread) [custom_components.bms_ble] confirm step for BP01
2024-12-08 14:57:05.740 DEBUG (MainThread) [custom_components.bms_ble] confirm step for BP01
2024-12-08 14:57:07.261 DEBUG (MainThread) [custom_components.bms_ble] confirm step for BP01
2024-12-08 14:57:07.262 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JEK8YH1XZ17KCPJ3YJKGBNPN version=1 domain=bms_ble title=BP01 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=60:6E:41:16:72:CD>
2024-12-08 14:57:07.262 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for BP01 (60:6E:41:16:72:CD) as Jiabaida Smart BMS
2024-12-08 14:57:07.262 DEBUG (MainThread) [custom_components.bms_ble] BP01: advertisement: {'name': 'BP01', 'address': '60:6E:41:16:72:CD', 'rssi': -72, 'manufacturer_data': {}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': '7C:2C:67:C9:6B:24', 'advertisement': AdvertisementData(local_name='BP01', service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-72), 'device': BLEDevice(60:6E:41:16:72:CD, BP01), 'connectable': True, 'time': 266839.348024387, 'tx_power': None}
2024-12-08 14:57:07.262 DEBUG (MainThread) [custom_components.bms_ble] BP01: BMS data update

```

andreas-bulling, 2024-12-08:
> @andreas-bulling
> 
> > BL00 and BL01
> 
> Wait your devices are called B**L**00 and 01? Not B**P**00 and 01?

no, sorry. BP00 and BP01

riogrande75, 2024-12-08:
Ok, some progress.
I had a look into the code and could see, that you only handle BLE devices called BP0**0** as seplos v2. I modified the code and now BP01 gets correctly recognized:
```
2024-12-08 15:27:27.678 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JEKAP2SY49RYD8GRFZXG833Z version=1 domain=bms_ble title=BP01 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=60:6E:41:16:72:CD>
2024-12-08 15:27:27.678 DEBUG (MainThread) [custom_components.bms_ble.plugins.seplos_v2_bms] Seplos Smart BMS V2 init(), BT address: 60:6E:41:16:72:CD
2024-12-08 15:27:27.678 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for BP01 (60:6E:41:16:72:CD) as Seplos Smart BMS V2
2024-12-08 15:27:27.678 DEBUG (MainThread) [custom_components.bms_ble] BP01: advertisement: {'name': 'BP01', 'address': '60:6E:41:16:72:CD', 'rssi': -73, 'manufacturer_data': {}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': '7C:2C:67:C9:6B:24', 'advertisement': AdvertisementData(local_name='BP01', service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=-127, rssi=-73), 'device': BLEDevice(60:6E:41:16:72:CD, BP01), 'connectable': True, 'time': 268575.028421723, 'tx_power': None}
2024-12-08 15:27:27.678 DEBUG (MainThread) [custom_components.bms_ble] BP01: BMS data update
2024-12-08 15:27:27.678 DEBUG (MainThread) [custom_components.bms_ble.plugins.seplos_v2_bms] Connecting BMS (BP01)
2024-12-08 15:27:27.678 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Connection attempt: 1
2024-12-08 15:27:28.744 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 1, last rssi: None)
2024-12-08 15:27:29.245 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Connection attempt: 2
2024-12-08 15:27:30.266 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 2, last rssi: None)
2024-12-08 15:27:30.767 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Connection attempt: 3
2024-12-08 15:27:31.729 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 3, last rssi: None)
2024-12-08 15:27:32.230 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Connection attempt: 4
2024-12-08 15:27:33.441 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 4, last rssi: None)
2024-12-08 15:27:33.955 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Connection attempt: 5
2024-12-08 15:27:34.991 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 5, last rssi: None)
2024-12-08 15:27:35.494 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Connection attempt: 6
2024-12-08 15:27:36.431 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 6, last rssi: None)
2024-12-08 15:27:36.932 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Connection attempt: 7
2024-12-08 15:27:38.048 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 7, last rssi: None)
2024-12-08 15:27:38.549 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Connection attempt: 8
2024-12-08 15:27:39.481 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 8, last rssi: None)
2024-12-08 15:27:39.983 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Connection attempt: 9
2024-12-08 15:27:40.836 DEBUG (MainThread) [bleak_retry_connector] 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Failed to connect: Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish, device_missing: False, backing off: 0.5 (attempt: 9, last rssi: None)
2024-12-08 15:27:41.338 DEBUG (MainThread) [custom_components.bms_ble] BP01: device communicating failed: 60:6E:41:16:72:CD - 60:6E:41:16:72:CD: Failed to connect after 9 attempt(s): Error ESP_GATT_CONN_FAIL_ESTABLISH while connecting: Connection failed to establish: Interference/range; External Bluetooth adapter w/extension may help; Extension cables reduce USB 3 port interference (BleakAbortedError)
2024-12-08 15:27:41.339 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching BP01 data in 13.660 seconds (success: False)
2024-12-08 15:27:41.339 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS device (BP01)
```
Unfortunately there is still a problem.
Can you pls. edit the code to identify all BP0x as seplos v2?

patman15, 2024-12-08:
> Can you pls. edit the code to identify all BP0x as seplos v2?

Well the limitation to BP00 was by intention, as I understood that this is always the main/master device. I assume BP00 is the master and BP01, BP02 are the battery packs individually. Correct?
The log indicates a Bluetooth issue, i.e. the adapter cannot connect to the device.

patman15, 2024-12-08:
> I also turned off both JK's, JBD (named 48V_Block) cannot be turned off afaik. It can be seen but I did not add it.

It's sufficient if you _disable_ it in the integration, by clicking on the 3 dots next to the device name.

riogrande75, 2024-12-08:
> Well the limitation to BP00 was by intention, as I understood that this is always the main/master device. I assume BP00 is the master and BP01, BP02 are the battery packs individually. Correct? The log indicates a Bluetooth issue, i.e. the adapter cannot connect to the device.

Yes, that's basically right.
I have 3 packs: master pack BP00 (wired via RS485 to a inverter) is connected to BP01 and furthermore to BP02. The master sums up all SoC's and devides it by number of packs. Also capacities are added together as well as charging/discharging amps. This values are forwarded to the inverter together with pending alarms.

My goal is to monitor **all** seplos BMS' with HA to see the cell voltages, diff. volts, etc. of each block. This reveals if there are bad cells, etc. - stuff the inverter seems not to be interrested in.
Therefor I need to connect to each bms directly imho.

I am not quite sure how the android app does this, as I am not too familiar with BLE traces: Usually I connect to BP00 and can see all cell related values. Then the app gives the option to move forward to BP01 and BP02.
Dunno if the app connects to other BP0x directly or it does it via BP00's ble connection.
As far as I understood the traces, there is traffic directly from android to each seplos via ble.

andreas-bulling, 2024-12-08:
I can also see the `seplos_v2_bms.py` in the `plugins` folder. I recorded my screen as the message shows only very briefly: It's the configuration wizard that  is immediately followed by the error message above.

I just reinstalled the custom component but to no effect

patman15, 2024-12-08:
Hey!
@riogrande75 ok thanks for the explanation, that matches my thoughts. I think (looking at the protocol) I can query all BPs throught the main unit and I already implemented that.
@andreas-bulling can you please let me know about your Bluetooth setup. Do you have proxy, can you check from the terminal the output of `bluetoothctl devices` if you can see your `BP00`. Next please post the output of `bluetoothctl info <MAC of BP>` and a `home-assistant.log`.

Since none of you yet reached the state where the integration could connect to the device, can we please first focus on the `BP00` device? I would like to narrow down the issues. Also it confuses me less talking to both of you in parallel. :wink: I promise to add the readout of the other battery packs as I have done it for the v3 version as well.

andreas-bulling, 2024-12-08:
Since @riogrande75 is at least one step further already, I will hold back for now and let the two of you sort out the remaining issues. Hopefully that makes communication easier for you.

riogrande75, 2024-12-08:
So I removed changes I made in your code, reset everything, even BMS itself and BT proxy.
BP00 shows up, get identified as "Seplos Smart BMS V2" but connection setup fails, just like before.
If I use gattool connection setup seems to be fine.
Log attached.
[BP00.log.txt](https://github.com/user-attachments/files/18053584/BP00.log.txt)

patman15, 2024-12-08:
This is a Bluetooth issue, not related to the integration. Can you try connecting without an ESP proxy? Since you say gattool works, I assume you have a BT transmitter aside from the proxy, right? Please remove the proxy and let's see if it works directly using BT.
Do you see the device using `bluetoothctl devices`? Can you connect using `bluetoothctl connect <MAC>`?

andreas-bulling, 2024-12-09:
I have the same problem. My HA server is in a different part of the building. I "only" have a dedicated ESP32 with BT proxy next to the battery box. And as I wrote - at the moment it doesn't even detect the two BMS (in contrast to the stock version that at least found them when setting up the integration...)

riogrande75, 2024-12-09:
Yeah, it's strange. Still don't get it, why JK and JBD get detected perfectly via BT proxy too but not seplos.

andreas-bulling, 2024-12-09:
The problem with this BT Proxy is that it's not transparent what it's doing and if it's working. I have enabled it on all ESP32 devices in our house (passive mode). I can get temperature readings from all sensors distributed throughout the house just fine.

But, for some reason, despite the proxy node sitting right next to the battery box, I cannot "see" the BMSs.

And I don't even know if running `bluetoothctl devices` should show them in the first place as this is a program that runs on the operating system level and usually works with a dongle plugged into the machine itself.

Does BT proxy "inject" devices that are not directly in range of the dongle, i.e. should they also be listed?

riogrande75, 2024-12-09:
I'll try to setup a RPi2 as a temp. HA server for dev purposes tomorrow.

patman15, 2024-12-09:
> Does BT proxy "inject" devices that are not directly in range of the dongle, i.e. should they also be listed?

No, devices only in range of the proxy will **not** show up using command line tools. Their information is injected directly into the HA bluetooth stack and only visible to the integrations, not to the CLI. This makes debugging BT issues much harder, proxy logs are needed for that path.
@andreas-bulling did you set `active: true` for the proxy next to the BMS? Can you provide any logs from the proxy?

andreas-bulling, 2024-12-09:
Yeah, that's what I reckoned...

I have set the proxy to active mode - and only this one: All others in the house are in passive mode. I will try to get some logs when I am back home later in the evening.

riogrande75, 2024-12-09:
Didn't wanna let you wait so I did it right away. RPi3 with new HA image placed next to my bms', BMS_BLE installed:
[image]
Awesome work - BP00 seems to work out-of-the box.
BP01 and BP02 do not yet work (get detected as JBD), but you stated that already.
Here the log.
[home-assistant_bms_ble_2024-12-09T19-23-58.017Z.log](https://github.com/user-attachments/files/18067026/home-assistant_bms_ble_2024-12-09T19-23-58.017Z.log)

May I ask you to implement this function too:
[image]
Only so we can see voltage,etc. of each cell.

patman15, 2024-12-09:
Awesome! Seems like I had a one-time right. :smile: 

> May I ask you to implement this function too:

Already there!

> Only so we can see voltage,etc. of each cell.

Click the `delta voltage`sensor and have a look at the attribute there ... :wink:
I can see it queries the first battery pack, but misses the second. Need to figure out why.

andreas-bulling, 2024-12-09:
Sounds/looks great! If it only also worked with BT Proxy :(

riogrande75, 2024-12-10:
> @riogrande75 can you let me know, do you have in total 16 cells, or x2 or x3?
I have 3 seplos mason boxes with 16 cells each = 48 cells.

Will have a look at the proxy problem too - my JK's dont work correct with it also while JBD is fine.

riogrande75, 2024-12-10:
> Click the `delta voltage`sensor and have a look at the attribute there ... 😉 I can see it queries the first battery pack, but misses the second. Need to figure out why.

Ahh.. I see:
[image]
But should the values not have 3 digits after the comma?

riogrande75, 2024-12-10:
Ok, thx.
Yea, without proxy all BP0x units can be seen stable.
With proxy (active=true), only JBD, JK and sporadically BP01 are detected. As it doesn't see BP00 I cannot give it a try. BP01 is not working anyway so far.

BTW (off topic): What is the reason, JK's with HW v10 are not supported? I see traffic seems to be ok, but values get not evaluated correctly.

patman15, 2024-12-10:
> Yea, without proxy all BP0x units can be seen stable.

Are you sure you are not hitting the 3 devices/proxy limit, see README?

> BTW (off topic): What is the reason, JK's with HW v10 are not supported? I see traffic seems to be ok, but values get not evaluated correctly.

Simply that it uses a different protocol than implemented. :wink:

patman15, 2024-12-10:
@riogrande75 pushed an update to readout all BPs. When you have time, can you provide me a log to check if it does it correctly?

riogrande75, 2024-12-11:
Thx. But unfortunately BP01 gets still detected as "Jiabaida Smart BMS".
Log attached.
[home-assistant_bms_ble_2024-12-11T07-52-16.515Z.log](https://github.com/user-attachments/files/18091618/home-assistant_bms_ble_2024-12-11T07-52-16.515Z.log)

Gave it a shot on my productive HA with BT proxy too (with BP00), but same as before (connection setup fails).
Guess this has nothing to do with the 3 devices limit as there is currently not other bt connection in use.

andreas-bulling, 2024-12-11:
Neither for me. I ensured that the BT Proxy is the only device with an active connection.

patman15, 2024-12-11:
@andreas-bulling I still miss logs from the proxy to get an idea what is the issue on your side. Do you know how to get them?

@riogrande75 did you disable the BP01 and BP02 after they are detected? If not they will consume 1 of the 3 connections each. The more connections the less reliable they are. That is something I experienced.

riogrande75, 2024-12-11:
In fact I removed all configured BMS' from my test-HA (RPI3 without any proxy) and then I added BP01.

I'd rather wait until BP01+BP02 detection works fine before I try move back to productive HA server and start tracing the issues of with BT proxy.

patman15, 2024-12-11:
You don't need them to be detected correctly (in fact they should not appear at all), all values are queried via BP00. For now I disabled them to avoid confusion and to ensure that the proxy is not bothered with useless connections. Please update, delete the devices and restart.

andreas-bulling, 2024-12-11:
> @andreas-bulling I still miss logs from the proxy to get an idea what is the issue on your side. Do you know how to get them?

I'd be happy to help but I am afraid I don't know how to generate the logs you need. Can you quickly explain?

patman15, 2024-12-11:
Go to the [ESPhome integration](https://my.home-assistant.io/redirect/integration/?domain=esphome), add the proxy as a device, then on the left click on "enable debug logging", wait a minute or so, then click again. This should bring up a download dialog for the home-assistant.log. Attach the file here. Hope that helps!

[66 more replies]

closed 2024-12-23 by patman15