patman15/BMS_BLE-HA #102: Add TDT BSM support

goblinmaks, 2024-11-29:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

```text
https://www.tdtbms.com/products-category/smart-bms.html
```

### Additional context

https://www.tdtbms.com/products-category/smart-bms.html
https://www.tdtbms.com/download.html

AndroidApp https://www.tdtbms.com/download-category/bluetooth-app.html

patman15, 2024-11-30:
It [looks as](https://github.com/syssi/esphome-seplos-bms/issues/71) if the TDT BMS uses a similar protocol to the Seplos v2.0. Thus, I will first try to finish #101 and then come back to your request. Maybe it's solved with one implementation.
Meanwhile, please provide the output of `bluetoothctl info <MAC>` and the [service explorer script](https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py).

goblinmaks, 2024-11-30:
Yep, looks itbvery similar if compare it by rs-485 part. I hope it will be similar for bluetooth proto.

goblinmaks, 2024-11-30:
[image]
[image]

patman15, 2024-12-10:
Hi @goblinmaks, just checking, still interested? I would have soon time to try an implementation ...

goblinmaks, 2024-12-10:
Still intereting, I will support with all that I can.

patman15, 2024-12-11:
Hey! I started to look at what the BT app does, but for me it looks not really similar to a Seplos v2 BMS, but maybe I'm to tired. Are you able to to [capture the Bluetooth traffic](https://www.caichinger.com/blog/2020/11/08/capture_bluetooth_traffic_from_android/) when using the app with the battery? That would be the simplest way for me to get it running.

goblinmaks, 2024-12-12:
Captured
[btsnoop_hci.log](https://github.com/user-attachments/files/18115472/btsnoop_hci.log)

patman15, 2024-12-12:
Excellent, thanks. Confirms my thoughts, but now I have some real data I can implement against, great!

patman15, 2024-12-13:
@goblinmaks first version available on the branch. If you can provide me a debug log and check values against the app, it would be great.

goblinmaks, 2024-12-13:
Super, I will do it tomorrow. Also I missed to add screnshoy from app when capture bluetooth traffic. [image]

goblinmaks, 2024-12-14:
[image]
Could you please share more info how to debug it ?

patman15, 2024-12-14:
Please have a look at the troubleshooting section of the README. I can help more later today. Most interesting is the output of `bluetoothctl info <MAC>`

goblinmaks, 2024-12-14:
[image]

goblinmaks, 2024-12-14:
ok, i have changed local name pattern to part of mac like "C0-D6*" and remove matching by id. Have some connecting progress 
[image]

patman15, 2024-12-14:
Ok, for some reason the device changes it's name. Which is strange. Can you provide me the `raw advertisement` info using [nRF connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp) and try the updated version I just pushed?

goblinmaks, 2024-12-14:
[image]
Automatically detects with latest changes, but not received any values.

patman15, 2024-12-14:
Did you check if you need to pair the device, i.e. `bluetoothctl pair <MAC>`? Note that this does not work with a proxy!

patman15, 2024-12-15:
@goblinmaks I checked your recording and saw that there is an enable code. I added it, can you try again?

goblinmaks, 2024-12-15:
ok, will try it now.

goblinmaks, 2024-12-15:
@patman15 Excellent!
Works correctly using bluetooth_proxy and correctly get initial data. 
Let me leave it for few hours to check long data collection. 

[image]
[image]

goblinmaks, 2024-12-15:
Looks good after few working hours and few reboots of bluetooth proxy and HA. 

Thank you very much for your work.

andreitoma-github, 2024-12-16:
Hi @patman15 !
Few weeks ago I used a Win10 machine and made a Wireshark capture between the Meta BMS (running on iPhone) and the a  WattCycle battery (12V, 100Ah with BT).
The BMS advertise the name XDZN_001_AC50 and I suspect that is a TDT BMS for 4 cells only.
I hope this capture will be useful for you to validate the new TDT BSM support feature.
[Capture1_connect_get_data_few_seconds_disconnect.zip](https://github.com/user-attachments/files/18146091/Capture1_connect_get_data_few_seconds_disconnect.zip)

goblinmaks, 2024-12-16:
@andreitoma-github right not this integration works but have some problem with initial connection. I suspect that it can be a problem with my local setup only. Could you please try to install integration from the branch and check with your battery ?

patman15, 2024-12-16:
> Ok, have some error when re-add integration
> 
Looks like your device was not correctly discovered, i.e. it has not announced the name via Bluetooth. That raises the error.

patman15, 2024-12-16:
> Hi @patman15 ! Few weeks ago I used a Win10 machine and made a Wireshark capture between the Meta BMS (running on iPhone) and the a WattCycle battery (12V, 100Ah with BT). The BMS advertise the name [...]

Hi @andreitoma-github! Thanks for the input. The issue is that the current implementation can only handle 16 cell batteries since I did not exactly know how the responses would look like for a different cell count. Thanks to you I now know, I can fix that quite easily.

patman15, 2024-12-16:
@andreitoma-github fixed 4S version, @goblinmaks there was something wrong with yesterdays version. Tests did not pass, probably try again with current one.

goblinmaks, 2024-12-16:
@patman15 re-adding works good with adapter. Some problem with bluetooth_proxy

andreitoma-github, 2024-12-17:
Hi @patman15 
The commit 0d29712c7d3f3520ab267812f7650bc2af8c0410 on '102-add-tdt-bsm-support' branch works  for me. 
All sensors look correct. 
The cell voltages and cell temperatures are correct. 
The BT connection gets disrupted when the phone app is open and fetches data, but the integration reconnects after phone app is closed. 
Thank you!
[image]
[image]

patman15, 2024-12-17:
@andreitoma-github thanks for the feedback, will soon merge the code and include it with the next release.

closed 2024-12-18 by patman15