patman15/BMS_BLE-HA #103: Make it also standalone like batmon-ha?

hacsler, 2024-11-30:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

It would be great to have it running like the batmon-ha standalone https://github.com/fl4p/batmon-ha/blob/master/doc/Standalone.md . Due to the fact that it seems that I have an ej_bms.py device, the adaption in the batmon-ha is a huge work for me as "read-only" programmer. If we can use it standalone it can be placed on a Pi Zero.

patman15, 2024-11-30:
Well for having stand-alone setups (without Home Assistant) there are batmon-ha and ESPhome components. :wink: If you need a remote device to capture Bluetooth, have a look at the [Bluetooth proxy](https://esphome.io/components/bluetooth_proxy.html). Not sure what your use case is, but I think it can be solved without a stand-alone variant.

hacsler, 2024-12-01:
Unfortunately Batmon-ha does not support your ej-bms.py . I also tried bluetooth proxy without success. It will not show the  ej bms. My use case is to have a system external (garage) which gather information from my ej bms and send it to HA in my home. I also testing with OpenMqttGateway. I 'see' the bms data with MQTT Explorer, but HA wants not to 'see' it. I will see what will guide me to a working solution.

patman15, 2024-12-01:
For the Bluetooth proxy, did you set
```yaml
bluetooth_proxy:
  active: true
```
in the configuration? If you provide me with debug logs of the integration and the proxy, I think we can resolve the issue. Also, a log from [nRF connect](https://play.google.com/store/search?q=nrf+connect&c=apps) would help. I guess that there might be an issue with the auto-detection not recognizing your BMS. I only had one example to test so there might be a bug. Thanks!

hacsler, 2024-12-01:
The HACS BLE Battery Management System (BMS) does not show any log entry. In the home-assistant.log I see successful conenction of ble-proxy to HA. Only an entry that the proxy swithed off 10:10 . If you see the log of the device it scanned als later 
[10:14:47][D][esp32_ble_tracker:273]: Starting scan...
[10:19:47][D][esp32_ble_tracker:273]: Starting scan...
[10:24:47][D][esp32_ble_tracker:273]: Starting scan...
Here the debug of the seriel
[logs_ble-bluetooth-proxy_run.txt](https://github.com/user-attachments/files/17968389/logs_ble-bluetooth-proxy_run.txt)
and the nRF xml
[nRF.xml.txt](https://github.com/user-attachments/files/17968390/nRF.xml.txt)

patman15, 2024-12-01:
> The HACS BLE Battery Management System (BMS) does not show any log entry.

- Did you follow the [troubleshooting guideline](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#troubleshooting), i.e. [enable the debug protocol](https://www.home-assistant.io/docs/configuration/troubleshooting/#debug-logs-and-diagnostics) for the [BLE Battery Management integration](https://my.home-assistant.io/redirect/integration/?domain=bms_ble) and reboot once before looking at the logs?
- Is the device awake, i.e. have you been connected via the app recently? 
- Did you disconnect any other Bluetooth device (mobile with app) from the BMS?

It looks like the proxy does not catch any signal. The nRF connect log is not really useful. I would need screenshots similar to these: https://github.com/patman15/BMS_BLE-HA/issues/82#issuecomment-2498299433

Regards, Patrick

hacsler, 2024-12-01:
I attche the screenshots . The BMS is from a 12V (1024 Wh) Batterie from Electronix - about 2 years old . I saw that the UUIDs should match your ej_bms lib. Debug is always on :D I'm always 'debugging'. With the OpenMqttGateway I got 
---cut---
2024-12-01 11:34:48.782 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on esphome/OMG_ESP32_BLE_195/BTtoMQTT (qos=0): b'{"bleconnect":true,"interval":55555,"adaptivescan":true,"intervalacts":55555,"intervalcnct":3600000,"scanduration":10000,"hasspresence":false,"prestopic":"presence/","presuseuuid":false,"minrssi":-100,"extDecoderEnable":true,"extDecoderTopic":"undecoded","pubuuid4topic":false,"ignoreWBlist":false,"forcepscn":false,"tskstck":2044,"crstck":3064,"enabled":true,"scnct":125}'
2024-12-01 11:37:37.865 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on esphome/OMG_ESP32_BLE_195/LWT (qos=0): b'offline'
---cut---
Now I changed to BlueTooth Proxy and see
---cut---
BLE-Bluetooth-Proxy Safe Mode Boot wechselte zu 1. Dezember 2024 um 12:10 ausgelöst durch Dienst Knopf: Drücken
12:10:00 - Vor 24 Minuten - hass
---cut---
Because I used the switch in the dashboard. I also see
---cut---
2024-12-01 11:36:35.273 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successfully connected to ble-bluetooth-proxy @ 192.168.2xx.196 in 0.106s
2024-12-01 11:36:35.393 INFO (MainThread) [aioesphomeapi.reconnect_logic] Successful handshake with ble-bluetooth-proxy @ 192.168.2xx.196 in 0.121s
---cut---
I also added in yaml and it is also enabled in the dashboard.
---cut---
binary_sensor:
  - platform: ble_presence
    mac_address: 02:00:xx:xx:xx:xx
    name: "BMS E&J"
---cut---
It is not important - yet another a bad weather project ;)
Best regards

hacsler, 2024-12-01:
[image]
[image]
[image]
[image]

patman15, 2024-12-01:
The log didn't really help me, because I can't see a single line from the integration, although there needs to be some evidence that it is running. Anyway, the screenshots helped. Can you try the version from the linked branch? (Follow the manual installation guideline to get it running.) I added your battery to be detected. If you experience issues, I really need a complete log. You can remove all lines not containing `bms_ble` in case you fear sensitive data to get public.

patman15, 2024-12-01:
Which tells me that you tried to add a device, but none was detected. The version on the branch hopefully fixes that. :smiley:

hacsler, 2024-12-01:
So cool; now it is visible as T-12V-1544.  Now I see traffic on the ble-proxy. On HA it says that no device or entities. Failed to setup. I assume it is caused by the log entries below?
---cut---
[17:16:33][C][mdns:117]:   Hostname: ble-bluetooth-proxy
[17:16:33][C][esphome.ota:073]: Over-The-Air updates:
[17:16:33][C][esphome.ota:074]:   Address: 192.168.x.x:3232
[17:16:33][C][esphome.ota:075]:   Version: 2
[17:16:33][C][esphome.ota:078]:   Password configured
[17:16:33][C][safe_mode:018]: Safe Mode:
[17:16:33][C][safe_mode:020]:   Boot considered successful after 60 seconds
[17:16:33][C][safe_mode:021]:   Invoke after 10 boot attempts
[17:16:33][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[17:16:33][C][api:140]: API Server:
[17:16:33][C][api:141]:   Address: 192.168..x.:6053
[17:16:33][C][api:143]:   Using noise encryption: YES
[17:16:51][I][bluetooth_proxy:283]: [0] [02:00:07:81:B5:4E] Connecting v3 with cache
[17:16:51][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[17:16:51][I][esp32_ble_client:067]: [0] [02:00:07:81:B5:4E] 0x00 Attempting BLE connection
[17:16:52][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_CONNECT_EVT
[17:16:52][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_OPEN_EVT
[17:16:52][I][esp32_ble_client:154]: [0] [02:00:07:81:B5:4E] Connected
[17:16:52][D][esp32_ble_tracker:273]: Starting scan...
[17:16:52][D][esp32_ble_client:188]: [0] [02:00:07:81:B5:4E] cfg_mtu status 0, mtu 23
[17:16:52][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_REG_FOR_NOTIFY_EVT
[17:16:52][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_WRITE_DESCR_EVT
[17:16:52][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_NOTIFY_EVT
[17:16:52][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_WRITE_CHAR_EVT
[17:16:52][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_NOTIFY_EVT
[17:17:02][I][esp32_ble_client:084]: [0] [02:00:07:81:B5:4E] Disconnecting.
[17:17:02][D][esp-idf:000][BTU_TASK]: W (8460857) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
[17:17:02][D][esp-idf:000][BTU_TASK]: W (8460864) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
[17:17:02][D][esp-idf:000][BTU_TASK]: W (8460867) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16
[17:17:02][D][esp-idf:000][BTU_TASK]: W (8460871) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x16
[17:17:02][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_CLOSE_EVT
[17:17:02][D][esp-idf:000][BTU_TASK]: W (8460916) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
[17:17:14][I][bluetooth_proxy:283]: [0] [02:00:07:81:B5:4E] Connecting v3 with cache
[17:17:14][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[17:17:14][I][esp32_ble_client:067]: [0] [02:00:07:81:B5:4E] 0x00 Attempting BLE connection
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_CONNECT_EVT
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_OPEN_EVT
[17:17:14][I][esp32_ble_client:154]: [0] [02:00:07:81:B5:4E] Connected
[17:17:14][D][esp32_ble_tracker:273]: Starting scan...
[17:17:14][D][esp32_ble_client:188]: [0] [02:00:07:81:B5:4E] cfg_mtu status 0, mtu 23
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_REG_FOR_NOTIFY_EVT
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_WRITE_DESCR_EVT
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_NOTIFY_EVT
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_WRITE_CHAR_EVT
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_NOTIFY_EVT
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_NOTIFY_EVT
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_NOTIFY_EVT
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_NOTIFY_EVT
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_NOTIFY_EVT
[17:17:14][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_NOTIFY_EVT
[17:17:24][I][esp32_ble_client:084]: [0] [02:00:07:81:B5:4E] Disconnecting.
[17:17:24][D][esp-idf:000][BTU_TASK]: W (8483082) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
[17:17:24][D][esp-idf:000][BTU_TASK]: W (8483088) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
[17:17:24][D][esp-idf:000][BTU_TASK]: W (8483090) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16
[17:17:24][D][esp-idf:000][BTU_TASK]: W (8483092) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x16
[17:17:24][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_CLOSE_EVT
[17:17:24][D][esp-idf:000][BTU_TASK]: W (8483131) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
[17:17:29][I][bluetooth_proxy:283]: [0] [02:00:07:81:B5:4E] Connecting v3 with cache
[17:17:29][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[17:17:29][I][esp32_ble_client:067]: [0] [02:00:07:81:B5:4E] 0x00 Attempting BLE connection
[17:17:30][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_CONNECT_EVT
[17:17:30][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_OPEN_EVT
[17:17:30][I][esp32_ble_client:154]: [0] [02:00:07:81:B5:4E] Connected
[17:17:30][D][esp32_ble_tracker:273]: Starting scan...
[17:17:30][D][esp32_ble_client:188]: [0] [02:00:07:81:B5:4E] cfg_mtu status 0, mtu 23
[17:17:30][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_REG_FOR_NOTIFY_EVT
[17:17:30][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_WRITE_DESCR_EVT
[... cut ...]
[17:18:01][D][esp-idf:000][BTU_TASK]: W (8519835) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
[17:18:01][D][esp-idf:000][BTU_TASK]: W (8519840) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
[17:18:01][D][esp-idf:000][BTU_TASK]: W (8519843) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16
[17:18:01][D][esp-idf:000][BTU_TASK]: W (8519848) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x16
[17:18:01][D][esp32_ble_client:110]: [0] [02:00:07:81:B5:4E] ESP_GATTC_CLOSE_EVT
[17:18:01][D][esp-idf:000][BTU_TASK]: W (8519881) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
---cut---

patman15, 2024-12-01:
As mentioned, without a `home-assistant.log` I cannot help you ...

hacsler, 2024-12-01:
Now I have a lot of debug log entires ;)
[ha-bms-ble.log](https://github.com/user-attachments/files/17969841/ha-bms-ble.log)

patman15, 2024-12-02:
I think the fix was simple, just pushed an update, can you check if that works for you? A debug log would be great in any case, so I have some real world values for later to test against.

patman15, 2024-12-02:
I just pushed another commit. If you could update once again and then monitor if the values look reasonable / compare to the app it would be great. I'll merge it for the next release, which will then close the issue. If you experience issues afterwards, just raise a new issue. Thanks for supporting!

patman15, 2024-12-06:
Thanks, had a look. There are some answers in between that come from the BT module itself. I now filter them. The major issue is that for certain periods the BMS does not seem to answer while it suddenly comes back again. No clue why. I also checked these spike in the energy, this is what the BMS reports, so no problem with the integration. :confused:
You can try the new version, maybe it enhances something but if not, I can only implement some sort of restart mechanism, so the BMS can recover.

patman15, 2024-12-07:
Ok that makes sense. Most probably it turns off Bluetooth when there is no activity to safe energy. Nothing I can influence with the integration. So I guess for the moment we are done here. I will close this issue, when merging to the main branch for the next release. Just open a new issue in case you have some findings! Thanks!

closed 2024-12-08 by patman15