patman15/BMS_BLE-HA #110: Add old JK bms (HW 10.X) support

riogrande75, 2024-12-12:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

It would be nice to have widespread (old) JK bms with HW10.X supported.
[image]

Only I have about 10 pcs. of JK-B2A16S20P and JK-B2A24S20P running.
Guess the protocol should be very similar to already implemented newer JK protocol.
There is some good working python code available in the [mpp-solar project](https://github.com/jblance/mpp-solar/blob/master/mppsolar/protocols/jk02.py).
Protocol description is available from [here](https://github.com/jblance/mpp-solar/tree/master/docs/protocols) and also attached.
[JK-BMS_RS485 communication protocol_v2.5.pdf](https://github.com/user-attachments/files/18114541/JK-BMS_RS485.communication.protocol_v2.5.pdf)

So far I'm reading them via mpp-solar, pushing data to mqtt and then pick em up with HA. I would highly appreachiate a direct implementation in BMS_BLE.

patman15, 2024-12-13:
Thanks, I will need a bit more information for the auto-detection, will give it a thought when time allows.

riogrande75, 2024-12-13:
Should I prepare a new set of btsnoop traces for JK's? Would these be helpful for you?

patman15, 2024-12-14:
Ok, you are really making it hard for me to resist implementing it. :wink: Thanks! Could you also provide the output of `bluetoothctl info <MAC>` so I know how to auto-detect it? Otherwise, great job, I think I have everything I need. I just need to look for time. :smile:

riogrande75, 2024-12-14:
Here you go:
[image]

patman15, 2024-12-16:
@riogrande75 you can try the linked branch! Let me know if it works and please provide a debug log as always. Thanks!

riogrande75, 2024-12-17:
WOW! That was fast. Thank you!
And it looks very good at first sight:
[image]

Log attached.
[home-assistant_bms_ble_2024-12-17T06-04-20.465Z.log](https://github.com/user-attachments/files/18160933/home-assistant_bms_ble_2024-12-17T06-04-20.465Z.log)

I will compare everything with android app's reading as soon as I'm home. But so far everything looks plausible.