patman15/BMS_BLE-HA #113: extra tempsensors not found.

spacecabbie, 2024-12-15:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

So I have JK-BMS (B2A20S20PR-HC) with 2 temp sensors, These are not found by the software.

Was just wondering if this is A bug/feature or am i missing something ?

patman15, 2024-12-15:
Have you checked the attributes of the existing temperature sensor, i.e. click on the sensor, expand the `attributes` section?
If yes and there are none, please send me a debug log to check.

spacecabbie, 2024-12-15:
> Have you checked the attributes of the existing temperature sensor, i.e. click on the sensor, expand the `attributes` section? If yes and there are none, please send me a debug log to check.

ahhh same like the cell voltage I missed it completly sorry about that, 
Quick followup attributes list 4 values: 14,8 / 16,6 / 12,7 / 0,0 and main value is 8.8 I asume the main is the pcb of the bms the extra values confuse me a bit I have only 2 external sensors what is the 3th value ?

patman15, 2024-12-15:
There should be 5 sensors in total. the first is the *power_tube* sensor (what ever that is), the others are I think external ones. Unfortunately, I do not know which sensor is where, I just compiled them from the BMS response. Still, you found a bug, if the sensors are not connected they should not be visible. Will fix that.

patman15, 2024-12-16:
> Bms, ext1,ext2 then ext3, ext4 the last 2 would be showing 0 or as you prefer be hidden since not connected.
>
That would be also my expectation. I'll setup my fake device to check which values the app shows.

> Also if you want pictures or logs let me know.
>
Debug logs are always helpful to gather sample values from different BMSs for my test suite. Thanks!

patman15, 2024-12-16:
The app does not accept my fake device. No idea why. So can't check it.
I pushed an update to the branch related to this ticket that removes the unavailable sensors. Would be interesting, how negative temperature values look like ...

spacecabbie, 2024-12-16:
> The app does not accept my fake device. No idea why. So can't check it. I pushed an update to the branch related to this ticket that removes the unavailable sensors. Would be interesting, how negative temperature values look like ...

[home-assistant_bms_ble_2024-12-16T13-54-34.654Z.log](https://github.com/user-attachments/files/18151010/home-assistant_bms_ble_2024-12-16T13-54-34.654Z.log)

> That would be also my expectation. I'll setup my fake device to check which values the app shows.
>
>
> Debug logs are always helpful to gather sample values from different BMSs for my test suite. Thanks!

[home-assistant_bms_ble_2024-12-16T13-54-34.654Z.log](https://github.com/user-attachments/files/18151034/home-assistant_bms_ble_2024-12-16T13-54-34.654Z.log)
din't check the log for credentials or stuff like that if there is any please delete it from this ticket.

spacecabbie, 2024-12-16:
> The app does not accept my fake device. No idea why. So can't check it. I pushed an update to the branch related to this ticket that removes the unavailable sensors. Would be interesting, how negative temperature values look like ...

Your more then welcome to login to my instance let me know.

spacecabbie, 2024-12-16:
So quick update: 
T
[image]
hat first reading I have no clue it didn't seem to change when I was manipulating the censors. It's definitely not on the BMS because that's directly in the outside temperature which is now around 9°. 
I've included the pictures of the situation.

spacecabbie, 2024-12-16:
[image]

patman15, 2024-12-17:
> That first reading I have no clue it didn't seem to change when I was manipulating the censors. It's definitely not on the BMS because that's directly in the outside temperature which is now around 9°.
>
Can you please check whether your SW/HW version is >=11? Probably there is a protocol issue.

spacecabbie, 2024-12-19:
> > That first reading I have no clue it didn't seem to change when I was manipulating the censors. It's definitely not on the BMS because that's directly in the outside temperature which is now around 9°.
> 
> Can you please check whether your SW/HW version is >=11? Probably there is a protocol issue.

Hw: V11A
SW: V11.54

Added screenshots.
https://imgur.com/a/EPogqLl

patman15, 2024-12-19:
Hmm, you have *MOS*, *Battery T1*, and *Battery T2*, makes 3 sensors in total. The integration shows also 3. Are the values incorrect or what is the issue? I got a bit lost of the actual problem.

spacecabbie, 2024-12-19:
No i have 4, unless i read this wrong..

https://imgur.com/a/UgrumbR

patman15, 2024-12-19:
> No i have 4, unless i read this wrong..

It's 3:
[image]
Readings are:
- 16.6°C
- 19.4°C
- 18.8°C

then there are two wrong ones, which I removed in the [newer version](https://github.com/patman15/BMS_BLE-HA/releases/tag/v1.11.0-beta.4)
- 0°C and
- 0°C

spacecabbie, 2024-12-20:
> It's 3: [...]

I follow what you're saying perhaps I wasn't clear about it but I count the main sensor, 
The one that says 11°C that's how I come to four. The 16.6°C is the stranger in my example, I cannot change it by manipulating the known sensors, warming the BMS itself changes the main one (11.0) and warming to sensor leads changes the mid 2 vallue's (19.4 / 18.8) the 16.6 and 0 values do not seem to change. 

Quick question how do I apply this beta release copy it over the plugin directory?

patman15, 2024-12-20:
> Quick question how do I apply this beta release copy it over the plugin directory?

Always copy the whole integration by following the manual installation guide. Or in this case for the beta release, you can just install via HACS by selecting redownload and pick the beta release.

The "main sensor" is just the average of the attributes. What is strange is, that the first attribute seems to be stuck. I'll check the implementation again. If you can provide a log together with real values I can also check this.

spacecabbie, 2024-12-21:
In the new version, it seems resolved.

https://imgur.com/a/Dwnxy2L

Thanks for the explanation. How the sensor readings work, personally, I
would rather have three separate sensors, I think this is a very clever way
of dealing with the sub-resistance and sup-voltage readings, also well
explained in the wiki.

From A user perspective, it doesn't make sense though for the temperatures
IMHO.

It would be wonderful to have a toggle option, so the sensors can be
displayed as separate entities or in this way, but this is just fine-tuning
of The interface, and is absolutely not a priority or even possibly needed.

 Anyways for me the issue is resolved.

patman15, 2024-12-21:
> From A user perspective, it doesn't make sense though for the temperatures IMHO.

I actually agree, but finding a simple, maintainable way is really difficult. The integration now support 12+ different BMSs and all have their specialties. Even if I separate cell temperatures from others, the other would still be strange to be summarized in any way, because they could be ambient, MOS, etc. temperatures. In short, context is key here. Thus I decided to just average all. In case you want to have them separate, have a look at the [FAQ](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#how-do-i-get-the-cell-voltages-as-individual-sensor-for-tracking). Similar method also works for temperatures. Hope that fine-tunes your setup!
Best regards, Patrick

closed 2024-12-21 by patman15