patman15/BMS_BLE-HA #122: The Daly BMS is not recognised by this BLE integration

ashinroy1993, 2024-12-21:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

Hi, my Daly BMS was earlier working with this integration, but after a core update this stopped working, but I tried reinstalling the home assistant and these integration from scratch but its still not working

patman15, 2024-12-21:
Hi! Which version of the integration did you have before and which one now? Can you provide data as described in the troubleshooting guide?

ashinroy1993, 2024-12-22:
Hi, I got this resolved by disabling passive scanning in my bluetooth adapter. thanks for reaching out. 
One issue I still observe is the battery is not detected in this integration while using esp32 bluetooth proxy.

patman15, 2024-12-22:
Again, please provide logs as described, otherwise I can't help. Most likely it's an issue with your proxy as the integration has no specifics regarding its usage.

closed 2025-02-01 by github-actions[bot]