patman15/BMS_BLE-HA #123: Ective Batteries not detected

unbreaky83, 2024-12-22:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

Hi,

I was trying to add my two Ective-Batteries to Home-Assistant. Unfortunately, if i try to add them as devices, i get the error "No supported devices found via Bluetooth".

After that, i checked the bluetooth information with bluetoothctl info MAC and compared the UUID and Manufacturer Data Key with the manifest file. Both information are matching. 

I also tried to connect to the devices with bluetooth connect MAC, but it did not work, but i also got the message "Unable to open mgmt_socket". Not sure if this is by design or an issue with my Raspberry Pi 4 since I'm new with Home-Assistant.

Any help/hints are appreciated
Best regards
Chris

patman15, 2024-12-23:
Hi! Can you provide the information shown by `bluetoothctl info <MAC>` please? Did you check that `passive scanning` is disabled on the Bluetooth adapter?
Best regards, Patrick

unbreaky83, 2024-12-23:
Hi Patrick

I checked the passive scanning and it is disabled

[core-ssh var]$ bluetoothctl info 30:55:44:37:62:XX
Unable to open mgmt_socket
Device 30:55:44:37:62:XX (public)
	Name: NWJ202212230103XX
	Alias: NWJ202212230103XX
	Paired: no
	Bonded: no
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Unknown		{0000ffe0-0000-1000-8000-00805f9b34fb}
	ManufacturerData Key: 0xffff (65535)
	ManufacturerData Value: 30 55 44 37 62 XX	0UD7b.
	RSSI: 0xffffffcd (-51)
	AdvertisingFlags: 05		.

[core-ssh var]$ bluetoothctl info 30:55:44:37:A2:XX
Unable to open mgmt_socket
Device 30:55:44:37:A2:XX (public)
	Name: NWJ202212230103XX
	Alias: NWJ202212230103XX
	Paired: no
	Bonded: no
	Trusted: no
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Unknown		{0000ffe0-0000-1000-8000-00805f9b34fb}
	ManufacturerData Key: 0xffff (65535)
	ManufacturerData Value: 30 55 44 37 a2 XX	0UD7..
	RSSI: 0xffffffb9 (-71)
	AdvertisingFlags: 05		.

Thank you
Best regards
Chris

patman15, 2024-12-23:
Ok, there was an issue with the naming. Can you try the version from the main branch, or do you need another beta release?

unbreaky83, 2024-12-23:
Thank you, i have replaced the new versions of the two files manually since I have limited access to the raspberry at the moment. 
The two batteries got discovered now. 

Unfortunately, if i try to add them, i get the following errors: "Invalid Flow Specified" or "Unknown Error Occurred" 

I have attached the Debug-Log
[home-assistant_bms_ble_2024-12-23T12-13-49.560Z.log](https://github.com/user-attachments/files/18229215/home-assistant_bms_ble_2024-12-23T12-13-49.560Z.log)

Thank you very much for your patience and help

patman15, 2024-12-23:
I found the issue, but I do not know how to fix it. There is a conflict with the internal integration for `kegtron`. Please try to run `pip install kegtron_ble` from the terminal and reboot. I think this fixes it. I'm trying to get support from HA developers as I'm not sure it is an issue with bms_ble. Let me know if the work around fixes the issue. Thanks!

unbreaky83, 2024-12-23:
I tried the command, but i seems not be available on the HomeAssistant OS. 

bash: pip: command not found

unbreaky83, 2024-12-23:
Yes. I tried with that Addon. Maybe Home Assistant OS don't have that python extension? Or is it a raspberry thing?

unbreaky83, 2024-12-24:
Sorry, my fault. I mixed up the different terminals. 

I was able to install the kegtron_ble extension. Now it works and I see the both batteries with voltage and SoC

Thank you very much for your help 
Best regards and merry Christmas 
Chris

patman15, 2024-12-24:
Ok, great! Still I need to figure out how to avoid this issue. Thanks for the confirmation that the workaround is ok.
Merry Christmas, Patrick

patman15, 2024-12-24:
@unbreaky83 which [installation method](https://www.home-assistant.io/installation/#advanced-installation-methods) did you use for HA? Are you using HA OS?

unbreaky83, 2024-12-24:
Yes, I am using the Home Assistant OS. 

I used the following method to prepare the Raspberry Pi 4: https://www.home-assistant.io/installation/raspberrypi/

patman15, 2024-12-25:
JFYI: I tried to reproduce the issue on a HA core installation as well as on freshly installed HAOS 14.1 (x86) but without any luck. I can reproduce it with my test though. Since I'm not sure the root cause for the message is the same for the tests, I will check them again.

patman15, 2024-12-26:
@unbreaky83 I tried to reproduce the issue, but still no luck. The issue with the tests has a different root cause. Since your immediate problem is solved, are you ok to close this topic?

unbreaky83, 2024-12-27:
Hi Patrick 

Strange, with which Hardware did you test? Yes, you can close that topic. Thank you very much for your help. 

Best regards 
Chris

patman15, 2024-12-27:
> Strange, with which Hardware did you test?,

I have an ESP32 faking an Ective battery. But the issue is not battery related, but related to the detection of it (which is proven by the fact that installing the Kegtron dependency that actually solve the issue). I opened an issue with the Kegtron integration that causes the conflict (see above), although I still do not understand, why HA did not automatically install the dependency on your system. For me it always did so in all configurations. (Only exceptions was for tests, where it is supposed to be like that.)

closed 2024-12-31 by patman15