patman15/BMS_BLE-HA #124: JK BMS V15 not working anymore with Version 1.11.0

downset, 2024-12-23:
### Checklist

- [X] I have tried the latest version of BMS_BLE-HA.
- [X] I have enabled debug logging for my installation.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [X] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

JK BMS V15.35

### Describe the issue

The BMS gets no connection with the new Release
The attached log is a snipped of the Bluetooth Proxy, witch repeats over and over

### Reproduction steps

```text
1.update to 1.11.0
2.restart Homeassistant
3.Looking at the log of the ESPhome bluetooth proxy
```

### Debug logs

[17:12:27][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:27][I][esp32_ble_client:084]: [0] [C8:47:80:0F:D5:C5] Disconnecting.
[17:12:27][D][esp-idf:000][BTU_TASK]: W (112139974) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13

[17:12:27][D][esp-idf:000][BTU_TASK]: W (112139981) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16

[17:12:27][D][esp-idf:000][BTU_TASK]: W (112139985) BT_APPL: gattc_conn_cb: if=4 st=0 id=4 rsn=0x16

[17:12:27][D][esp-idf:000][BTU_TASK]: W (112139987) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x16

[17:12:27][D][esp32_ble_client:110]: [0] [C8:47:80:0F:D5:C5] ESP_GATTC_CLOSE_EVT
[17:12:27][D][esp-idf:000][BTU_TASK]: W (112140024) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16

[17:12:27][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
17:12:27][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:28][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:28][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:28][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:28][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:28][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:29][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:29][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:29][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:29][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:29][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:30][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:30][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:30][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:30][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:30][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_NOTIFY_EVT
[17:12:30][I][esp32_ble_client:084]: [1] [C8:47:80:0D:08:6B] Disconnecting.
[17:12:30][D][esp-idf:000][BTU_TASK]: W (112143251) BT_HCI: hci cmd send: disconnect: hdl 0x1, rsn:0x13

[17:12:30][D][esp-idf:000][BTU_TASK]: W (112143256) BT_APPL: gattc_conn_cb: if=3 st=0 id=259 rsn=0x16

[17:12:30][D][esp-idf:000][BTU_TASK]: W (112143258) BT_APPL: gattc_conn_cb: if=4 st=0 id=260 rsn=0x16

[17:12:30][D][esp-idf:000][BTU_TASK]: W (112143262) BT_APPL: gattc_conn_cb: if=5 st=0 id=261 rsn=0x16

[17:12:30][D][esp32_ble_client:110]: [1] [C8:47:80:0D:08:6B] ESP_GATTC_CLOSE_EVT
[17:12:30][D][esp-idf:000][BTU_TASK]: W (112143299) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x16
[17:13:47][D][esp32_ble_tracker:222]: Pausing scan to make connection...
[17:13:47][I][esp32_ble_client:066]: [0] [C8:47:80:0F:D5:C5] 0x00 Attempting BLE connection
[17:13:48][D][esp32_ble_client:110]: [0] [C8:47:80:0F:D5:C5] ESP_GATTC_CONNECT_EVT
[17:13:48][D][esp32_ble_client:110]: [0] [C8:47:80:0F:D5:C5] ESP_GATTC_OPEN_EVT
[17:13:48][I][esp32_ble_client:154]: [0] [C8:47:80:0F:D5:C5] Connected
[17:13:48][D][esp32_ble_tracker:273]: Starting scan...
[17:13:48][D][esp32_ble_client:110]: [0] [C8:47:80:0F:D5:C5] ESP_GATTC_REG_FOR_NOTIFY_EVT
[17:13:48][D][esp32_ble_client:187]: [0] [C8:47:80:0F:D5:C5] cfg_mtu status 0, mtu 131
[17:13:48][D][esp32_ble_client:110]: [0] [C8:47:80:0F:D5:C5] ESP_GATTC_WRITE_DESCR_EVT
[17:13:48][D][esp32_ble_client:110]: [0] [C8:47:80:0F:D5:C5] ESP_GATTC_NOTIFY_EVT
[17:13:48][D][esp32_ble_client:110]: [0] [C8:47:80:0F:D5:C5] ESP_GATTC_WRITE_CHAR_EVT
[17:13:48][D][esp32_ble_client:110]: [0] [C8:47:80:0F:D5:C5] ESP_GATTC_NOTIFY_EVT

patman15, 2024-12-23:
Can you get me a debug log to see what the integration says? Thanks!

patman15, 2024-12-23:
I don't know why your BMS likes to send additional "AT\r\n" messages while others don't. Seems to be some strange setting. Can you try the branch, which filters them?

downset, 2024-12-23:
and there it is back
quck fix :)

maybe it is cause of the newest firmware v35?
This has also new values eg if it is in float or bulk charge mode and some more, but until now even the original app can`t show them

Thanks for your quick support

downset, 2024-12-25:
i have to reopen this again.
After some time with the fix, the BMS shows false values
Some are 0 some are extrem high e.g. Voltage of 4.294.967,3 V (witch could be right multiplied by 10^-5  ->42,949 )
Reloading the pack in the integration fixes it for some hours
Here is a lock wile it is in this State
[home-assistant_bms_ble_2024-12-25T11-58-10.275Z.log](https://github.com/user-attachments/files/18245817/home-assistant_bms_ble_2024-12-25T11-58-10.275Z.log)

patman15, 2024-12-25:
Excellent catch, thanks for the logs! Can you try this [branch](https://github.com/patman15/BMS_BLE-HA/tree/124-jk-bms-v15-not-working-anymore-with-version-1110)? By the way, this was not a newly introduced issue, but already existent in previous versions.

downset, 2024-12-25:
interesting, have had sometimes reconnect problems in the past, so that the BMS was offline for some hours, but thought, it is caused of the bad wifi in the Inverter and Battery Room.
But have never had this strange numbers bevor.

The Branch is installed, and the BMS both show up,
Will come back here after abut 24h to confirm if it keeps going

happy Christmas, and don't code to much in these days :)

Smurphy1965, 2024-12-26:
Sorry if this is a no go, but i want to share my problem, and not open a new issue. I have a perfect balanced 8S battery with winston cells and a xiaoxiang BMS. After installing the plugin,  my cell nr 5 started to deplete very fast, and unbalanced the battery. My first idea was som bad connection to the sensor wires. After re balancing the battery, and turned off the BMS BLE the battery stays balanced. 
[image]

patman15, 2024-12-26:
> Sorry if this is a no go, but i want to share my problem, and not open a new issue.

Thanks for sharing. To me this looks like a defective cell, not related to the integration at all. If you want me to look deeper, please [open a new issue](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=question&projects=&template=support.yml) and [provide debug logs](https://github.com/patman15/BMS_BLE-HA#in-case-you-have-severe-troubles), since this issue is not at all related to the topic of this thread, neither the BMS type nor the problem.

downset, 2024-12-26:
I think you can close and merge :)
not fully 24h later, but rock stable not even one point with missing data

PR #125 (2024-12-23): Filter AT commands at the end of JK messages
fixes #124
closed 2024-12-26 by patman15