patman15/BMS_BLE-HA #133: Status "unavailable" every three minutes...

Prokopchini, 2024-12-29:
### Checklist

- [X] I have tried the latest version of BMS_BLE-HA.
- [X] I have enabled debug logging for my installation.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [X] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

JK-B2A8S20P

### Describe the issue

Every three minutes entities in HA go into unavailable status and after 30 seconds appear again

### Reproduction steps

```text
1.
2.
3.
...
```

### Debug logs

Unexpected error fetching LiFePo4 data
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 379, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/bms_ble/coordinator.py", line 97, in _async_update_data
    battery_info = await self._device.async_update()
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/bms_ble/plugins/basebms.py", line 257, in async_update
    await self._connect()
  File "/config/custom_components/bms_ble/plugins/basebms.py", line 214, in _connect
    await self._init_connection()
  File "/config/custom_components/bms_ble/plugins/jikong_bms.py", line 194, in _init_connection
    -32 if int(self._bms_info.get("sw_version", "")[:2]) < 11 else 0
           ~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid literal for int() with base 10: '\x00\x00'

patman15, 2024-12-29:
Thanks for reporting, can you please attach a debug log of the last minute or so before the exception occurs? Thanks!

Prokopchini, 2024-12-29:
> Thanks for reporting, can you please attach a debug log of the last minute or so before the exception occurs? Thanks!

[home-assistant_bms_ble_2024-12-29T22-03-50.100Z.log](https://github.com/user-attachments/files/18271060/home-assistant_bms_ble_2024-12-29T22-03-50.100Z.log)

patman15, 2024-12-30:
Hey! Could you check the version on the linked branch? I hope it is a simple fix. :crossed_fingers:

Prokopchini, 2024-12-30:
> Hey! Could you check the version on the linked branch? I hope it is a simple fix. :crossed_fingers: 

Hi, I'm sorry - but I don't understand - what is this "version on the linked branch"

Prokopchini, 2024-12-30:
I think I got it - I need to copy the changes to my file. Thanks, I'll try

patman15, 2024-12-30:
> Hi, I'm sorry - but I don't understand - what is this "version on the linked branch"

Please select [this branch](https://github.com/patman15/BMS_BLE-HA/tree/133-status-unavailable-every-three-minutes) and follow the [manual install](https://github.com/patman15/BMS_BLE-HA#manual) instructions.

Prokopchini, 2024-12-30:
> > Hi, I'm sorry - but I don't understand - what is this "version on the linked branch"
> 
> Please select [this branch](https://github.com/patman15/BMS_BLE-HA/tree/133-status-unavailable-every-three-minutes) and follow the [manual install](https://github.com/patman15/BMS_BLE-HA#manual) instructions.

Thank you! Everything worked out! Everything has been working correctly for three hours now

commit 6f09c830 (2024-12-30): ensure fresh data after command (#136)

issue with JK BMS streaming data, fixes #133
closed 2024-12-30 by patman15