patman15/BMS_BLE-HA #137: Keine unterstützen Geräte via Bluetooth gefunden.

Heila68, 2024-12-31:
### Checklist

- [X] I have tried the latest version of BMS_BLE-HA.
- [X] I have enabled debug logging for my installation.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [X] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

Ective BMS

### Describe the issue

The error message  "Keine unterstützen Geräte via Bluetooth gefunden." shows up immediatly after trying to add the integration to HA.
I read the issue "Ective Batteries not detected #123" which was solved by a newer version, but since I am using the actual version, it seems to be an other problem in my case.
However, I provide you the output of "bluetoothctl info <MAC>" under "debug logs".

Thanks a lot in advance!
Best regards
Heinz 

### Reproduction steps

```text
1.
2.
3.
...
```

### Debug logs

Device 34:14:B5:9D:78:XX (public)
        Name: NWJ20200720020539
        Alias: NWJ20200720020539
        Paired: no
        Bonded: no
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Unknown                   (0000ffe0-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (f000ffc0-0451-4000-b000-000000000000)
        Modalias: bluetooth:v000Dp0000d0110
        ManufacturerData.Key: 0x0000 (0)
        ManufacturerData.Value: 34 14 b5 9d 78 XX 4...x.          
        AdvertisingFlags:  05 .

patman15, 2024-12-31:
You need to install the version from the main branch. There is currently no release with the fix.

Heila68, 2024-12-31:
Hi Patrick,
please excuse my ignorance, I thougth I install the main branch when adding the repository link to HACS?
Could you help me how to do the right installation of the main branch in HACS?
Best regards
Heinz

patman15, 2024-12-31:
Hi Heinz, HACS always installs the latest release, not necessarily the latest code. If you want to do that, you need to manually install, see the [README](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual). Meanwhile I generated a [new release](https://github.com/patman15/BMS_BLE-HA/releases) v1.11.3 which you can install via HACS.

Heila68, 2024-12-31:
Hi Patrick,
thanks a lot for your answers, unfortunately my problem was not solved with this upgrade.
I have found that in my output from "bluetoothctl info <MAC>" the manufactor is not 65535 like it is set in the manifest.json, so I changed it to the reported id of 0, but this also didn't help.
Any other suggestions?
Heinz

patman15, 2024-12-31:
Hi Heinz, ok sorry, I oversaw that the guys changed their manufacturer ID within two years from 0xFFFF to 0x0, both being wrong. :man_facepalming: I adapted the code to detect old and new version in [attached branch](https://github.com/patman15/BMS_BLE-HA/tree/137-keine-unterst%C3%BCtzen-ger%C3%A4te-via-bluetooth-gefunden). Are you able to update according to the [manual install instructions](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual) or do you need a beta release?

Anker-dot, 2024-12-31:
Hi all,

I had the same problem today. I used version 1.10.0 (installed manually). This afternoon the connection was lost.
I deleted the bms_ble folder and reinstalled different versions, including 1.11.3. No changes..
BMS_BLE-HA-137-keine-unte... did not fix the problem, I constantly see the message "No supported devices found via Bluetooth".

patman15, 2024-12-31:
@Anker-dot can you please provide the output of `bluetoothctl info <MAC>`?

Heila68, 2024-12-31:
@patman15 Hi Patrick, your attached brand worked for me perfectly. Thank you very much!
Best regards
Heinz

Anker-dot, 2024-12-31:
.               

> @Anker-dot can you please provide the output of `bluetoothctl info <MAC>`?
> 
> > This afternoon the connection was lost.
> 
> If you lost the connection, this sounds more like a BT issue. Do you have a graph of the `link quality` and/or `RSSI` sensor?

[core-ssh ~]$ bluetoothctl devices
Unable to open mgmt_socket
Device 10:23:81:09:65:99 BP01
Device 5C:6E:1E:20:D3:0B 5C-6E-1E-20-D3-0B
Device 41:AA:66:3B:0E:38 Bowie H1i
Device 18:EB:CD:B4:D5:8A 18-EB-CD-B4-D5-8A
Device 15:C0:A0:1D:FB:75 15-C0-A0-1D-FB-75
Device 61:72:E7:4A:D9:E4 LE_WH-1000XM5
Device C8:47:8C:EB:01:23 JK_B2A24S15P
Device 21:53:E1:9F:14:A9 21-53-E1-9F-14-A9
Device 02:C3:F9:56:66:57 02-C3-F9-56-66-57
Device 2C:FF:3D:B2:43:53 2C-FF-3D-B2-43-53
Device 10:23:81:09:67:E7 BP00
Device 16:AA:22:02:23:45 ANT-BLE16ZMUB
Device 16:AA:22:01:00:08 ANT-BLE16ZMUB-0008
[core-ssh ~]$ 
[core-ssh ~]$ bluetoothctl info 10:23:81:09:67:E7
Unable to open mgmt_socket
Device 10:23:81:09:67:E7 (public)
        Name: BP00
        Alias: BP00
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
        RSSI: 0xffffffae (-82)
        AdvertisingFlags:
  00                                               .               
[core-ssh ~]$ bluetoothctl info C8:47:8C:EB:01:23
Unable to open mgmt_socket
Device C8:47:8C:EB:01:23 (public)
        Name: JK_B2A24S15P
        Alias: JK_B2A24S15P
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Unknown                   (0000ffe0-0000-1000-8000-00805f9b34fb)
        UUID: Tencent Holdings Limited. (0000fee7-0000-1000-8000-00805f9b34fb)
        ManufacturerData Key: 0x0b65 (2917)
        ManufacturerData Value:
  88 a0 c8 47 8c eb 01 23                          ...G...#        
        RSSI: 0xffffffae (-82)
        AdvertisingFlags:
  06                                               .

Anker-dot, 2024-12-31:
I simultaneously restored a copy of the VM from Saturday. There is version 1.10.0 with support for Seplos v2 and everything works. Updated to 1.11.3 and still good. This is weird..

patman15, 2025-01-01:
@Anker-dot so your problem is not related to the Ective BMS type this issue was created for? You seem to have a Seplos and a JK, correct?

Also, from your last comment I can see you have a lot of BT devices around and I assume you are using a Raspberry Pi, is that correct?
If so, please check out the [Home Assistant Bluetooth documentation on performance limitations](https://www.home-assistant.io/integrations/bluetooth/#cypress-based-adapters). It can happen that the adapter filters some messages and thus devices can disappear. Restart can recover/select other devices to be filtered.

patman15, 2025-01-02:
Please open a new issue if you suspect a problem with the integration. To me the error hints more to an issue with the Bluetooth stack / driver. Batmon uses a different version of the Bluetooth stack / library. My integration uses the one that HA or you installed, maybe that gives you a hint.

closed 2025-01-02 by patman15