patman15/BMS_BLE-HA #139: JK BMS Reconnect every...

Prokopchini, 2025-01-03:
### Checklist

- [X] I have tried the latest version of BMS_BLE-HA.
- [X] I have enabled debug logging for my installation.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [X] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

JK-B2A8S20P

### Describe the issue

Hi, I'm back again with the "unavailable" status problem. I've attached the log file. Thanks for your help.

### Debug logs

[home-assistant_bms_ble_2025-01-03T14-23-19.714Z.log](https://github.com/user-attachments/files/18300713/home-assistant_bms_ble_2025-01-03T14-23-19.714Z.log)

patman15, 2025-01-03:
Hey! Thanks, there is indeed another bug that I missed. Although I appreciate that your setup digs out such issues, are you aware that you have very frequent reconnects that should not appear? Did you check the `RSSI` sensor, can you post a diagram of 24hrs? What BT adapter are you using? Looks like you have a very flaky BT connection.

Prokopchini, 2025-01-03:
> Hey! Thanks, there is indeed another bug that I missed. Although I appreciate that your setup digs out such issues, are you aware that you have very frequent reconnects that should not appear? Did [...]

I can't say for 24 hours, I just turned on this parameter. For 10 minutes - the signal level is 84dbm.

Adapter : SENA UD-100

Prokopchini, 2025-01-03:
> I can't say for 24 hours, I just turned on this parameter. For 10 minutes - the signal level is 84dbm.
>
> Adapter : SENA UD-100

strange behavior in device card - rssi and signal level are updated and available, all other readings are "unavailable"

patman15, 2025-01-03:
> strange behavior in device card - rssi and signal level are updated and available, all other readings are "unavailable"

That is ok, because the signal strength is not coming from the BMS but from the system. If you have a quite bad connection, e.g. your -84dBm this is exactly the effect. You should definitely try to improve your wireless connection quality, unless you want to stay my favorite tester. :wink:

Prokopchini, 2025-01-03:
> That is ok, because the signal strength is not coming from the BMS but from the system. If you have a quite bad connection, e.g. your -84dBm this is exactly the effect. You should definitely try to [...]

Improved the signal and everything became stable) Thank you very much!

patman15, 2025-01-03:
Ok, great to hear, I will still fix the issue to make the integration more robust. Thanks for bringing it up!

closed 2025-01-04 by patman15