patman15/BMS_BLE-HA #141: Liontron Batteries

jassi0001, 2025-01-04:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

I have two Liontron Batteries (). The ESPHome Project "github://syssi/esphome-jbd-bms@main" integrated it together with me.
 device_description: "Monitor and control a Xiaoxiang Battery Management System (JBD-BMS) via BLE"

The ESPHome communication is working as expected.
Maybe you can implement it an i can test? 
in the other Project there where timing issues. after changing that, the communication worked....

### Additional context

github://syssi/esphome-jbd-bms@main
https://liontron.com/
https://liontron.com/download/german/LISMART1240LX.pdf and 
https://liontron.com/download/german/LISMART1255LX.pdf
unfortunatly i do not have a ble protocol.

patman15, 2025-01-04:
If it is a JBD then the protocol is already there, so I guess it is only the detection that does not work. Can you provide the output of `bluetoothctl info <MAC>`? Could be a simple update.

jassi0001, 2025-01-04:
this should help: 
[17:32:28][D][esp32_ble_client:110]: [0] [70:3E:97:D1:BE:20] Found device
[17:32:28][D][esp32_ble_tracker:669]: Found device 70:3E:97:D1:BE:20 RSSI=-92
[17:32:28][D][esp32_ble_tracker:690]:   Address Type: PUBLIC
[17:32:28][D][esp32_ble_tracker:692]:   Name: '0223400011'
[17:32:28][D][esp32_ble_tracker:695]:   TX Power: 2
[17:32:28][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[17:32:28][D][esp32_ble_tracker:219]: Pausing scan to make connection...
[...]
[17:32:29][D][esp32_ble_client:110]: [0] [70:3E:97:D1:BE:20] ESP_GATTC_SEARCH_CMPL_EVT
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0x1800
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0x1  end_handle: 0x7
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0x1801
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0x8  end_handle: 0xb
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0x180A
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0xc  end_handle: 0xe
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 0xFF00
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0xf  end_handle: 0x16
[17:32:29][V][esp32_ble_client:223]: [0] [70:3E:97:D1:BE:20] Service UUID: 00010203-0405-0607-0809-0A0B0C0D1912
[17:32:29][V][esp32_ble_client:225]: [0] [70:3E:97:D1:BE:20]  start_handle: 0x17  end_handle: 0x1b
[17:32:29][I][esp32_ble_client:227]: [0] [70:3E:97:D1:BE:20] Connected
[17:32:29][V][esp32_ble_client:069]: [0] [70:3E:97:D1:BE:20]  characteristic 0xFF01, handle 0x11, properties 0x12
[17:32:29][V][esp32_ble_client:069]: [0] [70:3E:97:D1:BE:20]  characteristic 0xFF02, handle 0x15, properties 0x6
[17:32:29][V][esp32_ble_client:121]: [0] [70:3E:97:D1:BE:20] gattc_event_handler: event=18 gattc_if=3
[17:32:29][D][esp32_ble_client:188]: [0] [70:3E:97:D1:BE:20] cfg_mtu status 0, mtu 128
[17:32:29][V][esp32_ble_client:121]: [0] [70:3E:97:D1:BE:20] gattc_event_handler: event=38 gattc_if=3
[17:32:29][D][esp32_ble_client:110]: [0] [70:3E:97:D1:BE:20] ESP_GATTC_REG_FOR_NOTIFY_EVT

bluetoothctl info 

Device 70:3E:97:D1:BE:20 (public)
        Name: 0223400011
        Alias: 0223400011
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Unknown                   (0000ff00-0000-1000-8000-00805f9b34fb)
        ManufacturerData.Key: 0x3e70 (15984)
        ManufacturerData.Value:
  97 d1 be 20                                      ...             
        RSSI: 0xffffffc0 (-64)
        TxPower: 0x003f (63)
        AdvertisingFlags:
  06                                               .

patman15, 2025-01-04:
> Einrichtungsfehler, wird erneut versucht: device communicating failed: 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Failed to connect after 4 attempt(s): TimeoutError (BleakNotFoundError)

Not sure how you reached that state, because that should not happen ...

Can you try the linked branch? That should work. If not, please provide a debug log! Thanks.

jassi0001, 2025-01-05:
Here are the Debug Logs:

2025-01-05 19:54:54.684 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JFJKNJ65RKV2YP17MDRD3227 version=1 domain=bms_ble title=0223400011 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=70:3E:97:D1:BE:20>
2025-01-05 19:54:54.684 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] 0223400011[BE:20]: initializing Jiabaida Smart BMS, BT address: 70:3E:97:D1:BE:20
2025-01-05 19:54:54.684 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for 0223400011 (70:3E:97:D1:BE:20) as Jiabaida Smart BMS
2025-01-05 19:54:54.684 DEBUG (MainThread) [custom_components.bms_ble] 0223400011: advertisement: {'name': '0223400011', 'address': '70:3E:97:D1:BE:20', 'rssi': -58, 'manufacturer_data': {15984: b'\x97\xd1\xbe '}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'BC:2B:02:77:41:62', 'advertisement': AdvertisementData(local_name='0223400011', manufacturer_data={15984: b'\x97\xd1\xbe '}, service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=63, rssi=-58), 'device': BLEDevice(70:3E:97:D1:BE:20, 0223400011), 'connectable': True, 'time': 971911.577254009, 'tx_power': 63}
2025-01-05 19:54:54.684 DEBUG (MainThread) [custom_components.bms_ble] 0223400011: BMS data update
2025-01-05 19:54:54.684 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] 0223400011[BE:20]: connecting BMS
2025-01-05 19:54:54.684 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 1
2025-01-05 19:54:55.475 DEBUG (MainThread) [custom_components.bms_ble] Bluetooth device detected: <BluetoothServiceInfoBleak name=0223400011 address=70:3E:97:D1:BE:20 rssi=-127 manufacturer_data={15984: b'\x97\xd1\xbe '} service_data={} service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'] source=hci0 connectable=True time=971509.582638687 tx_power=None>
2025-01-05 19:54:55.476 DEBUG (MainThread) [custom_components.bms_ble] Bluetooth device detected: <BluetoothServiceInfoBleak name=0923550208 address=70:3E:97:D1:C1:8C rssi=-127 manufacturer_data={15984: b'\x97\xd1\xc1\x8c'} service_data={} service_uuids=['00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000ff00-0000-1000-8000-00805f9b34fb', '00010203-0405-0607-0809-0a0b0c0d1912'] source=hci0 connectable=True time=971509.582638687 tx_power=None>
2025-01-05 19:55:14.686 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Timed out trying to connect (attempt: 1, last rssi: None)
2025-01-05 19:55:14.686 DEBUG (MainThread) [bleak_retry_connector.bluez] 0223400011 - 70:3E:97:D1:BE:20: Waited 2.8953072614967823e-05 seconds to disconnect
2025-01-05 19:55:14.937 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 2
2025-01-05 19:55:34.939 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Timed out trying to connect (attempt: 2, last rssi: None)
2025-01-05 19:55:34.939 DEBUG (MainThread) [bleak_retry_connector.bluez] 0223400011 - 70:3E:97:D1:BE:20: Waited 5.365500692278147e-05 seconds to disconnect
2025-01-05 19:55:35.190 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 3
2025-01-05 19:55:55.192 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Timed out trying to connect (attempt: 3, last rssi: None)
2025-01-05 19:55:55.192 DEBUG (MainThread) [bleak_retry_connector.bluez] 0223400011 - 70:3E:97:D1:BE:20: Waited 3.07250302284956e-05 seconds to disconnect
2025-01-05 19:55:55.443 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 4
2025-01-05 19:55:59.832 DEBUG (MainThread) [custom_components.bms_ble] Bluetooth device detected: <BluetoothServiceInfoBleak name=0923550208 address=70:3E:97:D1:C1:8C rssi=-54 manufacturer_data={15984: b'\x97\xd1\xc1\x8c'} service_data={} service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'] source=BC:2B:02:77:41:62 connectable=True time=971976.286673571 tx_power=63>
2025-01-05 19:55:59.855 DEBUG (MainThread) [custom_components.bms_ble] Device 0923550208 (70:3e:97:d1:c1:8c) detected as 'Jiabaida Smart BMS'
2025-01-05 19:55:59.855 DEBUG (MainThread) [custom_components.bms_ble] confirm step for 0923550208
2025-01-05 19:56:15.445 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Timed out trying to connect (attempt: 4, last rssi: None)
2025-01-05 19:56:15.445 DEBUG (MainThread) [bleak_retry_connector.bluez] 0223400011 - 70:3E:97:D1:BE:20: Waited 3.1574047170579433e-05 seconds to disconnect
2025-01-05 19:56:15.696 DEBUG (MainThread) [custom_components.bms_ble] 0223400011: device communicating failed: 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Failed to connect after 4 attempt(s): TimeoutError (BleakNotFoundError)
2025-01-05 19:56:15.696 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching 0223400011 data in 81.012 seconds (success: False)
2025-01-05 19:56:15.696 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS device (0223400011)
2025-01-05 19:56:15.697 DEBUG (MainThread) [custom_components.bms_ble] Bluetooth device detected: <BluetoothServiceInfoBleak name=0223400011 address=70:3E:97:D1:BE:20 rssi=-58 manufacturer_data={15984: b'\x97\xd1\xbe '} service_data={} service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'] source=BC:2B:02:77:4 [... cut ...]
2025-01-05 19:56:47.167 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 1
2025-01-05 19:57:07.169 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Timed out trying to connect (attempt: 1, last rssi: None)
2025-01-05 19:57:07.169 DEBUG (MainThread) [bleak_retry_connector.bluez] 0223400011 - 70:3E:97:D1:BE:20: Waited 4.9202004447579384e-05 seconds to disconnect
2025-01-05 19:57:07.421 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 2

patman15, 2025-01-05:
Seems like BT has troubles connecting, so far I could not spot an issue with the integration. Can you please
- make sure nothing else is connected to the batteryies, e.g. ESPHome
- wait at least 5 minutes
- attach the log file of those 5 minutes?

jassi0001, 2025-01-06:
Hi, the problem seems to be the same as in the ESPHome Project. There where Timing Problems while getting data, which could be solved there... i don't know, if you find it in the code.... i linked the artikel in my first post....
When i connect the ESPHome installation, it connects just as i start the connection. Xiaoxing App on Iphone as well (to both Batteries)

2025-01-06 07:27:10.106 DEBUG (MainThread) [custom_components.bms_ble] Bluetooth device detected: <BluetoothServiceInfoBleak name=0923550208 address=70:3E:97:D1:C1:8C rssi=-52 manufacturer_data={15984: b'\x97\xd1\xc1\x8c'} service_data={} service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'] source=BC:2B:02:77:41:62 connectable=True time=1013447.31705354 tx_power=63>
2025-01-06 07:27:10.106 DEBUG (MainThread) [custom_components.bms_ble] Device 0923550208 (70:3e:97:d1:c1:8c) detected as 'Jiabaida Smart BMS'
2025-01-06 07:27:10.107 DEBUG (MainThread) [custom_components.bms_ble] confirm step for 0923550208
2025-01-06 07:28:06.433 DEBUG (MainThread) [custom_components.bms_ble] confirm step for 0223400011
2025-01-06 07:28:08.648 DEBUG (MainThread) [custom_components.bms_ble] confirm step for 0223400011
2025-01-06 07:28:08.648 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JGX4S8T88JXGVGZ3715N38NN version=1 domain=bms_ble title=0223400011 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=70:3E:97:D1:BE:20>
2025-01-06 07:28:08.648 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] 0223400011[BE:20]: initializing Jiabaida Smart BMS, BT address: 70:3E:97:D1:BE:20
2025-01-06 07:28:08.648 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for 0223400011 (70:3E:97:D1:BE:20) as Jiabaida Smart BMS
2025-01-06 07:28:08.648 DEBUG (MainThread) [custom_components.bms_ble] 0223400011: advertisement: {'name': '0223400011', 'address': '70:3E:97:D1:BE:20', 'rssi': -60, 'manufacturer_data': {15984: b'\x97\xd1\xbe '}, 'service_data': {}, 'service_uuids': ['0000ff00-0000-1000-8000-00805f9b34fb'], 'source': 'BC:2B:02:77:41:62', 'advertisement': AdvertisementData(local_name='0223400011', manufacturer_data={15984: b'\x97\xd1\xbe '}, service_uuids=['0000ff00-0000-1000-8000-00805f9b34fb'], tx_power=63, rssi=-60), 'device': BLEDevice(70:3E:97:D1:BE:20, 0223400011), 'connectable': True, 'time': 1013500.902369023, 'tx_power': 63}
2025-01-06 07:28:08.649 DEBUG (MainThread) [custom_components.bms_ble] 0223400011: BMS data update
2025-01-06 07:28:08.649 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] 0223400011[BE:20]: connecting BMS
2025-01-06 07:28:08.649 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 1
2025-01-06 07:28:28.650 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Timed out trying to connect (attempt: 1, last rssi: None)
2025-01-06 07:28:28.650 DEBUG (MainThread) [bleak_retry_connector.bluez] 0223400011 - 70:3E:97:D1:BE:20: Waited 2.816098276525736e-05 seconds to disconnect
2025-01-06 07:28:28.902 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 2
2025-01-06 07:28:48.903 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Timed out trying to connect (attempt: 2, last rssi: None)
2025-01-06 07:28:48.903 DEBUG (MainThread) [bleak_retry_connector.bluez] 0223400011 - 70:3E:97:D1:BE:20: Waited 4.1215913370251656e-05 seconds to disconnect
2025-01-06 07:28:49.155 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 3
2025-01-06 07:29:09.157 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Timed out trying to connect (attempt: 3, last rssi: None)
2025-01-06 07:29:09.157 DEBUG (MainThread) [bleak_retry_connector.bluez] 0223400011 - 70:3E:97:D1:BE:20: Waited 2.8908951207995415e-05 seconds to disconnect
2025-01-06 07:29:09.408 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 4
2025-01-06 07:29:29.410 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Timed out trying to connect (attempt: 4, last rssi: None)
2025-01-06 07:29:29.410 DEBUG (MainThread) [bleak_retry_connector.bluez] 0223400011 - 70:3E:97:D1:BE:20: Waited 4.640000406652689e-05 seconds to disconnect
2025-01-06 07:29:29.661 DEBUG (MainThread) [custom_components.bms_ble] 0223400011: device communicating failed: 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Failed to connect after 4 attempt(s): TimeoutError (BleakNotFoundError)
2025-01-06 07:29:29.661 DEBUG (MainThread) [custom_components.bms_ble] Finished fetching 0223400011 data in 81.013 seconds (success: False)
2025-01-06 07:29:29.662 DEBUG (MainThread) [custom_components.bms_ble] Shutting down BMS device (0223400011)
2025-01-06 07:29:35.050 DEBUG (MainThread) [custom_components.bms_ble] Setup of <ConfigEntry entry_id=01JGX4S8T88JXGVGZ3715N38NN version=1 domain=bms_ble title=0223400011 state=ConfigEntryState.SETUP_IN_PROGRESS unique_id=70:3E:97:D1:BE:20>
2025-01-06 07:29:35.050 DEBUG (MainThread) [custom_components.bms_ble.plugins.jbd_bms] 0223400011[BE:20]: initializing Jiabaida Smart BMS, BT address: 70:3E:97:D1:BE:20
2025-01-06 07:29:35.050 DEBUG (MainThread) [custom_components.bms_ble] Initializing coordinator for 0223400011 (70:3E:97:D1:BE:20) as Jiabaida Smart BMS
[... cut ...]
2025-01-06 07:34:48.144 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 1
2025-01-06 07:35:08.147 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Timed out trying to connect (attempt: 1, last rssi: None)
2025-01-06 07:35:08.147 DEBUG (MainThread) [bleak_retry_connector.bluez] 0223400011 - 70:3E:97:D1:BE:20: Waited 3.4476048313081264e-05 seconds to disconnect
2025-01-06 07:35:08.398 DEBUG (MainThread) [bleak_retry_connector] 70:3E:97:D1:BE:20 - 70:3E:97:D1:BE:20: Connection attempt: 2

patman15, 2025-01-06:
> Hi, the problem seems to be the same as in the ESPHome Project. There where Timing Problems while getting data, which could be solved there... i don't know, if you find it in the code.... i linked the artikel in my first post....

I have had a look at the issue, but I can't see the relation (yet). The logs show that HA is unable to even make a connection. The integration doesn't reach the state to send a single command. For me it looks like either
- the battery requires pairing, 
- you still have another device connected to it, or 
- there is something really strange that I'm not aware of.

Unfortunately, I do not see much because even the connect command fails.
Which HA version and what BT adapter are you using? Did you check that the device is visible using `bluetoothctl devices`? Can you connect using `bluetoothctl connect <MAC>` ?

jassi0001, 2025-01-06:
hci0 70:3E:97:D1:C1:8C type LE Public connect failed (status 0x08, Timeout)
Pairing is not neccesary .... i do not know, what excactly the timing Problem was and how it was solved, but i can ask.... 
There is no other device connected and when i try with the iphone app, connect ist established prompt... Same with ESPHome....

patman15, 2025-01-08:
@jassi0001 I talked to @syssi yesterday. I cannot see any relation between the changes that were done and the current issue. Given the fact that you cannot connect using `bluetoothctl` strongly hints at an issue with the Bluetooth stack.
Can you please give more details:
- `bluetoothctl devices` to see how many others are around
- BT adapter you are using
- Home Assistant version
- Home Assistant installation type
- Are you using any other BT devices with Home Assistant that work?

jassi0001, 2025-01-09:
bluetoothctl devices to see how many others are around
BT adapter you are using Realtec mapped through proxmox to a HA virtual installation.
Home Assistant version : 2025.1
Home Assistant installation type . Proxmox virtual HAOS
Are you using any other BT devices with Home Assistant that work? you're right . i activated it only for battery monitor. So at the moment no other connected and working.
bluetoothctl devices shows up 15 Devices arround... but i can not connect to one of them. all timeout. I will check here... 

as you said, it is a problem with the adapter or the stack....

jassi0001, 2025-01-09:
problem seems to be inside the HAOS... connection from Proxmox works.... i Think you can close here and i have to check in my installation

patman15, 2025-01-09:
Closing is up to the reporter, unless I fixed something. :smile: Close at your needs, I'm happy to help you with further input, if you have questions. If you haven't done so, I recommend to "pass-through" the whole USB device to the VM. At least for me the setup works.
You also might want to take a look at the [high performance adapters recommended by HW](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters)

jassi0001, 2025-04-14:
Hello again,

i tried the Batterie now in a different environment. Connection to both works now but after a short time the batterie goes offline. After activaton the first time data is read out. After offline and try to reinitialise ha tries to reinitialize but gers timeout. 
I have an android handy in place as well....  i used the jbd_bms.py file from the branch you created and copied the identification information to a jdb_bms from version 1.15. With this the batteries were discovered.
Overall i have a timing problem with the batteries... so how can i debug here, now?
Hope you can give me some infos.

patman15, 2025-04-16:
@jassi0001 I reopened this issue and updated the [branch](https://github.com/patman15/BMS_BLE-HA/tree/141-liontron-batteries) to the latest state to continue. The I would need a debug log to see what's going on.

patman15, 2025-04-17:
Hi! Sorry, but these were not the log files I was looking for. See the [troubleshooting guide](https://github.com/patman15/BMS_BLE-HA#in-case-you-have-troubles-youd-like-to-have-help-with) on how to enable debugging for the integration.
Still we might need your capabilities in case this log does not provide the necessary information.
(Btw. I'd like to stay in English for the community)

jassi0001, 2025-04-17:
Thanks for your response... here are the debug logs from HA

[home-assistant_bms_ble_2025-04-17T12-55-30.461Z.log](https://github.com/user-attachments/files/19796596/home-assistant_bms_ble_2025-04-17T12-55-30.461Z.log)

patman15, 2025-04-17:
@jassi0001, you wrote:

> BT adapter you are using Realtec mapped through proxmox to a HA virtual installation.

See this [warning](https://www.home-assistant.io/integrations/bluetooth/#realtek-rtl8761bu-adapters)!
Further, the device `0223400011` has an `rssi=-93` which incredibly bad, see the [reference table](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#provided-information).
If you want to get the Realtek working, you need to reboot the host! Note it will happen again. Consider a USB-BT adapter with [high-performance](https://www.home-assistant.io/integrations/bluetooth/#known-working-high-performance-adapters)! You can get them for <10€, as mentioned in https://github.com/patman15/BMS_BLE-HA/issues/141#issuecomment-2580142670. 😃

jassi0001, 2025-04-17:
It is a different Environment now. It is Running in HAOS on Raspberrypi and an esp-btproxy.

The Batterie with 0223….. is gar away from the Adapter. The one with 09xxxx is near…. But the behavior is the Same….

patman15, 2025-04-17:
Ok, but are you still using the Realtek adapter? It's known for doing strange things.

patman15, 2025-04-17:
I don't know how to help you, I'm pretty sure it is a BT setup issue. I can see from the log 2 things:
1. the two batteries come in via different adapters
2. `0223400011` battery is once received correctly, which let's me conclude that there is no obvious SW issue

I would recommend to deactivate the internal BT adapter of the RaspberryPi to avoid interferences. Maybe that helps.

jassi0001, 2025-04-18:
Hi,
i changed the setup no: HAOS in Raspberry, ESPHome with BTProxy, BT on Raspi disabled. 
 
log shows: (Attach).

I had this in the other project (github://syssi/esphome-jbd-bms@main) as well, that no stable connection could be established. After we changed the Configuration like this: it worked with that project without disconnection. Thats what i ment with timing Problems. The Liontron Batteries seem to have a Problem there....

jbd_bms_ble:
  - id: bms0
    ble_client_id: client0
    update_interval: 10s

[LOG.txt](https://github.com/user-attachments/files/19810472/LOG.txt)

patman15, 2025-04-18:
What I can tell you from this log:
- `rssi: -69` -> ok
- first attempt, timeout
- BT connection is lost
- `rssi: -70` -> ok
- second attempt, success
- third attempt, no response

> After we changed the Configuration like this: it worked with that project without disconnection. Thats what i ment with timing Problems. The Liontron Batteries seem to have a Problem there....

Ok, this is the time between the attempts, but that is anyway 30s in this case.

I now increased the timeout for individual messages from 2.5s to 5s. Can you update, or just change the value like in this commit https://github.com/patman15/BMS_BLE-HA/commit/c27d934c0e40dabe313bd7e937fd0f922b53a3dd ?
Also can you try to ping your BT proxy and see how large this value is?
If you have discord, you can also ping me there to do a live session if you want.

Happy Easter,
Patrick

jassi0001, 2025-04-18:
Hi Patrick,

i changed the value without success. If you like, we can do a live session and i can show you some things. I have installed the syssi Project on esp, parallel with proxy on the same system. some times your project works for one cycle, most times not. Sissy works everytime.....

i asked to contact via discord.

patman15, 2025-04-18:
> I have installed the syssi Project on esp, parallel with proxy on the same system.

I would guess that this causes the issue because you cannot access the battery twice so the proxy and syssi project compete over the device. Please do either or!

jassi0001, 2025-04-18:
discord contact is patman15 ? Mine is jassi0001

patman15, 2025-04-18:
No, wasn't free anymore -> https://discordapp.com/users/693860058479722517

jassi0001, 2025-04-18:
hmm... think we have to use the same servers? I click on the link, but no name is shown in discord ? Strange....

patman15, 2025-04-18:
I'm on the HA server: https://discord.gg/7hx9S8k7

patman15, 2025-04-18:
Ok, just to note it down: the battery seems to go in stand-by when not queried often enough. Timeout 9s did work.

closed 2025-04-19 by patman15