patman15/BMS_BLE-HA #142: No supported devices found via Bluetooth.

ivankovff, 2025-01-04:
### Checklist

- [X] I have tried the latest version of BMS_BLE-HA.
- [X] I have enabled debug logging for my installation.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issue](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3Abug).
- [X] I have filled out the issue template to the best of my ability.

### Provide the type of BMS you used:

Seplos 3v

### Describe the issue

I can't run the add-on. I have a similar problem. In the terminal I see my bms but nothing happens.
[image]

### Reproduction steps

```text
1.installed the app
2.checked in the terminal
3.device missing
...
```

### Debug logs

The journal cannot be started.

ivankovff, 2025-01-04:
usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 537, in handle_manifest_list
    raise int_or_exc
homeassistant.loader.IntegrationNotFound: Integration 'ble_monitor' not found.
2025-01-05 00:29:57.741 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140332633615968] Error handling message: Unknown error (unknown_error)

ivankovff, 2025-01-04:
[image]

patman15, 2025-01-05:
Hi! Please [manually install](https://github.com/patman15/BMS_BLE-HA?tab=readme-ov-file#manual) the [attached branch](https://github.com/patman15/BMS_BLE-HA/tree/142-Seplos-v3-SP51-pattern-not-recognized) and provide me a debug log for verification.
Your device name `SP51...` was not recognized as valid. I added it to the list of supported devices now.

ivankovff, 2025-01-05:
Thank you very much.
This helped.

patman15, 2025-01-05:
Ok, great, I will integrate that with the next release.

closed 2025-01-10 by patman15