patman15/BMS_BLE-HA #145: daly bms smart support ( blue version )

bxmatus, 2025-01-10:
### Checklist

- [X] I need support with using the integration.
- [X] My issue is not a [feature request](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Feature%2BRequest&projects=&template=feature_request.yml)
- [X] I'm not avoiding to fill out the [bug report](https://github.com/patman15/BMS_BLE-HA/issues/new?assignees=&labels=Bug&projects=&template=bug.yml) form.

### Describe the issue

Hi does this software support blue daly bms ( with active balancing ) ? Add device showing no supported devices, thanjz

patman15, 2025-01-10:
Hi! No, currently not. If you own such a device and are willing to help me gather the necessary information, I can most probably add it.
See the feature request template for what I would need. Starting point would be an output of `bluetoothctl info <MAC>`. Assuming the protocol is similar to the implemented one, no traces required at the moment.
Best regards, Patrick

bxmatus, 2025-01-10:
Hi, thanks for answer, im sending requested into:

➜  ~ bluetoothctl info 50:18:12:01:A3:B3 
Device 50:18:12:01:A3:B3 (random)
        Alias: 50-18-12-01-A3-B3
        Paired: no
        Bonded: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        ManufacturerData.Key: 0x0104 (260)
        ManufacturerData.Value:
  01 50 18 12 01 a3 b3 4a 48 42                    .P.....JHB      
        RSSI: 0xffffffd2 (-46)
        AdvertisingFlags:
  06                                               .               
➜  ~ 

device name in bluetooth start with "JHB-(mac)"

patman15, 2025-01-11:
@bxmatus, I enabled the detection of the BMS, not sure how it will react, but let's give it a try. Please follow the [manual installation instructions](https://github.com/patman15/BMS_BLE-HA#manual) and install the [linked branch](https://github.com/patman15/BMS_BLE-HA/tree/145-daly-bms-smart-support-blue-version). Please provide me a debug log to either find a, likely to occur, issue or to verify that everything is fine. Thanks!

bxmatus, 2025-01-11:
Hello, thanks for response.
Branch is working, i'm able to see data ( i'm not near the device so i can't confirm values yet - will do it later ) but generally it looks great !
Few questions:
- Device does have 2 temperature sensors but i can see only one value. Can i send you some data dump so you can add it ?
- I'm not able to see voltage of individual cells, BMS is providing it for sure, is it possible to add it there ?

thank you very much

bxmatus, 2025-01-11:
This is debug output from unit:

BMS data sample {'temp#0': 3.0, 'voltage': 13.3, 'current': 7.9, 'battery_level': 71.8, 'cycle_charge': 28.7, 'cell_count': 4, 'temp_sensors': 2, 'cycles': 0, 'delta_voltage': 0.001, 'temp#1': 0.0, 'temp#2': 0.0, 'cell#0': 3.344, 'cell#1': 3.343, 'cell#2': 3.344, 'cell#3': 3.344, 'cycle_capacity': 381.71, 'power': 105.07, 'battery_charging': True, 'temperature': 1.0}

patman15, 2025-01-11:
> Device does have 2 temperature sensors but i can see only one value. Can i send you some data dump so you can add it ?

Yes, please provide me a complete debug log file (attach it) and the values they should have. I can see it detects 2 sensors, but obviously the values are wrong. I need the log to check why.

> I'm not able to see voltage of individual cells, BMS is providing it for sure, is it possible to add it there ?

See the [README](https://github.com/patman15/BMS_BLE-HA#provided-information), they are as attribute to the `delta_voltage` sensor.

> can you add temporarily all temperature values into HA

All detected are, similar to the cell voltages, as attribute to the `temperature` sensor. There seems to be a protocol mismatch, so I need the log and the values they should have (from the app).

Best regards,
Patrick

Tom-T, 2025-01-19:
[home-assistant_bms_ble_2025-01-19T05-23-41.727Z.log](https://github.com/user-attachments/files/18467788/home-assistant_bms_ble_2025-01-19T05-23-41.727Z.log)

I have three, using proxy bluetooth.
I'm not sure if it matters, but I've previously renamed them JHB-1 JHB-2 and JHB-3 in the app and that seems to be the same naming here. I've removed from HACS and added your custom branch. Included is my log file, sorry my HA is a mess.

It looks like I may only be seeing the main one, not all three. I have them in parallel. 

[image]

I really appreciate you adding this new 

I'd be happy to do any testing when I have time if OP can't get to it.

Tom-T, 2025-01-19:
Looks like they did eventually show data

[image]

patman15, 2025-01-19:
> Looks like they did eventually show data
> 
The logs and reported values look ok to me. Your signal strength is a bit low. In combination with the 3 devices per proxy limit, your BT is supposedly a bit on the edge, which could explain the long time till the devices were redetected.
So far nothing I see as an issue of the integration, but related to your BT setup.

Do you have any additional issues/questions?

Tom-T, 2025-01-19:
Using the debug log I can confirm cell voltages0-15, temp1, temp2, as well as the commonly referenced data such as SOC. I couldn't confirm an exact match, as I had to disconnect and reconnect. However the values seemed to be what one would expect from moment to moment. 

It would be great to see it added in the future version. 

In the near future I hope to add cell voltage displays. I see readme section on it, but I'm not sure when I'll have the time and patience to do that soon. It seems complicated and I'm trying to not overdo it today. :)

Thanks for the very quick reply and update!

Tom-T, 2025-01-19:
Update:

The process is well documented, and was easy enough to follow. I'll eventually make it look pretty (or not)

Thanks again.
[image]

patman15, 2025-01-19:
> The process is well documented, and was easy enough to follow.

Great to hear, thanks for the feedback! Since the values are ok for you, I'll soon integrate it to the release.
@bxmatus I would be still interested in a debug log from you, since you had issues with the values. At the moment there is at least one known good.

bxmatus, 2025-01-21:
Hello, implementation is working perfectly fine. unit is reporting two temperatures ( showing right now as attribute ) and "main temperature value" which is just overal temperature made from both sensors.  Implementation is now grabing data every 30 seconds. Is it possible for user to change this value ?

thanks

m.

patman15, 2025-01-21:
Thanks for the feedback!

> Implementation is now grabing data every 30 seconds. Is it possible for user to change this value ?

You can, by editing https://github.com/patman15/BMS_BLE-HA/blob/main/custom_components%2Fbms_ble%2Fconst.py#L29 but it will be overwritten by updates. Minimum is 5s enforced by HA, real minimum depends on link quality and other factors.

closed 2025-01-25 by patman15