patman15/BMS_BLE-HA #149: New parameters - Charge Status -  in latest JK BMS Firmware and app

downset, 2025-01-11:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

In the latest BMS Firmware (since V15.35 - now i am running V15.38) there are some new Parameters:

-Charge Status (Bulk-Absorbtion-Float)
-Charge Status Time (in Sec, indicates the time in Absorbtion mode)
-And the Status of the two dry Contacts (not interesting for me personal, but listed for compleatness)

[image]
Here you can see them at the bottom of the pic

Here a debug log from newest 15.38 where i reactivated both BMS at the beginning to have a full connecting cycle in the log
[home-assistant_bms_ble_2025-01-11T09-31-35.657Z.log](https://github.com/user-attachments/files/18385275/home-assistant_bms_ble_2025-01-11T09-31-35.657Z.log)

I assume you need more information for implementing, but this is all i have at the moment, maybe you can read something out of that.

Would be nice to log the charge status parameters for debug an fine tune the absorption timer settings

Best regards
downset

patman15, 2025-01-11:
> Charge Status (Bulk-Absorbtion-Float)

Makes sense to me. What about adding it as an attribute to the `battery_charging` sensor? Would that be fine with you?

The others sound too geeky and are very unique to the JK.

downset, 2025-01-11:
Fine for me as Attribute

Would like to see the Timer too but I understand the uniqueness argument :)

greets
downset

patman15, 2025-01-11:
Ok, I managed to fake BMS to the app, but my app does not seem to show these parameters. Where did you get it from?

downset, 2025-01-11:
I am using IOS, there it was updated on the 5.1.25 to V4.30 direkt from the App store

patman15, 2025-01-11:
Hmm, I fear, I cannot do much until the Android app is updated. Unless you are able to provide me 3 debug logs each for one of the states `Bulk-Absorbtion-Float`. Then I might be able to guess it from the binary data.

downset, 2025-01-22:
I just found a source for the 4.3 android beta APK

https://www.pgyer.com/Cfq3

it is linked here:
https://diysolarforum.com/threads/jkbms-inverter-edition-problems-issues-no-support-help-to-fix-major-issues-do-not-buy-warning-as-of-oct-12-2024.88573/page-43

and as wrtten there, it support the new parameters.

Greets 
downset

patman15, 2025-01-23:
Thanks @downset will have a look next week!

patman15, 2025-02-02:
Status update: I'm having troubles getting my JK emulation working, thus I cannot decode the new status flag. If your @downset or someone else has the patience to generate **separated** debug logs for the three states `Bulk-Absorbtion-Float` it would help / speed-up the implementation quite a lot.
Will continue fixing my emulation ...

downset, 2025-02-12:
Yesterday I installed another Battery to the System and used the chance to get some logs as I had to equilize the new Bat with the rest and charged all up from Grid :)

just look at the first Battery "BatteryOben-00" as it seems only the Master reports the charge stage not the slave Batterys, or at least the IOS App only shows it for the Master

[JK Float.log](https://github.com/user-attachments/files/18769427/JK.Float.log)
[JK Adsorb.log](https://github.com/user-attachments/files/18769428/JK.Adsorb.log)
[JK Bulk.log](https://github.com/user-attachments/files/18769426/JK.Bulk.log)

Hope this Helps
Best regards
downset

patman15, 2025-02-13:
Had a first look, but it's not that obvious. Still can't get my emulator working to play around.
Will continue to try, let's see if others are successful: https://github.com/syssi/esphome-jk-bms/issues/672

syssi, 2025-04-12:
@downset Could you provide another set of logs? It looks like your first capture wasn't accurate and doesn't included the labeled state.

downset, 2025-04-12:
do you need all three states new or one special?

downset, 2025-04-12:
Here is a new Bulk, that i could generate instant :)

[Bulk.log](https://github.com/user-attachments/files/19718619/Bulk.log)

syssi, 2025-04-12:
I would be happy about all state for additional confidence

downset, 2025-04-12:
ok I will try... today is the last sunny day for the next Week so I have also to charge my Car :)
I don`t think that I get the batteries to 100% and Float today maybe i can get an absorbtion phase

downset, 2025-04-28:
Here is an adsorbtion phase as requested @syssi 
"BatteryOben-00:D5C5" is the master which is showing the state in the APP
maybe i can get also a float today

[adsorbtion.log](https://github.com/user-attachments/files/19940028/adsorbtion.log)

downset, 2025-04-28:
and finally here is the float: @syssi 

[float.log](https://github.com/user-attachments/files/19941232/float.log)

syssi, 2025-05-27:
@downset Good job!

This is the implementation at `esphome-jk-bms`: https://github.com/syssi/esphome-jk-bms/pull/746

```
    // 246+32  2   0x00 0x00              Charge status time
    this->publish_state_(this->charge_status_time_elapsed_sensor_, (float) jk_get_16bit(246 + offset));

    // 248+32  1   0x00                   Charge status                                0x00: Bulk
    //                                                                                 0x01: Absorption
    //                                                                                 0x02: Float
    this->publish_state_(this->charge_status_id_sensor_, (float) data[248 + offset]);
    this->publish_state_(this->charge_status_text_sensor_, this->charge_status_id_to_string_(data[248 + offset]));
```

patman15, 2025-05-27:
@downset I implemented the hint from @syssi in the branch attached to this ticket. Can you try? You should see an attribute to the charging sensor.

closed 2025-05-29 by patman15