patman15/BMS_BLE-HA #150: JK BMS Control

Silverwings007, 2025-01-12:
### Checklist

- [X] I have filled out the template to the best of my ability.
- [X] I have understood that this integration is only about battery management systems with Bluetooth Low Energy support.
- [X] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [X] This issue is not a duplicate of any [previous feature requests](https://github.com/patman15/BMS_BLE-HA/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Please describe the enhancement you would like to have.

I've just started using your integration on my JK BMS Jk-pb2a16s20p

All working well so far!

I would love to see control of the charge/balance/discharge of the BMS via toggle switches. BATMON can do this, not sure if it's possible to copy the core? Do you think this might be possible?

Silverwings007, 2025-01-14:
Hey, thanks for the detailed reply. My particular use case would be to turn off the charging on the BMS (via automation) so that all my solar gets exported and not used to charge the battery. It wouldn't be used in the safety case.

If it was relatively easy to implement, could it be used in a  'do not rely on for safety' case? Or 'experimemtal' perhaps? Thanks

patman15, 2025-01-15:
I can check how much effort it is, as I would need to implement the new platform and more importantly an information flow in the other direction HA -> BMS. The use-case would be (as anyway in the license terms) with no liability etc.
Most important I could probably do a patch for you to maintain (or do you maybe want to start a fork with adding the functionality?) in the sense, that future versions (of the HW, FW or integration) could break it. I fear that if I open that bottle, a lot of people will request further settings from the app and honestly, I'm sure I will lack the time to maintain it. It's not only the code itself, but the current version [runs 283 tests](https://github.com/patman15/BMS_BLE-HA/actions/runs/12725709753/job/35473251129) and I do not consider them "enough" to avoid issues when changing code.

So long story short, I can have a look, maybe do a one-time patch for you and you can apply it yourself, but I won't put it into any release as I (apart from the risk that people use it wrong) do not want to promise things I cannot keep up for a long period of time, i.e. maintenance.

Let me check that next week, I'm also working on a `fault` sensor so, my capacity is currently booked. :wink:

Silverwings007, 2025-01-15:
Anything you can do would be amazing. I don't mind doing a test or simply a one off patch if you get the time.

I think we all really appreciate the time you put into this.

Thanks

patman15, 2025-01-16:
Thanks for the motivation, will check aligned with my availability.

vchaban88, 2025-02-03:
Hi I have a problem with the temperature sensor on the JK BMS. Maybe someone know how to solve it? 
I have several JK BMS conected,  some work well, but others - incorrect temperature sensore value ~ 2550 C`

philtao-dev, 2025-03-10:
Hi, I have the same need to be able to control charge/discharge for automation purpose (all the safety aspects are handled via CANbus and Victron CerboGX).
I am using your integration with two JK BMS (JK-PB2A16S20P) in parallel and it has worked flawlessly for the last two months.
The data generated from your integration and Grafana reporting made me aware that when the JK BMS goes into "Float" mode it creates an instability with battery current oscillating between -15A (discharge) to +15A (charge) every second or so... which cannot be good for the battery.  therefore I need to disable charge once the battery is reported as full (via CANbus)... until SOC goes below a set point. 
The other use case is to stop charging the battery at a set SOC and inject all surplus to the grid (also based on spot selling price that fluctuates throughout the day).

Have put more thoughts into it?
I will check with JK the possibity to fix that on their side... and I will also consider getting back into Python coding

patman15, 2025-03-11:
> Have put more thoughts into it?

It's still on my list to do as one-time effort (not integrated into a release), when I have enough time. So far updates consumed all my available spare-time ...

stomko11, 2025-03-14:
Found this thread as I am interested in the same thing. I am not using batmon as it requires BT adapter, and I already have esphome proxies around the house.

patman15, 2025-03-15:
Hi guys!

I spent this rainy day to create a patch to enable JK BMS control.

> [!CAUTION]
> This is completely untested, my harm your battery, you, others, *ABSOLUTELY* no warranty, liability if you use it. THE ENTIRE RISK AS TO THE QUALITY AND PERFORMANCE OF THIS PATCH IS WITH YOU.
> I cannot keep this up-to-date, so please do not expect maintenance for it, I **will not** include it in a release, so any update of might render the patch unusable.

> [!WARNING]
> There are multiple firmware versions of the JK BMS and I do not know for which one this code could work.

That being said, have fun (patch is on top of version 1.13.0) and let me know if it was worth spending that big amount of time: [JK_ctrl.zip](https://github.com/user-attachments/files/19265400/JK_ctrl.zip)

stomko11, 2025-03-16:
Can you provide any instructions on how to apply this?

philtao-dev, 2025-03-16:
This is what I have done to apply the patch... 

1. extract the patch file from zip: JK_ctrl.patch
2. copy the file "JK_ctrl.patch" in the "homeassistant" directory (I use mobaXterm to do that)
3. ssh from the "homeassistant" directory run : git apply JK_ctrl.patch

but I get an error as the file "switch.py" does not exist so cannot be patched.

patman15, 2025-03-16:
@philtao-dev Sorry, the patch was not fully correct. I fixed that and updated to 1.14.0. Should work now.

philtao-dev, 2025-03-16:
Thanks a lot for that work.  I really apreciate it 🥇 
Patch installed fine on 1.14.0
The switches are there and quick test show them as operational.
I will do some further testing in the next few days and report here

stomko11, 2025-03-17:
for some reason, this is not working for me. I updated to 1.14.0, however when I try to apply patch, I get following:

➜  /homeassistant git apply JK_ctrl.patch 
error: patch failed: custom_components/bms_ble/manifest.json:172
error: custom_components/bms_ble/manifest.json: patch does not apply
error: custom_components/bms_ble/switch.py: No such file or directory

I do not see switch.py under /bms_ble/ directory.

philtao-dev, 2025-03-17:
Have you re-downloaded the **JK_ctrl.zip** file since @patman15 has updated it 19 hours ago?

stomko11, 2025-03-19:
You are right. I have downloaded new version, but accidentaly uploaded previous one. Now patch was installed without issues.

nelgi, 2025-03-19:
Thank you for this great piece of software, i have installed the patch. The buttons appeared:
[image]

but after a reboot, without toggling the switches they have unknown state:
[image]

I have JKBMS (JK_BD6A2+S15P), HW: V11A, SW: V11.54

My BMS's are connected through BT-PROXY

MikeHotel0815, 2025-03-19:
Thank you so much, for taking the time to code this patch! The installation worked flawlessly liek @philtao-dev stated in his post. I had to restart HA though.

The switches showed up and changed to "toggle" switches. 

[image]

As i am on my battery right now, i have to wait to test the "discharge" button. Althoug the "balancing" switch works fast and reliable, the "charge" button changes position, but switches back after a few seconds with no effect on the BMS at all. I'll check the "charge" tomorrow.

patman15, 2025-03-19:
> without toggling the switches they have unknown state

@nelgi, correct, the **state** of the switches **is not queried**, the patch just allows you to actively set the state. This keeps the patch maximal independent of the other code. @MikeHotel0815 this is also why I do not recommend to change the switch type.

MikeHotel0815, 2025-03-19:
[@MikeHotel0815](https://github.com/MikeHotel0815) this is also why I do not recommend to change the switch type.

I did not change the type. They changed on first click automatically.

patman15, 2025-03-19:
> I did not change the type. They changed on first click automatically.

@MikeHotel0815, true, this was an issue with reloading the integration, due to the patch. I updated the patch in the original post.

patman15, 2025-03-21:
The easiest way is to install version 1.14.0 of the integration the way you have done it last time and re-apply the patch.

MikeHotel0815, 2025-03-22:
> The easiest way is to install version 1.14.0 of the integration the way you have done it last time and re-apply the patch.

I did that. I removed my battery and removed the Integration via HACS (three dots remove). 
I reinstalled the INtegration via HACS. 
Applied the patch. (downloaded today)

[image]

[image]

[image]

no error!

Bit i still got the toggle switches:

[image]

And the BMS is toggeling from one state to another

[image]

[image]

sunnysingh-84, 2025-04-02:
Is this patch compatible with v1.15.0 ?

sunnysingh-84, 2025-05-07:
Is this patch compatible with v1.16.0 and onwards and if so, does it need to be reapplied after each upgrade?

patman15, 2025-05-07:
See https://github.com/patman15/BMS_BLE-HA/issues/150#issuecomment-2726945262 and yes, you need to reapply after each update.

closed 2025-05-17 by patman15